ros-navigation/navigation2 #1990: AMCL save_pose_rate parameter

pschalkwijk, 2020-09-16:
## Feature request

#### Feature description
Implement the save_pose_rate parameter for nav2_amcl. In the navigation package for ROS1 the save_pose_rate parameter is used to write the last published pose to parameters (~initial_pose_* and ~initial_cov_*), in nav2_amcl the parameter is declared and exists in documentation, but it isn't used. 
In documentation: 
Maximum rate (Hz) at which to store the last estimated pose and covariance to the parameter server, in the variables ~initial_pose_* and ~initial_cov_*. This saved pose will be used on subsequent runs to initialize the filter (-1.0 to disable).

#### Implementation considerations
I think saving the pose to a parameter could be done when the pose is published in AmclNode::publishAmclPose

Please correct me if I'm wrong and this is already implemented

SteveMacenski, 2020-09-16:
Please use TF2 to get the robot pose, don't rely on parameters. That wouldn't be something we'd be open to merging.

closed 2020-09-16 by SteveMacenski

mikeferguson, 2020-09-16:
@SteveMacenski in ROS1, the use of this is not to get the robot position - it is so that when AMCL restarts, it can re-localize to the same position as when it shut down. I'm not sure that such a use case works in ROS2 though since you don't have a common parameter server.

SteveMacenski, 2020-09-16:
Yeah, if you were to shutdown and bring up AMCL again, I'm not 100% sure what would happen. Since the parameters are declared, maybe they'll stick around, but I'm not sure when the participant shuts down if ROS2 under the hood maintains that somehow, my best understanding though is that this isn't possible since there's no central authority to "remember"

mikeferguson, 2020-09-16:
> My best understanding though is that this isn't possible since there's no central authority to "remember".

perhaps a feature could be added to save the pose to a file on shutdown (or rather, when the lifecycle shuts down) and reload it from that file (if the file exists)?

SteveMacenski, 2020-09-16:
Yeah, that's a fair assessment, though its easy to ask the question then "what else would I want to save" and a number of things come to mind. So I think this question quickly digresses into a general way to save values to persist between runs and that's a little more than I think we need to do here. This discussion has come up before, but no one's volunteered to make some generic way of saving a set of values to file and reloading on a new launch. 

But I could be wrong, if people think this is something a FOSS implementation would be nice for, we could work on it.

pschalkwijk, 2020-09-17:
Thanks for the fast responses; I indeed want to use this to reinitialize AMCL with the last known position in case of a unclean shutdown. I have other parameters I want to save (outside of navigation), and will use a generic method to save them based on the ros2cli dump verb. Writing to a database is a realistic option for me, but I'll start out with writing to a yaml file first

I opened this issue since the parameter 'save_pose_rate' suggests nav2_amcl supports this behavior. If there is no intention on implementing this, maybe this parameter should be removed.

SteveMacenski, 2020-09-17:
I think that's fair, a PR to remove this would be appreciated. I'm not sure how we overlooked that.