ros-navigation/navigation2 #1991: loading backup recovery in behavior tree xml blocks the bt_navigator configuration

tal-grossman, 2020-09-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.3-1focal.20200911.235003
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
- implement backup recovery the loaded behavior tree xml (could be the only recovery or one of many recoveries)
 

```
        <SequenceStar name="RecoveryActions">
          <BackUp backup_dist="-0.4" backup_speed="0.2"/>
        </SequenceStar>
```

#### Expected behavior
navigation launch will be configured and loaded correctly, and when needed robot will back up as defined.
#### Actual behavior
bt_navigator in stuck in configuration step due to  action_client_->wait_for_action_server() not getting any response. 

#### Additional information

#####  bug reason:
Action client _/bt_navigator_rclcpp_node_is defined to action named: **_back_up_**
while
Action server _/recoveries_server_ is defined to action named: **_backup_**

##### Proposed Fix:
in _nav2_behavior_tree/plugins/action/back_up_action.cpp_ name the action client **_backup_**:
```
#include "behaviortree_cpp_v3/bt_factory.h"
BT_REGISTER_NODES(factory)
{
  BT::NodeBuilder builder =
    [](const std::string & name, const BT::NodeConfiguration & config)
    {
      return std::make_unique<nav2_behavior_tree::BackUpAction>(
        name, "backup", config);
    };

  factory.registerBuilder<nav2_behavior_tree::BackUpAction>("BackUp", builder);
}
```

SteveMacenski, 2020-09-17:
Is your configuration in recovery server incorrect? https://github.com/ros-planning/navigation2/blob/main/nav2_recoveries/include/nav2_recoveries/recovery.hpp#L108 it pulls the name from your own configuration file. It looks like `backup` appears in both places https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/bringup/params/nav2_params.yaml#L267.

SteveMacenski, 2020-09-17:
I do see in foxy-devel that its different (https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_behavior_tree/plugins/action/back_up_action.cpp#L78). I'm not 100% sure the right action here. On one hand, you're right, by default it doesn't work correctly. On the other, its an ABI breaking change that shouldn't happen within a distribution.

@mikeferguson any thoughts?

SteveMacenski, 2020-10-06:
Closing from lack of response

closed 2020-10-06 by SteveMacenski