ros-navigation/navigation2 #1992: Feature request within nav2_waypoint_follower

jediofgever, 2020-09-22:
## Feature request

#### Feature description
`nav2_waypoint_follower` is useful. I had like to see an additional parameter for changing behavior of robot in between given poses.  The parameter can be basically a double which should stand for a amount of time for robot to wait/sleep before going to next pose, if the parameter is zero the behavior of robot stays as in original implementation.  

#### Implementation considerations
The feature implemenation is relatively simple but I am not sure what are your opinions, or whether this is useful for greater part of navigation developers. 
kindly let me know

SteveMacenski, 2020-09-22:
Feel free to submit a PR, I'd merge it. Shouldn't be more than 10 lines and 1 new parameter.

jediofgever, 2020-09-23:
Thanks @SteveMacenski, Please see https://github.com/ros-planning/navigation2/pull/1993, for implementation. 
As for the option you described;
I agree that a service/action like mechanism will be better as we will have direct feedback instead of a fixed amount of time to take the decision. 
I might work on that, but not sure if these two should coexist or only the latter should take the place.

closed 2020-10-08 by SteveMacenski