ros-navigation/navigation2 #1994: Navigation2 foxy launch fail on Yocto dunfill

rafaelhuang-intel, 2020-09-23:
## Bug report

**Required Info:**

- Operating System:
  - Yocto
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - 0.4.3
- DDS implementation:
  - fastrtps
#### Steps to reproduce issue

```
Run command:
ros2 launch nav2_bringup bringup_launch.py use_sim_time:=True autostart:=True map:=/home/root/test/map/map.yaml
```

Fail log:

[navigation2_simulate_error_log_20200923_1.txt](https://github.com/ros-planning/navigation2/files/5266155/navigation2_simulate_error_log_20200923_1.txt)

I also report this issue to meta-ros (https://github.com/ros/meta-ros/issues/743 ). 

after check navigation2 launch guide (https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_bringup/bringup), it's seems still use dashing source and TB3 package, but TB3 still in porting yet, so what can I launch navigation2 in simulate of ROS2 foxy?

SteveMacenski, 2020-09-23:
The errors relate to rosidl not nav2, closing as this isn't in our scope.

closed 2020-09-23 by SteveMacenski

rafaelhuang-intel, 2020-09-24:
@SteveMacenski 

I see navigation2  foxy launch guide (https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_bringup/bringup), it's seems still use dashing source and TB3 package, but TB3 still in porting yet, so what can I launch navigation2 in simulate of ROS2 foxy?

rafaelhuang-intel, 2020-09-24:
@SteveMacenski 

I list all Yocto dunfill dashing & foxy rosidl package list, after compare, it's seems only version update and foxy more than dashing, so could you let me know which rosidl packge will be depended by Navigation2?

[navigation2_launch_fail_issue_rosidl_package_list_20200924_1.txt](https://github.com/ros-planning/navigation2/files/5272987/navigation2_launch_fail_issue_rosidl_package_list_20200924_1.txt)

SteveMacenski, 2020-09-24:
I don't have a good answer for you. This relates to yocto probably. We don’t need any particular rosidl stuff beyond typical compiling and loading of messages generated. There’s nothing special about Nav2 in that regard.

rafaelhuang-intel, 2020-09-24:
@SteveMacenski 

for my other question is how to run navigation2 foxy version on simulation (gazebo), I see the tutorails (https://navigation.ros.org/tutorials/index.html), no simulate in list?  and TB3 still in porting, so what can I do while no TB3 package in foxy?