ros-navigation/navigation2 #1999: Nav2 plans path through walls when there is no other way to reach the goal

pfedom, 2020-09-27:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version: Foxy
#### Steps to reproduce issue
Set goal pose to a position the robot can not reach. For example close the door after the robot entered the room and than set the goal pose outside the room. 
It also happens when setting the goal pose outside of the turtlebot arena in the "Getting Started" Tutorial.

#### Expected behavior
The robot searches for a path to get to the point and stops with an error message when it can't reach the goal. 
#### Actual behavior
The pathplanner plans a way through a wall. The robot drives to the wall, slows down a bit but keeps going and hits the wall. 
#### Additional information
I use an omnidirectional robot platform with one laserscanner in front (180 degrees) and the Intel T265 camera as odometry signal. While this test slam_toolbox was running in online_async mode.
I encountered the same behavior when running nav2 without slam and prerecorded map.

----
Is there a parameter or something else that needs to be set so the wall has maximum cost and the robot can never plan a path through? 
Picture of the wrong planned path:
[image]

Here my nav2_params.yaml:
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.25
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 25.0
    laser_min_range: 0.05
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "omnidirectional"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.9
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan
amcl_map_client:
  ros__parameters:
    use_sim_time: False
amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False
[... 199 lines cut ...]
  ros__parameters:
    use_sim_time: False
recoveries_server:
  ros__parameters:
    costmap_topic: local_costmap/costmap_raw
    footprint_topic: local_costmap/published_footprint
    cycle_frequency: 10.0
    recovery_plugins: ["spin", "backup", "wait"]
    spin:
      plugin: "nav2_recoveries/Spin"
    backup:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2
robot_state_publisher:
  ros__parameters:
    use_sim_time: False

SteveMacenski, 2020-09-28:
Please reformat your ticket and provide specific reproduction instructions in the TB3 default world (and any parameter changes from default you had to make to have this occur).

pfedom, 2020-09-29:
With turtlebot its all still default. 
On the real robot, the major changes should be in the DWB Controller the velocities and accelerations. Furthermore the robot has a rectengular footprint and not a radius like the turtlebot. 
Like mentioned in the edit: The turtlebot slows down but keeps moving and than hits the wall (see the gif I added) 
I think setting a bigger inflation radius in my real setup should slow down the process because the robot slows down earlier. Do I understand this right?

SteveMacenski, 2020-09-29:
I don't see any attempt at planning into that space from your gif, I just see a robot driving into a wall which is inside of the minimim range of the laser. It wouldn't surprise me if the laser gazebo plugin doesn't properly handle collision objects there resulting in that little nub in the map. But I don't see any action items here.

gramss, 2020-09-30:
Cann't get it to display nice, but hope this makes this issue clearer
[image]

SteveMacenski, 2020-09-30:
This is super strange, I was just testing on my branch working on Hybrid-A* and I don't see this at all, but current main branch does, thanks for bringing this up. So something's changed letting this happen from after Jun 17.

naiveHobo, 2020-09-30:
It looks like the bug was introduced in e850db02ba88e03e2b6519e2b2ad51bbdc9f97f6 after the fix in `RecoveryNode` logic. I reverted this patch and everything seems to work as expected but that only happens because `RecoveryNode` doesn't retry `ComputePathToPose` again after it fails the first time and the planner specific recovery is called (which is what this patch was supposed to fix).

Even after reverting e850db02ba88e03e2b6519e2b2ad51bbdc9f97f6,  this bug can still be reproduced by increasing the number of retries in the planning branch to anything > 1:

```xml
<RecoveryNode number_of_retries="2" name="ComputePathToPose">
    <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
    <ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
</RecoveryNode>
```

naiveHobo, 2020-09-30:
So I figured out the actual reason behind this bug.

Currently, when you give a goal when there's no valid path to the goal the first `ComputePathToPose` call will fail since the planner is not able to plan a valid path. The `RecoveryNode` then calls `ClearEntireCostmap` and clears the global costmap. In the very next iteration within a fraction of a second 'RecoveryNode` retries 'ComputePathToPose' and this time the planner plans on an empty global costmap since the global costmap was cleared and hasn't spawned back up yet. The planner is able to find a straight path to the invalid goal, `ComputePathToPose` succeeds and we have the bug!!

I added a 1 second sleep just above this line to test this and it check out, everything works as expected:
https://github.com/ros-planning/navigation2/blob/506384f2bcf32978428278f65567bfaf429d6d63/nav2_planner/src/planner_server.cpp#L238

SteveMacenski, 2020-10-01:
Where was this originally introduced? Was it from a BT update to include clearing costmaps as a failure action in planning? I don't see how the commit hash you linked to could cause that issue (since N has been > 1 I believe always). 

I see the server_timeout for clear costmaps is 10ms https://github.com/ros-planning/navigation2/blob/f237751f557b2e921e08f96d4749bfe3f2b2b091/nav2_bt_navigator/src/bt_navigator.cpp#L126 and looking at the service caller https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L102 we shouldn't be returning from the service node until after its completed the clear costmaps action.

So I'd be curious what the return code from `check_future` is. If its timeout, we just need to increase that timeout, if its success, then there's something wrong in costmap_2d because we should not be returning from the service until things are good again.

naiveHobo, 2020-10-01:
> Where was this originally introduced?

https://github.com/ros-planning/navigation2/pull/1823 fixed a recovery node error because of which the main child of the recovery node was not retried after executing the second child. Before this patch, the flow of the recovery node was:
first child (fails) -> second child (succeeds) -> recovery node (succeeds)

After this path, the flow of the recovery node was changed to:
first child (fails) -> second child (succeeds) -> first child (succeeds) -> recovery node (succeeds)

>  I don't see how the commit hash you linked to could cause that issue (since N has been > 1 I believe always).

The number of retries for the planner specific recovery node has always been 1 because of which this issue was masked before https://github.com/ros-planning/navigation2/pull/1823
https://github.com/ros-planning/navigation2/blob/a30816f2939e441f2054df0f40775adb76b57997/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml#L11

I don't think the clear costmap service returns a timeout, the service executes correctly and ends up clearing the costmap. I think this issue is caused because of the time it takes to rebuild the costmap after it has been cleared. The planner server calls `getPlan()` right after the clear costmap service executes which makes me think that the planner actually ends up planning on the cleared costmap before the sensor data is received and the costmap is rebuilt again.

The timeout I added just before `getPlan()` is called gives the costmap enough time to receive sensor data and rebuild its layers after which the planner is able to plan correctly. This should confirm that the planner ends up planning on an empty costmap if it's called right after the clear costmap service executes and returns success.

SteveMacenski, 2020-10-01:
> After this path, the flow of the recovery node was changed to:
> first child (fails) -> second child (succeeds) -> first child (succeeds) -> recovery node (succeeds)

For  the recovery node, I don't think that's good behavior, I think the original was correct. If the planner is now passing, we shouldn't be triggering the clear costmap again. Imagine this recovery was backing up or something, we'd be backing up, in a good state, and then backing up again.

> The timeout I added just before getPlan() is called gives the costmap enough time to receive sensor data and rebuild its layers after which the planner is able to plan correctly. This should confirm [...]

I would just make sure its not timing out, because that's something we can more easily do something about. If it really just needs time, that becomes more complicated to deal with. We'd need to keep flags between clears and updates so that we can wait until the costmap is valid again to use.

naiveHobo, 2020-10-01:
> If the planner is now passing, we shouldn't be triggering the clear costmap again. Imagine this recovery was backing up or something, we'd be backing up, in a good state, and then backing up again.

I think you misinterpreted my comment. The `RecoveryNode` is the parent node which triggers its second child when the first child fails. The second child is assumed to be the recovery behavior that you want to execute when your main execution branch (`RecoveryNode`'s first child) fails. Taking the planning branch as an example:

```xml
<RecoveryNode number_of_retries="1" name="ComputePathToPose">
    <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
    <ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
</RecoveryNode>
```

Before #1823, the `RecoveryNode` would have returned success right after `ClearEntireCostmap` returns success without re-ticking `ComputePathToPose`. The flow in that case would be:
ComputePathToPose (fails) -> ClearEntireCostmap (succeeds)
At this point, `RecoveryNode` just returns success without going back to `ComputePathToPose`.

After #1823, the flow becomes:
ComputePathToPose (fails) -> ClearEntireCostmap (succeeds) -> ComputePathToPose (succeeds)
The first child of `RecoveryNode` is executed again after a successful recovery (second child) executes. At this point `RecoveryNode` returns success. This should be the expected behavior right?

> I would just make sure its not timing out, because that's something we can more easily do something about.

I checked the return code from the `ClearEntireCostmap` service, it returns success. 100% sure now that it's the planner planning on an empty costmap issue.

SteveMacenski, 2020-10-01:
> This should be the expected behavior right?

That does sound right I misread ` recovery node (succeeds)` part - I thought that was meaning the recovery behavior action. God, the nomenclature we used in this is going to be messing with my head for years to come, isn't it? 

> I checked the return code from the ClearEntireCostmap service, it returns success. 100% sure now that it's the planner planning on an empty costmap issue.

Ok. 

My knee-jerk solution would be to add a param to `Costmap2DROS` about whether it was cleared, and if so, it needs to wait for a new update before planning.

SteveMacenski, 2020-10-14:
@pfedom @naiveHobo please review / test this patch

simutisernestas, 2020-11-02:
@SteveMacenski I'm getting similar behavior at regular intervals. I thought that this patch could fix that, but unfortunately, it doesn't help at all. The main difference is that I'm not using amcl, but the map and localization is provided by SLAM at 1 Hz. The rest of the configuration is the same. By any chance, these are related? You can see path planning failures in attached screenshots, sometimes it just goes straight through obstacles. I could provide step by step reproduction here if that's relevant or should I fill in a new issue?

[image]

[image]

[image]

SteveMacenski, 2020-11-09:
I'd very much like to merge a solution for this this week, will be more actively dealing with #2009 and #2044 

@simutisernestas please use our default configs of things to test against these patches. We don't need to introduce any additional changes to fix unrelated bugs

SteveMacenski, 2020-11-13:
https://github.com/ros-planning/navigation2/pull/2090 merging imminently to resolve

closed 2020-11-13 by SteveMacenski

Kaetera, 2021-05-27:
Hello, 
I'm still seeing this behaviour on Foxy. In particular, the path is planned through walls if there is no path possible in the local costmap. 

OS: Unbuntu 20.04 64bit, running on Intel i5 x86_64
ROS version: ROS2 Foxy, installed by apt
Nav2 version:  ros-foxy-navigation2  0.4.7-1focal.20210423.031543
DDS: Default Fast-RTPS

When a path through a wall is planned, I got warnings in the console like:
`
[WARN] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[WARN] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-5.18, -0.64)
[WARN] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
`
so the fact that the path is invalid is effectively detected, but the robot still starts to follow it (cmd_vel computed) and it goes straight into the wall. 
I was expecting that if the path planning fails, either no path is publish, or at least the robot won't move. Am I correct ? Or am I misunderstanding something ? Or maybe the binary packages do not include yet this patch ?

[image]

nedo99, 2021-10-05:
Is there a solution for Foxy? PRs here cannot be applied to `foxy-devel` branch.

ghhhgfgh, 2025-09-05:
How to avoid this problem? Can I turn off some nodes or modify some code in BT to avoid it? Because the robot I use is Ubuntu 20, I can only use the foxy version of ROS.