ros-navigation/navigation2 #2003: behavior tree PipelineSequence doesn't reset properly on NavigateToPose action cancel.

Jconn, 2020-09-30:
## Bug report

**Required Info:**

- Operating System:
Ubunto 18.04
- ROS2 Version:
Foxy
- Version or commit hash:
98e994fa691
- DDS implementation:
Any

#### Steps to reproduce issue
- Set a goal through rviz
- Cancel goal 
- set a goal through rviz

I am running the [navigate_w_replanning_and_recovery](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml) behavior tree. I find that when I set a goal through rviz, cancel that goal through rviz, and then set another goal, My FollowPath node is ticked before my ComputePathToPose node completes. 

This makes my robot follow my old path until my ComputePathToPose action completes (which on my setup ends up being only one loop of the local planner). 

If I let a goal complete successfully, and then send another goal, FollowPath is properly gated on ComputePathToPose completing. 

I think the problem is that `last_child_ticked_` in [pipeline_sequence.cpp
](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp) is only being reset on a node failure or if all nodes have ticked success, but does not reset on a halt. 

If I add `last_child_ticked_ = 0;` to `PipelineSequence::halt()`, my issue stops reproducing.

SteveMacenski, 2020-09-30:
@Jconn can you verify @naiveHobo 's patch fixes this?

SteveMacenski, 2020-10-01:
OK, I'll link the PR so it closes when we merge it

closed 2020-10-02 by SteveMacenski