ros-navigation/navigation2 #2006: CodeCov project has been failing on every PR for weeks

SteveMacenski, 2020-10-01:
For a couple weeks, the codecoverage patch report has been OK but the project report has not been succeeding. Its been failing after 1s of running and fails to produce results causing PRs to not be all green. It also isn't updating the repo's main report for the coverage statistics. 

@ruffsl any idea?

ruffsl, 2020-10-01:
@SteveMacenski , from looking at the most recent nightly debug test job, it seems that the coverage results are still being uploaded to codecov successfully.

https://app.circleci.com/pipelines/github/ros-planning/navigation2/3854/workflows/7e50ddd4-7252-4648-8e14-87d8587b352e/jobs/14911/parallel-runs/0/steps/0-117

You can even view the reports via the link in the stdout next to: `View reports at ...`

https://codecov.io/github/ros-planning/navigation2/commit/a30816f2939e441f2054df0f40775adb76b57997

I'm not sure why these results aren't being reflected from the analytics panel for the main branch statistics. Perhaps @crdelsey may spot the configuration error. Did this regression occur around the time we migrated from master to main as the default branch name? https://github.com/ros-planning/navigation2/issues/1873

ruffsl, 2020-10-01:
It looks like codecov still thinks the master branch as the default branch name?

[image]

https://codecov.io/gh/ros-planning/navigation2/branches?view=contributed&_pjax=%23pjax-inner

@SteveMacenski , do you have access to the admin panel for codecov to update that? Or I think we could update the `codecov.yml` as well:

https://docs.codecov.io/docs/codecov-yaml#master-copy

The `Yaml not found` message from the test job's stdout is still odd though:

```
==> Circle CI detected.
    project root: .
    Yaml not found, that's ok! Learn more at http://docs.codecov.io/docs/codecov-yaml
    -> Found 1 reports
```

SteveMacenski, 2020-10-01:
Yes, I was able to change it, good catch. Maybe that's what caused it?

ruffsl, 2020-10-01:
The `Yaml not found` message might be related to this: https://github.com/codecov/codecov-bash/issues/100

ruffsl, 2020-10-01:
> Yes, I was able to change it, good catch. Maybe that's what caused it?

Did you change that from an admin panel? I wonder what the behavior is like for folks just now forking the repo, and what codecov things to initialize as the default branch.

SteveMacenski, 2020-10-01:
I changed it from settings, I'm not sure the admin-ness of it.

closed 2020-10-02 by SteveMacenski