ros-navigation/navigation2 #2011: nav2_waypoint_follower fails to build in binding of goalResponseCallback function

jediofgever, 2020-10-05:
I am unsure if this is specific to my platform, I tried to build latest `main` without success, I update && upgraded my system but seems like I cannot build `nav2_waypoint_follower`.

The error is being triggered when binding `goalResponseCallback`
 
Is there anyone meeting same problem. Thank you for your time. 

## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04

- ROS2 Version: 
  Foxy Debian install,  

 - Version or commit hash:
   https://github.com/ros-planning/navigation2/commit/64871a7283a08eb08d7365c6279d0b330155f7f6

- DDS implementation:
  ( Fast-RTPS,.)

#### Steps to reproduce issue
Follow steps on  [Manually Build Main](https://navigation.ros.org/build_instructions/index.html)
at `navigation.ros.org`

#### Expected behavior
Build all nav2 packages without errors. 

#### Actual behavior
`nav2_waypoint_follower` package fails 

#### Additional information

`colcon build --symlink-install
`

gives following stderr output; 

```
--- stderr: nav2_waypoint_follower                                
/home/atas/colcon_ws/src/navigation2/nav2_waypoint_follower/src/waypoint_follower.cpp: In member function ‘void nav2_waypoint_follower::WaypointFollower::followWaypoints()’:
/home/atas/colcon_ws/src/navigation2/nav2_waypoint_follower/src/waypoint_follower.cpp:196:87: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<nav2_msgs::action::NavigateToPose>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>’} and ‘std::_Bind_helper<false, void (nav2_waypoint_follower::WaypointFollower::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&), nav2_waypoint_follower::WaypointFollower*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>’})
  196 |         std::bind(&WaypointFollower::goalResponseCallback, this, std::placeholders::_1);
      |                                                                                       ^
In file included from /usr/include/c++/9/future:48,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/atas/colcon_ws/install/nav2_util/include/nav2_util/node_thread.hpp:20,
                 from /home/atas/colcon_ws/install/nav2_util/include/nav2_util/lifecycle_node.hpp:22,
                 from /home/atas/colcon_ws/src/navigation2/nav2_waypoint_follower/include/nav2_waypoint_follower/waypoint_follower.hpp:22,
                 from /home/atas/colcon_ws/src/navigation2/nav2_waypoint_follower/src/waypoint_follower.cpp:15:
/usr/include/c++/9/bits/std_function.h:462:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(const std::function<_Res(_ArgTypes ...)>&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >}]’
  462 |       operator=(const function& __x)
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:462:33: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (nav2_waypoint_follower::WaypointFollower::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&), nav2_waypoint_follower::WaypointFollower*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>’} to ‘const std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>&’
  462 |       operator=(const function& __x)
      |                 ~~~~~~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:480:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::function<_Res(_ArgTypes ...)>&&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >}]’
  480 |       operator=(function&& __x) noexcept
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:480:28: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (nav2_waypoint_follower::WaypointFollower::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&), nav2_waypoint_follower::WaypointFollower*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>’} to ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>&&’
  480 |       operator=(function&& __x) noexcept
      |                 ~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:494:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::nullptr_t) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >}; std::nullptr_t = std::nullptr_t]’
  494 |       operator=(nullptr_t) noexcept
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:494:17: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (nav2_waypoint_follower::WaypointFollower::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&), nav2_waypoint_follower::WaypointFollower*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>’} to ‘std::nullptr_t’
  494 |       operator=(nullptr_t) noexcept
      |                 ^~~~~~~~~
[... 3 lines cut ...]
/usr/include/c++/9/bits/std_function.h:523:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9/bits/std_function.h: In substitution of ‘template<class _Res, class ... _ArgTypes> template<class _Cond, class _Tp> using _Requires = typename std::enable_if<_Cond::value, _Tp>::type [with _Cond = std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>::_Callable<std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>, std::__invoke_result<std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>&, std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > > > >; _Tp = std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>&; _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >}]’:
/usr/include/c++/9/bits/std_function.h:523:2:   required by substitution of ‘template<class _Functor> std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>::_Requires<std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>::_Callable<typename std::decay<_Tp>::type, std::__invoke_result<typename std::decay<_Tp>::type&, std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > > > >, std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>&> std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::Navi [... cut ...]
      |                                                                                       ^
make[2]: *** [CMakeFiles/waypoint_follower_core.dir/build.make:82: CMakeFiles/waypoint_follower_core.dir/src/waypoint_follower.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: CMakeFiles/waypoint_follower_core.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
---
Failed   <<< nav2_waypoint_follower [3.84s, exited with code 2]

Summary: 38 packages finished [11.7s]
  1 package failed: nav2_waypoint_follower
  1 package had stderr output: nav2_waypoint_follower

```

naiveHobo, 2020-10-05:
Following #1997 there were some changes introduced in upstream ROS2 packages which introduced the `nav2_waypoint_follower` bug in `goalResponseCallback` which were subsequently fixed in #2000. When you build navigation2 main from source, you're expected to have built ROS2 master from source as well (https://navigation.ros.org/build_instructions/index.html#manually-build-main).

jediofgever, 2020-10-05:
I see , thanks for pointing it out.

closed 2020-10-05 by jediofgever