ros-navigation/navigation2 #2013: Fail to build eloquent-devel branch

mirellameelo, 2020-10-05:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
  - 
- ROS2 Version: Eloquent
  - 
- Version or commit hash: eloquent-devel
  - 
- DDS implementation:
  - 

#### Steps to reproduce the issue

I was following the tutorial from the [build instructions](https://github.com/ros-planning/navigation2/blob/eloquent-devel/doc/BUILD.md#quickstart-using-initial-setup-script) file. **Since master branch do not exist**, I changed:
```
wget https://raw.githubusercontent.com/ros-planning/navigation2/master/tools/ros2_dependencies.repos
to
wget https://raw.githubusercontent.com/ros-planning/navigation2/eloquent-devel/tools/ros2_dependencies.repos
```
Then I faced the following error:

```
ros2_nav_dependencies_ws/src/image_common/camera_info_manager/src/camera_info_manager.cpp:46:10: fatal error: rcpputils/get_env.hpp: No such file or directory
 #include "rcpputils/get_env.hpp"
```
[In ros2_dependencies.repos file](https://github.com/ros-planning/navigation2/blob/eloquent-devel/tools/ros2_dependencies.repos#L13), I changed the version "ros2" for "eloquent" regarding the repositories "image_common" and "gazebo_ros_pkgs" - because I also had errors with that.
I succeeded in building the dependencies. So I moved on in the  [README](https://github.com/ros-planning/navigation2/blob/eloquent-devel/doc/BUILD.md#quickstart-using-initial-setup-script) file. So I got new errors. Such as:

```
navigation2_ws/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:85:35: error: no matching function for call to ‘nav2_behavior_tree::PipelineSequence::haltChildren()’
         ControlNode::haltChildren();

navigation2_ws/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:109:29: error: no matching function for call to ‘nav2_behavior_tree::PipelineSequence::haltChildren()’
   ControlNode::haltChildren();

navigation2_ws/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:50:13: error: ‘struct BT::Tree’ has no member named ‘rootNode’; did you mean ‘root_node’?
       tree->rootNode()->halt();

navigation2_ws/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:54:20: error: ‘struct BT::Tree’ has no member named ‘rootNode’; did you mean ‘root_node’?
     result = tree->rootNode()->executeTick();

```

#### Expected behavior
build the navigation2 packages - eloquent version

#### Additional information
I believe the main problem regards the dependencies' previous build since I could not find the master branch required in the [build tutorial](https://github.com/ros-planning/navigation2/blob/eloquent-devel/doc/BUILD.md#quickstart-using-initial-setup-script). I appreciate if you could verify it and update instructions. Thanks in advance!

SteveMacenski, 2020-10-05:
I'll merge a PR to fix the eloquent-devel instructions if you'd be so kind.

On the dependencies for `ros2`, I think you did the right thing. Can't speak to that too much though - you should be able to `apt install` those though since eloquent has the build farm binaries available.

On BT.CPP: What version are you on? I think you need 3.5.1. The BT.CPP made an API breaking release that impacted us as well so it impacts historical users that aren't on Foxy or newer.

mirellameelo, 2020-10-05:
@SteveMacenski Thank you for the fast reply!
Regarding BT.CPP I was using branch "ros2-3.1.1", but I changed for the version you suggested (and I also tried with branch "ros2-version-2.x"), and then I built again the dependencies. When building the navigation2 packages I got new errors in the costmap2d package, all related to Eigen, and basically, the same: "error: enum constant in boolean context ". Once I ignore this package, I succeeded.

```
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: enum constant in boolean context [-Werror=int-in-bool-context]
     MayLinearVectorize = bool(MightVectorize) && MayLinearize && DstHasDirectAccess
```

I'm trying to figure it out, but updating it here just in case you have any more suggestions to solve it. Maybe a specific Eigen version is necessary? I'm using 3.3.4.

And, yes, I tried with [apt install](https://navigation.ros.org/getting_started/index.html). To run the example, I had an additional step of building from source the [turtlebot3 packages](https://github.com/ROBOTIS-GIT/turtlebot3/tree/eloquent-devel).

Thank you!

SteveMacenski, 2020-10-05:
Where did that Eigen error come from?

mirellameelo, 2020-10-05:
Here the full error (the other are basically the same and in this same **observation_buffer file**):
 
```
In file included from /usr/local/include/eigen3/Eigen/Core:420:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /opt/ros/eloquent/include/tf2_sensor_msgs/tf2_sensor_msgs.h:37,
                 from /navigation2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/observation_buffer.hpp:47,
                 from /navigation2_ws/src/navigation2/nav2_costmap_2d/src/observation_buffer.cpp:37:
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘struct Eigen::internal::copy_using_evaluator_traits<Eigen::internal::evaluator<Eigen::Matrix<float, 3, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, 3, 1> >, Eigen::internal::assign_op<float, float> >’:
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:607:49:   required from ‘class Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, 3, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, 3, 1> >, Eigen::internal::assign_op<float, float>, 0>’
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:739:10:   required from ‘void Eigen::internal::call_dense_assignment_loop(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<float, 3, 1>; SrcXprType = Eigen::Matrix<float, 3, 1>; Functor = Eigen::internal::assign_op<float, float>]’
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:879:31:   required from ‘static void Eigen::internal::Assignment<DstXprType, SrcXprType, Functor, Eigen::internal::Dense2Dense, Weak>::run(DstXprType&, const SrcXprType&, const Functor&) [with DstXprType = Eigen::Matrix<float, 3, 1>; SrcXprType = Eigen::Matrix<float, 3, 1>; Functor = Eigen::internal::assign_op<float, float>; Weak = void]’
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836:49:   required from ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<float, 3, 1>; Src = Eigen::Matrix<float, 3, 1>; Func = Eigen::internal::assign_op<float, float>]’
/usr/local/include/eigen3/Eigen/src/Core/PlainObjectBase.h:728:41:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set_noalias(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<float, 3, 1>; Derived = Eigen::Matrix<float, 3, 1>]’
/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:278:27:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&&) [with _Scalar = float; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’
/usr/local/include/eigen3/Eigen/src/Geometry/Transform.h:442:102:   required from ‘const typename Eigen::internal::transform_right_product_impl<Eigen::Transform<Scalar, Dim, Mode, _Options>, OtherDerived>::ResultType Eigen::Transform<Scalar, Dim, Mode, _Options>::operator*(const Eigen::EigenBase<OtherDerived>&) const [with OtherDerived = Eigen::Matrix<float, 3, 1>; _Scalar = float; int _Dim = 3; int _Mode = 2; int _Options = 0; typename Eigen::internal::transform_right_product_impl<Eigen::Transform<Scalar, Dim, Mode, _Options>, OtherDerived>::ResultType = Eigen::Matrix<float, 3, 1>]’
/opt/ros/eloquent/include/tf2_sensor_msgs/tf2_sensor_msgs.h:80:52:   required from here
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:86:63: error: enum constant in boolean context [-Werror=int-in-bool-context]
     MayLinearVectorize = bool(MightVectorize) && MayLinearize && DstHasDirectAccess
```

SteveMacenski, 2020-10-05:
> from /opt/ros/eloquent/include/tf2_sensor_msgs/tf2_sensor_msgs.h:37,

So this error isn't from Nav2, its from TF2 sensor msgs. Costmap 2d just happens to use it. You might not have aligning version of tf2_sensor_msgs with Eigen or nav2.

mirellameelo, 2020-10-06:
@SteveMacenski hmm, I just update my Eigen version to the new release (3.3.8), and the packages were built. I'll double-check it tomorrow and hopefully submit the PR. Thanks for your help!

mirellameelo, 2020-10-06:
Just submitted the PR #2018 
Closing it. Tks!

closed 2020-10-06 by mirellameelo