ros-navigation/navigation2 #2019: Codecov covering tests now

SteveMacenski, 2020-10-06:
@ruffsl from the changes, I think the codecov yaml file is being ignored. We're seeing the tests now counting towards the coverage statistics when they weren't before (and regex entries to remove them). Are you sure that it wasn't being used before? Seems like this change indicates that it was. 

https://codecov.io/gh/ros-planning/navigation2

ruffsl, 2020-10-06:
Could you link to a specific example? Perusing through the landing page of the link you sent, I wasn't able to find a test folder or files among the files indexed using the web UI. You'll also want to make sure that any PR's your observing this have been rebased off of main, so that their CI config is also current.

ruffsl, 2020-10-06:
Do you know why there doesn't seem to be any coverage reports for the last 7 days view for main branch?

https://codecov.io/gh/ros-planning/navigation2/branch/main

Is something is still misconfigured from the branch migration.

SteveMacenski, 2020-10-06:
https://codecov.io/gh/ros-planning/navigation2/tree/932d278ac0d0f4dfab25745a509841f633c01003/nav2_costmap_2d

For example, the costmap_2d package in the PR for fixing navfn pending. After the changes were merged, the Nav2 coverage shot up 4.5% in the repo because at least some of the test files were being included.

ruffsl, 2020-10-06:
Looks like there's a strange discrepancy in where codecov finds vs reads the `codecov.yml` file. From the commit linked, under the `build` tab, we can see the `.txt` file that the CI uploaded to codecov cites the config path on the first line:

`src/navigation2/.circleci/codecov.yml`
https://codecov.io/gh/ros-planning/navigation2/commit/932d278ac0d0f4dfab25745a509841f633c01003/builds

As does the codecov script:

https://app.circleci.com/pipelines/github/ros-planning/navigation2/3901/workflows/fbd2596f-c750-453c-bb6f-009c00863e78/jobs/15055/parallel-runs/0/steps/0-117

Yet, the same file is not used by the web front end. My guess is that there is a hard assumption that the root dir specified for the script must be the root of the respective git repo, the outer workspace path being erroneous for whatever post-process-filtering that reads the config file a second time from a separate git repo checkout. Looks like this might be from running the codecov script from outside the repo's root dir?

SteveMacenski, 2020-10-06:
Is there a reason we can't move the code cov yaml to the root of this repo? That seems like a fine enough solution to me

SteveMacenski, 2020-10-06:
As much as I love to that our coverage percentage is artificially inflated, I'd rather have it be actually representative and hit these numbers for real.

closed 2020-10-07 by ruffsl