ros-navigation/navigation2 #2035: Should have ability to associated unique goal checker plugin with each controller plugin

justinIRBT, 2020-10-12:
## Feature request
Each controller plugin should be able to have its own goal checker plugin if the controller_server is configured with multiple controller_plugins.

I had an implementation where I was using two controller_server plugins, the default FollowPath as well as a RotateToPath.  My behavior tree was executing the RotateToPath controller till success then initiating FollowPath.  I had a custom success criteria for RotateToPath previously implemented through nav2_core::Controller's isGoalReached API, I no longer see a way to have a custom exit criteria for my RotateToPath controller that differs from the FollowPath controller, as it appears the controller_server now takes a single goal_checker_plugin.

#### Feature description
As of this commit https://github.com/ros-planning/navigation2/commit/d7494ec8c501522200089c54947232fe86c0ef94#diff-4477edf09e4f3611fe9449cfb0bb3e0e, the controller_server accepts multiple controller_plugins but only one goal_checker_plugin.

Previously the nav2_core::Controller API included isGoalReached, so each custom controller could have its own goal reached criteria.

#### Implementation considerations
Perhaps the goal_checker_plugin could be configured against the controller id and the 
https://github.com/ros-planning/navigation2/blob/be38a7a91811e27742ca4be01dbef16436a461fe/nav2_controller/src/nav2_controller.cpp#L433
Could become:
return goal_checker[current_controller_]_->isGoalReached(pose.pose, end_pose_, velocity);

I also derived benefit from accessing local information from the custom controller (like which side of the path it started on) in determining my success criteria, so I would also need my GoalChecker to have access to the information in my Controller class.  I am assuming this could be achieved with multiple inheritance.

SteveMacenski, 2020-10-12:
>  return goal_checker[current_controller_]->isGoalReached(pose.pose, end_pose, velocity);

 That would be fine. Though I'd like to have the option that if only 1 is specified it applies it to all of the potentially loaded controllers. It sounds like if you're just rotating to align with the goal, the goal checker would do that for you because the goal would just be the rotation?

justinIRBT, 2020-10-12:
The goal exit criteria I was using with my custom RotateToPath controller was more of a crossed plane check on the path angle vs. achieving a specific orientation with a tight threshold.  This was to help for the scenario when the robot was getting close to something and replanning the path, but then FollowPath was trying to cut the corner on the newly planned path, rotating just past the path angle allowed the robot to cleanly follow the path away from the new obstacle.

Another approach could be to have a second instance of controller_server that just implements my RotateToPath controller plugin and its custom goal checker.  I would need to make the https://github.com/ros-planning/navigation2/blob/be38a7a91811e27742ca4be01dbef16436a461fe/nav2_controller/src/nav2_controller.cpp#L167 "follow_path" server name configurable to have a second name instance I could interact with from the behavior tree.

SteveMacenski, 2020-10-12:
That would be remappable to change names. Though a second instance would be a bad idea, then you'd have 2 different local costmaps operating and those are by far the heaviest things in the navigation system. It's better to have a solution to goal checking.

justinIRBT, 2020-10-12:
What would be the best option for having my goal checker access state from my controller as part of its exit criteria?  It looks like it's instantiating unique instances of the goal_checker and controller plugin with loader_.createUniqueInstance, so multiple inheritance wouldn't work because it would have a separate instance of the class for goal and control.

SteveMacenski, 2020-10-13:
I think this is a good topic for the working group discussions.  I think having some additional input from other users would be good. I can see a number of ways this could play out (any of these stick out at you?):

- Push goal checker back into the controllers statically
- Push goal checker plugin interfaces into controllers so that its their problem to load their own goal checker plugin
- Have a map of goal checkers so that each in the controller server can have its unique goal checker if it wants it
- Have a single goal checker and play around with the API giving it the controller pointer so that it can decide what to do
- Other potential options.

We can always make it a shared pointer. The unique pointer stuff was me on a binge about a year ago with overuse of shared pointers for unshared objects.

justinIRBT, 2020-10-14:
The goal checker plugin managed by the individual controller (which additionally enables a custom controller to report goal status without a plugin) or the `OR` option both sound like they would work for me.

Right now, I am trying to get similar behavior within the current design and I am running into a couple of more things that might be interesting inputs.

- I was deriving my RotateToPath goal from the full path (to get a starting angle to face the path direction).  Currently the GoalChecker API is just passing the endpoint Pose every isGoalReached call.  I could also image scenarios where you would want to have a crossed plane check for the last waypoint as a goal succeeded if you had a tight convergence radius that local ODOA avoided, where the plane would be established as the line perpendicular to the line from (end-1)->(end).  In that case, ideally the GoalChecker API would have access to the full path to derive its goal.  In this scenario, you wouldn't want to recompute the goal every cycle, so passing the Path in the `reset()` function call would allow for goal computations to happen on path changes (whether that is just caching the end pose or doing some math against the path, like to establish a plane).  My temporary plan to work around this is to send the path to my RotateToPath as just a single point with orientation that is derived from the path starting direction, then I can use the same goal plugin for both RotateToPath and FollowPath (which gets the full path).  There is still a risk the controller will be out of sync with the separate goal check plugin in terms of which side of the line it starts on, but in that case it might oscillate, but still succeed.

- A second goal criteria I was using in my RotateToPath controller was looking at the starting condition and if the robot was travelling faster than a low speed and the angle difference between robot heading and the new path was relatively small, I was returning goal succeeded immediately, effectively ignoring RotateToPath and just transitioning to the next FollowPath for a smooth transition.  This one I am having a harder time working into the current design.  I can't embed this policy in a shared goal check plugin.  My next thought was to simply have the behavior tree action node check the starting condition and return success immediately without making the server call if the starting condition was present.  Unfortunately, it looks like I will have to write my own implementation of BtActionNode to achieve this that will mostly be code duplication.  It would be nice if the API for BtActionNode interface allowed the initial `on_tick()` call from the child to return a SUCCESS/FAILED/RUNNING (returning RUNNING would be the same behavior as current, returning SUCCESS/FAIL would return that status immediately as the return value of `tick()` and not make the server call.  I might be able to just make a condition node for this condition and combine it with the tree in the right way to get this behavior, I will have to think a little harder about that.

- While mentioning condition nodes, I made a custom condition isPathBlocked that checked the first 3 meters of the global path against the costmap.  I was using the costmap subscriber (which I saw online is a little heavy, but it's a prototype for now) so I used the is_stuck_condition as an example of a condition with a subscription and I ran into the issue where my subscription never updated.  I tracked this down to the fact that the node in the blackboard isn't spun by default as part of tree ticking, it is only spun in the action node, so if you have a tree cycle that isn't running an action node (just checking conditions) the blackboard node won't be spun.  It would be nice if that was taken care of in the root, or at least the is_stuck_condition should be updated to spin the node.

SteveMacenski, 2020-10-14:
> The goal checker plugin managed by the individual controller (which additionally enables a custom controller to report goal status without a plugin) or the OR option both sound like they would work for me.

That seems like a reasonable direction to start off in. 

The bullets I didn't fully follow you. It sounds like your in the depths of working through a problem so the description is a cliff notes. If the following suggestion isn't helpful because I didn't register your aim, feel free to ignore it. It sounds like you might want to goal checker to have access to the path, that sounds like something we could potentially do.

What it sounds like to me is that when you get a path generated from a holonomic 2D planner, you want your differential drive robot to rotate to the path's starting orientation before following so you don't need your local trajectory planner to swing the robot around. I've done this before in the past pretty easily with a shimmed plugin. Essentially create a new `justin_rbt_plugin` that has itself 2 plugins: rotate to orientation and the path follower. For any new request, check the starting robot vs. path orientation. If bigger than `N` radians, send command to rotate controller. If less than N, send to path follower. Then as the robot is rotating, once it hits the threshold, it just immediately starts following giving you a smooth motion.

I'd also merge a shim-plugin like that into Nav2 as well, given you want this and I've made it before, it sounds like this is a consistent need people have, lets just support it.

justinIRBT, 2020-10-19:
The shim-plugin approach seems like it should work, I will attempt it.  Thanks!

closed 2020-10-19 by justinIRBT

SteveMacenski, 2020-10-26:
Got it - an aside: have you taken any metrics on navigation2? I'd be curious on any of your results. We haven't had much time as a community to create any performance metrics yet.