ros-navigation/navigation2 #2039: Action Server crashes when BT reloaded

mdtoom, 2020-10-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy Source
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. modify `navigateToPose` in `bt_navigator.cpp` so that it reloads the BT when executing a new goal. 
diff:
```
@@ -240,7 +240,7 @@ BtNavigator::navigateToPose()
         return true;
       }
 
-      return action_server_->is_cancel_requested();
+      return action_server_->is_cancel_requested() || action_server_->is_preempt_requested();
     };
 
   auto bt_xml_filename = action_server_->get_current_goal()->behavior_tree;
@@ -260,11 +260,11 @@ BtNavigator::navigateToPose()
   std::shared_ptr<Action::Feedback> feedback_msg = std::make_shared<Action::Feedback>();
 
   auto on_loop = [&]() {
-      if (action_server_->is_preempt_requested()) {
-        RCLCPP_INFO(get_logger(), "Received goal preemption request");
-        action_server_->accept_pending_goal();
-        initializeGoalPose();
-      }
+      // if (action_server_->is_preempt_requested()) {
+      //   RCLCPP_INFO(get_logger(), "Received goal preemption request");
+      //   action_server_->accept_pending_goal();
+      //   initializeGoalPose();
+      // }
       topic_logger.flush();
 
       // action server feedback (pose, duration of task,
@@ -304,7 +304,14 @@ BtNavigator::navigateToPose()
 
     case nav2_behavior_tree::BtStatus::CANCELED:
       RCLCPP_INFO(get_logger(), "Navigation canceled");
-      action_server_->terminate_all();
+      if (action_server_->is_preempt_requested())
+      {
+          action_server_->terminate_current();
+      }
+      else
+      {
+          action_server_->terminate_all();
+      }
       break;
   }
 }
```

It should look like:
```
void
BtNavigator::navigateToPose()
{
  initializeGoalPose();

  auto is_canceling = [this]() {
      if (action_server_ == nullptr) {
        RCLCPP_DEBUG(get_logger(), "Action server unavailable. Canceling.");
        return true;
      }

      if (!action_server_->is_server_active()) {
        RCLCPP_DEBUG(get_logger(), "Action server is inactive. Canceling.");
        return true;
      }

      return action_server_->is_cancel_requested() || action_server_->is_preempt_requested();
    };

  auto bt_xml_filename = action_server_->get_current_goal()->behavior_tree;

  // Empty id in request is default for backward compatibility
  bt_xml_filename = bt_xml_filename == "" ? default_bt_xml_filename_ : bt_xml_filename;

  if (!loadBehaviorTree(bt_xml_filename)) {
    RCLCPP_ERROR(
      get_logger(), "BT file not found: %s. Navigation canceled",
      bt_xml_filename.c_str(), current_bt_xml_filename_.c_str());
    action_server_->terminate_current();
    return;
  }

  RosTopicLogger topic_logger(client_node_, tree_);
  std::shared_ptr<Action::Feedback> feedback_msg = std::make_shared<Action::Feedback>();

  auto on_loop = [&]() {
      // if (action_server_->is_preempt_requested()) {
      //   RCLCPP_INFO(get_logger(), "Received goal preemption request");
      //   action_server_->accept_pending_goal();
      //   initializeGoalPose();
      // }
      topic_logger.flush();

      // action server feedback (pose, duration of task,
      // number of recoveries, and distance remaining to goal)
      nav2_util::getCurrentPose(
        feedback_msg->current_pose, *tf_, global_frame_, robot_frame_, transform_tolerance_);

      geometry_msgs::msg::PoseStamped goal_pose;
      blackboard_->get("goal", goal_pose);

      feedback_msg->distance_remaining = nav2_util::geometry_utils::euclidean_distance(
        feedback_msg->current_pose.pose, goal_pose.pose);

      int recovery_count = 0;
      blackboard_->get<int>("number_recoveries", recovery_count);
      feedback_msg->number_of_recoveries = recovery_count;
      feedback_msg->navigation_time = now() - start_time_;
      action_server_->publish_feedback(feedback_msg);
    };

  // Execute the BT that was previously created in the configure step
  nav2_behavior_tree::BtStatus rc = bt_->run(&tree_, on_loop, is_canceling);
  // Make sure that the Bt is not in a running state from a previous execution
  // note: if all the ControlNodes are implemented correctly, this is not needed.
  bt_->haltAllActions(tree_.rootNode());

  switch (rc) {
    case nav2_behavior_tree::BtStatus::SUCCEEDED:
      RCLCPP_INFO(get_logger(), "Navigation succeeded");
      action_server_->succeeded_current();
      break;

    case nav2_behavior_tree::BtStatus::FAILED:
      RCLCPP_ERROR(get_logger(), "Navigation failed");
      action_server_->terminate_current();
      break;

    case nav2_behavior_tree::BtStatus::CANCELED:
      RCLCPP_INFO(get_logger(), "Navigation canceled");
      if (action_server_->is_preempt_requested())
      {
          action_server_->terminate_current();
      }
      else
      {
          action_server_->terminate_all();
      }
      break;
  }
}
```
2.
Create 2 behaviour tree XML files.
bt_1.xml
```
<?xml version="1.0"?>
<root main_tree_to_execute="BehaviorTree">
    <BehaviorTree ID="BehaviorTree">
        <Action ID="Fibonacci" order="10"/>
    </BehaviorTree>
</root>
```
bt_2.xml
```
<?xml version="1.0"?>
<root main_tree_to_execute="BehaviorTree">
    <BehaviorTree ID="BehaviorTree">
        <Action ID="Fibonacci" order="11"/>
    </BehaviorTree>
</root>
```
3. Run the fibonacci_action_server (`ros2 run action_tutorials_cpp fibonacci_action_server`) and the bt_navigator
4. Start the first BT:
```
ros2 action send_goal /navigate_to_pose nav2_msgs/action/NavigateToPose "behavior_tree: '{PATH_TO_BT_XML}/bt_1.xml'"
```
5. While the BT runs interrupt it from another terminal by executing bt_2.xml:
```
ros2 action send_goal /navigate_to_pose nav2_msgs/action/NavigateToPose "behavior_tree: '{PATH_TO_BT_XML}/bt_2.xml'"
```

#### Expected behavior
The action server should preempt and restart `navigateToPose` and execute the new BT.

#### Actual behavior

The fibonacci action server crashes with:
```
[INFO] [1602574356.496290013] [fibonacci_action_server]: Publish feedback
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.0/src/rmw_response.cpp:124, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/service.c:342

```

#### Additional information

What seems to happen is that after the cancel request has accepted and the action is terminated, the fibonacci action server (and as far as I tested any server using rclcpp::action) still sends a response to the old action client (the one in the behaviour tree node).
But because the behavior tree has been removed the fibonnaci action server cannot send the response back and crashes.

----

mdtoom, 2020-10-13:
Fix proposed in https://github.com/ros-planning/navigation2/pull/2033

SteveMacenski, 2020-10-13:
So much has changed here, there's no support we can offer for this configuration. If you'd like to upstream some of those changes because they add some features you require, we can discuss that. But for now, you've changed the behavior trees substantially, the BT navigator substantially, and a toy BT node not in our stack. We obviously can't really support or dive much into that. Very little here is the same.

But it seems like your actual issue is that you want to preempt a running navigation request with a new behavior tree in that request and its not working correctly- is that accurate? 

If so, I have no idea why that PR is playing with the simple action server, that should have nothing to do with it. Further, we're aware of that issue and have some acknowledgement  that preemption requests with new BTs will ignore the new BT (@naiveHobo found that). That's an issue we can work on.

But your PR has nothing to do with enabling new BTs on a preemption request. You're spinning to get a result, which is unrelated. Also your description in that PR is not the same as the issue you described here. I think you need to work through a concise and specific statement(s) of issue(s) that we can work on addressing. Please provide us with a bulleted list of **separate** issues you're seeing that you would like a resolution to. 

It seems like at minimum you have an issue with changing BTs on preemption requests and some other issue relating to the SimpleActionServer that shouldn't be fixed by requiring all users get a result but rather properly fixing it in SimpleActionServer.

mdtoom, 2020-10-14:
I agree with you.
Is changing BT's going to be addressed in https://github.com/ros-planning/navigation2/pull/2010?
It looks like a good idea to wait for that MR before changing anything in the bt_navigator.

Regarding the crashing action server. The Fibonacci action server does not use the SimpleActionServer but rclcpp_action::Server<T>. I just used it as an example. I thought it might be good to address this issue, because it might hinder others in creating their own BT action nodes (for example to move an arm) and interface with already existing action servers that don't use the SimpleActionServer. But agree with you that it can better be fixed in rclcpp and #2033 is a symptom fix.
I will try to make a minimal example and open an issue there.

SteveMacenski, 2020-10-14:
No, #2010 is just a refactor. We can deal with changing BTs in preemption in another PR. That would involve a substantive change.  

A crashing issue seems more important than a refactor, we should figure this out first. You didn't provide any specific and enumerated issues you think need to be resolved. Please do so, so we can get started in working through these issues you're experiencing. Else, we can't help.

SteveMacenski, 2020-10-14:
https://github.com/ros-planning/navigation2/issues/1971 this ticket deals with the BT change on preemption issue

SteveMacenski, 2020-10-16:
@mdtoom any feedback?

mdtoom, 2020-10-17:
I am sorry for the late reaction.
Moving to #1971 for the BT change on preemption issue.

Regarding the crashing issue:
- It also happens when i use `<Action ID="BackUp" backup_dist="1.0" backup_speed="-0.2"/> ` in the behavior XML instead of <fibonaccy> and then preempt the bt_navigator with that BT.
- however the problem does not occur when using cyclone, so it is related to fastrtps.
- But I am not able to trigger the crash when writing a minimal example that follows the flow of the preempted bt_navigator and looks like: 
```
#include <action_tutorials_interfaces/action/fibonacci.hpp>
#include <rclcpp/rclcpp.hpp>
#include <rclcpp_action/rclcpp_action.hpp>

#include <memory>

void request_fib(rclcpp::Node::SharedPtr node)
{
    auto action_client_ =
        rclcpp_action::create_client<action_tutorials_interfaces::action::Fibonacci>(node, "/fibonacci");
    action_client_->wait_for_action_server();

    action_tutorials_interfaces::action::Fibonacci::Goal goal;
    goal.order              = 10;
    auto future_goal_handle = action_client_->async_send_goal(goal);

    if (rclcpp::spin_until_future_complete(node, future_goal_handle) != rclcpp::executor::FutureReturnCode::SUCCESS)
    {
        throw std::runtime_error("send_goal failed");
    }

    // Sleep to make the goal running.
    rclcpp::sleep_for(std::chrono::seconds(3));

    // Cancel the current goal
    auto future_cancel = action_client_->async_cancel_goal(future_goal_handle.get());
    if (rclcpp::spin_until_future_complete(node, future_cancel) != rclcpp::FutureReturnCode::SUCCESS)
    {
        RCLCPP_ERROR(node->get_logger(), "Failed to cancel action server");
    }

    // I suspected this was the cause, but that isn't the case appearantly
    action_client_.reset();
}

int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);

    auto node = std::make_shared<rclcpp::Node>("test");
    request_fib(node);
    request_fib(node);
    rclcpp::shutdown();

    return 0;
}
```
My hunch here was that the response that is send by the Fibonacci action server after the cancel cannot be send because the `action_client_` was cleared causing the crash. But it doesn't look like that is the case.
- You can close the issue, as it is not related directly to navigation2. I will look into it when I have time. When I can reproduce it I will open an issue on fast_rtps' repo.

SteveMacenski, 2020-10-26:
> however the problem does not occur when using cyclone, so it is related to fastrtps.

Closing since Fast-DDS related and not a Nav2 bug. Please switch to Cyclone then.

closed 2020-10-26 by SteveMacenski