ros-navigation/navigation2 #2045: DWB: Improve behavior (tuning, critic improvement, more samples, etc)

SteveMacenski, 2020-10-15:
After getting more familiar with DWB in recent weeks, I think DWB is worth our time to really drill into and improve behavior and performance. DWA in ROS1 had a bunch of issues and overall didn't have the kind of collision avoidance and path following behavior that ultimately made people want to use it in a product or refined research robot.

I'd like to change that for DWB in ROS2. That may involve tuning, adding, and replacing DWB critics to better support better collision avoidance behavior and path following stability. It could also include new trajectory generators (also to support ackermann style vehicles), more samples and optimize runtime with parallelism, or other.

SteveMacenski, 2020-10-16:
anti-wobble critics? A critic that keeps track of a local window of past trajectories and weights to keep going in a direction for smooth motion

BriceRenaudeau, 2021-09-17:
Good critics could be max_vel

dt1729, 2022-03-30:
Hi all, are you guys planning to implement  these in any particular order of preference or can we pick any feature and start working on that?

SteveMacenski, 2022-03-30:
I don't think there is a particular order, but if you have access to hardware robots, helping with tuning the system as it exists today (which critics to use, weights, sample numbers, etc) would be of the most and significant value to the project. 

But certainly if you were interested in contributing in other places, I would never decline community involvement. If you wanted to write some code on this, I think the ackermann trajectory generator would be a good place to start and wouldn't impact any tuning.

dt1729, 2022-04-02:
Hi Steve, I will start work on implementing Ackermann trajectory generator and present something in the next nav stack meeting. Where can I look for a Class tree of the repository? ROSDocs?

SteveMacenski, 2022-04-04:
We don't have one self hosted, but if you generate the doxygen, I believe that will generate them. But you should only need to worry about the trajectory generator plugin base class that you're implementing (See the 2 example generators already existing for examples)

SteveMacenski, 2022-04-04:
@DLu I don't know if you have some time, but helping to tune the defaults for DWB for good out of the box behavior would be super, super helpful to making sure folks get a feel for its power

DLu, 2022-04-11:
@SteveMacenski Is there an existing demo where DWB doesn't do great?

SteveMacenski, 2022-04-11:
The current demo in Nav2 parameters. I threw them onto a robot as part of another personal project and it really did not do very well, especially in constrained environments like a hallway with a box in the way where there is plenty of space for the robot to get through. I would not classify that as "constrained", but if it can't do that, it definitely would have issues with doorways or other less trivial spaces. 

I think overall the entire Nav2 parameters for DWB need some love, I don't think anyone remembers where the settings we use came from, but I think it was from some extremely coarse tuning by Intel for the very first turtlebot3 demo.

SteveMacenski, 2022-09-28:
Another good addition are supporting path plan orientations for reversing segments from feasible smac planners.

stevedanomodolor, 2023-01-17:
@SteveMacenski, should the tunning be based on a real robot?, or is simulation robot okay?, The problem is that at the moment I don't have a physical mobile robot with me.

SteveMacenski, 2023-01-18:
Ah yes, that would be a problem. Maybe this isn't a great project then!

SteveMacenski, 2023-08-30:
I'm largely abandoning this in exchange for the new MPPI, which after some baking time I hope to be the new default controller in Nav2. As such, this is deprioritized to the point of not being actionable for core maintainers to spend days on. 

I'm more than happy to review / merge PRs from users that have tuned DWB better for their applications that we can take advantage of as our defaults, but for now, its not something maintainers' time will be spent on considering newer alternatives.

closed 2023-08-30 by SteveMacenski