ros-navigation/navigation2 #2052: nav2_waypoint_follower build fails on main branch.

garyrightbot, 2020-10-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy source 
- Version or commit hash:
  - e1e1be4880a703c9fba44bb0ed68f4463d31eb95
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

``` 
gary@gary-desktop:~/navigation2_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Starting >>> nav2_gazebo_spawner
Finished <<< nav2_common [1.56s]                                   
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid                                          
Finished <<< nav2_voxel_grid [0.56s]                               
Finished <<< nav2_gazebo_spawner [2.22s]                           
Finished <<< nav_2d_msgs [2.36s]                                   
Starting >>> dwb_msgs                              
Finished <<< dwb_msgs [2.11s]                                       
Finished <<< nav2_msgs [3.40s]                   
Starting >>> nav2_util
Finished <<< nav2_util [0.66s]                      
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server                                
Starting >>> nav2_amcl
Finished <<< nav2_lifecycle_manager [1.26s]                            
Starting >>> nav2_rviz_plugins
Finished <<< nav_2d_utils [1.42s]                                   
Finished <<< nav2_amcl [1.55s]                                              
Finished <<< nav2_map_server [1.74s]                                         
Starting >>> nav2_costmap_2d
Finished <<< nav2_rviz_plugins [0.87s]                                        
Finished <<< nav2_behavior_tree [2.57s]                                       
Starting >>> nav2_bt_navigator
Finished <<< nav2_costmap_2d [1.16s]                                       
Starting >>> nav2_core                                       
Starting >>> costmap_queue
Finished <<< nav2_bt_navigator [0.55s]                                    
Finished <<< nav2_core [0.39s]                                  
Starting >>> dwb_core
Starting >>> nav2_controller                                            
Starting >>> nav2_navfn_planner
Starting >>> nav2_planner
Starting >>> nav2_recoveries                                             
Starting >>> nav2_waypoint_follower
Starting >>> smac_planner
Finished <<< costmap_queue [1.82s]                                       
Finished <<< nav2_navfn_planner [1.97s]                              
Finished <<< nav2_planner [5.04s]                                    
Finished <<< dwb_core [5.80s]                                     
Starting >>> dwb_critics
Starting >>> dwb_plugins                                                   
--- stderr: nav2_waypoint_follower                                          
/home/gary/navigation2_ws/src/navigation2/nav2_waypoint_follower/src/waypoint_follower.cpp: In member function ‘void nav2_waypoint_follower::WaypointFollower::followWaypoints()’:
/home/gary/navigation2_ws/src/navigation2/nav2_waypoint_follower/src/waypoint_follower.cpp:195:87: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<nav2_msgs::action::NavigateToPose>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> > >)>’} and ‘std::_Bind_helper<false, void (nav2_waypoint_follower::WaypointFollower::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&), nav2_waypoint_follower::WaypointFollower*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (nav2_waypoint_follower::WaypointFollower::*(nav2_waypoint_follower::WaypointFollower*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::NavigateToPose> >&)>’})
  195 |         std::bind(&WaypointFollower::goalResponseCallback, this, std::placeholders::_1);
      |                                                                                       ^
In file included from /usr/include/c++/9/future:48,
                 from /home/gary/ros2_foxy/install/rclcpp/include/rclcpp/executors.hpp:18,
[... 38 lines cut ...]
  195 |         std::bind(&WaypointFollower::goalResponseCallback, this, std::placeholders::_1);
      |                                                                                       ^
make[2]: *** [CMakeFiles/waypoint_follower_core.dir/build.make:63: CMakeFiles/waypoint_follower_core.dir/src/waypoint_follower.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:109: CMakeFiles/waypoint_follower_core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_waypoint_follower [18.2s, exited with code 2]
Aborted  <<< dwb_plugins [42.1s]                                                                                           
Aborted  <<< dwb_critics [46.0s]                                                                                               
Aborted  <<< nav2_recoveries [54.6s]                                                                                            
Aborted  <<< smac_planner [59.3s]                                                                                        
Aborted  <<< nav2_controller [1min 2s]                                           

Summary: 20 packages finished [1min 11s]
  1 package failed: nav2_waypoint_follower
  5 packages aborted: dwb_critics dwb_plugins nav2_controller nav2_recoveries smac_planner
  1 package had stderr output: nav2_waypoint_follower
  4 packages not processed ``` 

```

#### Expected behavior
The nav2 stack should build successfully
#### Actual behavior
nav2_waypoint_follower build fails

SteveMacenski, 2020-10-21:
Main builds on ros2 master. If you use any other distribution (eg foxy) you have no promise of compatibility. See our build guides on navigation.ros.org.

closed 2020-10-21 by SteveMacenski

garyrightbot, 2020-10-22:
Oh sorry my bad.

garyrightbot, 2020-10-22:
@SteveMacenski I was able to merge most of the changes from the main branch on the foxy-devel branch.

Could not get the waypoint_task_executor_plugin to work though.

My changes are [here](https://github.com/garyrightbot/navigation2/tree/foxy-devel) in case someone wants to use the main branch features (like SmacPlanner) on foxy. I hope my repo could come in handy for the next foxy release too.