ros-navigation/navigation2 #2058: How can I subscribe LaserScan topic at nav2_behavior_tree custom action ?

minwoominwoominwoo7, 2020-10-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.
- ROS2 Version:
  - foxy
- Version or commit hash:
  - minwoo@minwoo-NUC8i7BEH:~/turtlebot3_ws/src/navigation2$ git -C navigation2 rev-parse HEAD
fa1a64ad4d47a3793d22ce7da3f7238c92294af7
- DDS implementation:
  - Fast-RTPS, I used default dds 

#### Steps to reproduce issue
hello, 
I make nav2_behavior_tree custom action. 
I can subscribe some topic, It's ok. but I can not subscribe  <sensor_msgs::msg::LaserScan> type topic, 
If I try subscribe <sensor_msgs::msg::LaserScan> topic. I can meet Execution error. 

erro log is below 
[recoveries_server-9] [INFO] [1603724537.526578450] [recoveries_server]: Configuring wait
[planner_server-8] [INFO] [1603724537.530311066] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
[lifecycle_manager-12] [INFO] [1603724537.532356586] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-10] [INFO] [1603724537.532537433] [bt_navigator]: Configuring
[bt_navigator-10] [ERROR] [1603724537.552957661] []: Caught exception in callback for transition 10
[bt_navigator-10] [ERROR] [1603724537.552980368] []: Original error: Could not load library: /home/minwoo/turtlebot3_ws/install/nav2_behavior_tree/lib/libnav2_ball_search_action_bt_node.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_message_type_support_handleIN11sensor_msgs3msg10LaserScan_ISaIvEEEEEPK29rosidl_message_type_support_tv
[bt_navigator-10] [WARN] [1603724537.553003554] []: Error occurred while doing error handling.
[bt_navigator-10] [FATAL] [1603724537.553012813] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-12] [ERROR] [1603724537.553416742] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-12] [ERROR] [1603724537.553440277] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[rviz2-3] [INFO] [1603724537.840370000] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1603724537.840596552] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1603724537.873144454] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [WARN] [1603724538.033498180] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.

Is there a limit on the size or type of topics that can be subscribed to in the nav2_behavior_tree action node?
please help me. 

I attach my code, Thank you ~

////////ball_search_action.cpp ///////////

#include "nav2_behavior_tree/bt_action_node.hpp"
#include "nav2_msgs/action/wait.hpp"
#include "nav2_behavior_tree/plugins/action/ball_search_action.hpp"

namespace nav2_behavior_tree
{

  BallSearch::BallSearch(
    const std::string & xml_tag_name,
    const std::string & action_name,
    const BT::NodeConfiguration & conf)
  : BtActionNode<nav2_msgs::action::Wait>(xml_tag_name, action_name, conf)
  {
    ball_sub_ = node_->create_subscription<gb_visual_detection_3d_msgs::msg::BoundingBoxes3d>(
      "/darknet_ros_3d/bounding_boxes", 10, std::bind(&BallSearch::callback_updated_ball, this, _1));

    //When the below code is added, the execution erro happen. I can't understand.
    scan_sub_ = node_->create_subscription<sensor_msgs::msg::LaserScan>(
      "scan", rclcpp::SensorDataQoS(), std::bind(&BallSearch::callback_scan, this, _1));
      
  } 
  
  void
  BallSearch::callback_scan(const sensor_msgs::msg::LaserScan::SharedPtr msg)
  {
    //last_scan_received_ = *msg;
    RCLCPP_INFO(node_->get_logger(), std::to_string(msg->ranges[90]).c_str());
  }      
  
  '''''''''''''''''''''''''''''''''''''''''''''''''''''''
  
  
  
 ////////ball_search_action.hpp///////////
  
 // Copyright (c) 2018 Samsung Research America
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#ifndef NAV2_BEHAVIOR_TREE__PLUGINS__ACTION__BALL_SEARCH_ACTION_HPP_
#define NAV2_BEHAVIOR_TREE__PLUGINS__ACTION__BALL_SEARCH_ACTION_HPP_

#include <string>

#include "nav2_behavior_tree/bt_action_node.hpp"
#include "nav2_msgs/action/wait.hpp"
#include "nav_msgs/msg/odometry.hpp"
#include "gb_visual_detection_3d_msgs/msg/bounding_boxes3d.hpp"
#include "geometry_msgs/msg/pose_stamped.hpp"
#include "nav2_util/geometry_utils.hpp"
#include "sensor_msgs/msg/laser_scan.hpp"

namespace nav2_behavior_tree
{

using std::placeholders::_1;

class BallSearch : public BtActionNode<nav2_msgs::action::Wait>
{
public:
  BallSearch(
    const std::string & xml_tag_name,
    const std::string & action_name,
    const BT::NodeConfiguration & conf);

  void on_tick() override;

  // Any BT node that accepts parameters must provide a requiredNodeParameters method
  static BT::PortsList providedPorts()
  {
    return providedBasicPorts(
      {
        BT::InputPort<int>("wait_duration", 1, "Wait time"),
        BT::OutputPort<geometry_msgs::msg::PoseStamped>(
          "output_goal",
          "Received Ball Goal by subscription"),        
      });
  }

private:
  BT::NodeStatus tick() override ;
    
  void callback_updated_ball(const gb_visual_detection_3d_msgs::msg::BoundingBoxes3d::SharedPtr msg);
  rclcpp::Subscription<gb_visual_detection_3d_msgs::msg::BoundingBoxes3d>::SharedPtr ball_sub_;
  gb_visual_detection_3d_msgs::msg::BoundingBoxes3d last_ball_received_; 
  int recieve_ball_count ; 

  void callback_scan(const sensor_msgs::msg::LaserScan::SharedPtr msg);
  rclcpp::Subscription<sensor_msgs::msg::LaserScan>::SharedPtr scan_sub_;
  sensor_msgs::msg::LaserScan last_scan_received_;  

};  

}  // namespace nav2_behavior_tree

#endif  // NAV2_BEHAVIOR_TREE__PLUGINS__ACTION__BALL_SEARCH_ACTION_HPP_

SteveMacenski, 2020-10-27:
> [bt_navigator-10] [ERROR] [1603724537.552980368] []: Original error: Could not load library: /home/minwoo/turtlebot3_ws/install/nav2_behavior_tree/lib/libnav2_ball_search_action_bt_node.so: [...]

Did you actually link things correctly? I also don't see any `BT_REGISTER_NODES`. Please review the basics of BehaviorTree.CPP

SteveMacenski, 2020-11-04:
Closing - no response from user

closed 2020-11-04 by SteveMacenski