ros-navigation/navigation2 #2060: Different behavior between ROS2 DWB local planner and ROS1 DWA local planner in simulation environment

Phymin, 2020-10-28:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu18.04
- ROS2 Version:
  -  ros1-melodic ros2-dashing

- Version or commit hash:
  -  942ea483863e3268023cb0cd504c93632fcbf155

- DDS implementation:
  - Fast-DDS

#### Steps to reproduce issue

```
1. first launch a gazebo simulation environment
2. then launch the navigation2 with rviz2, I use the launch file from nav2_bringup package
3. give goals from rviz2 
```

#### Expected behavior
With all the parameters almost the same between ROS1 and ROS2, the expect hehavior in ROS2 should almost the same as in ROS1

#### Actual behavior
In ROS2, when the vehicle reaches near the goal, the vehicle will frequently take recovery actions(such as spin), and sometimes may even report navigation failed  in simulation environment, but in ROS1, there will be no recovery actions in simulation environment

#### Additional information
1. From my test, the parameter infected with this problem is the goal tolerance parameter, if I increase the goal tolerance, the possibility of recovery will happen less.
2. I have also test this problem in foxy, it's almost the same with dashing.
3. Here is my navigation parameter file: [https://gist.github.com/Phymin/d5e4f3bc7f67177fb08e30a786fa118d](url)

#### PS
I don't know if I should put this problem here or on the ROS anwser, but I have asked serveral questions about ROS2 on ROS anwser, but most of them  have no response, so I think I should put it here first. Thank you.

SteveMacenski, 2020-10-28:
DWA is not DWB. Please read the docs or reference David Lu’s ROSCon talk.

closed 2020-10-28 by SteveMacenski

SteveMacenski, 2020-10-28:
These are different planners (hence the name change). There is no expectation that your old configuration files will have the same behavior, in fact the vast majority of your old parameters are no longer even used.

Phymin, 2020-10-29:
> These are different planners (hence the name change). There is no expectation that your old configuration files will have the same behavior for a completely different controller, in fact the vast majority of your old parameters are no longer even used.

@SteveMacenski Yes, I know, what I mean the same is that the core parameters are the same. and the parameter file was copied from the turtlebot3 package, which is the standard tutorial package, if this parameter are not suitable, where should I refer to tuning the navigation parameters? Thanks you!