ros-navigation/navigation2 #2061: [eloquent-devel/tools/ros2_dependencies.repos] Incorrect BehaviorTree.CPP Branch

mjshiggins, 2020-10-28:
## Bug report

#### Expected behavior
The [ros2_dependencies.repos file](https://github.com/ros-planning/navigation2/blob/eloquent-devel/tools/ros2_dependencies.repos) should reference the appropriate branch / tag for the BehaviorTree.CPP library (3.5.3) due to internal updates to match the new BT API.

#### Actual behavior
The `ros2_dependencies.repos` file instead refers to the `ros2-version-2.x` branch, which is not API compatible with the current `eloquent-devel` branch or tag `0.3.5`, installed when using the apt-package.

#### Additional information
This is particularly important when dealing with the underlying issues of the behaviortree_cpp_v3 binary release. Also, it looks like this was addressed in the discussion on issue #2013 , but those erroneous changes still made it in.

Merge request incoming, will update this issue once I've posted it.

mjshiggins, 2020-10-28:
Added PR #2062

SteveMacenski, 2020-10-28:
Thanks, well discuss in the PR

closed 2020-10-28 by SteveMacenski