ros-navigation/navigation2 #2063: [bt_navigator] undefined symbol at runtime

KvE-sax, 2020-10-28:
## Bug report

**Required Info:**
    Operating System:
        Ubuntu 20.04 LTS
    ROS2 Version:
        Foxy binaries
- Version or commit hash:
  - 26b975b78af4ac0b839a8b825f621a2a2e611ee1
- DDS implementation:
  - Fast RTPS

#### Steps to reproduce issue
- created new behavior tree condition module (based on existing condition modules and their structure)
- register to cmake file of the package, behavior tree xml, bt_navigator.cpp and nav2_params.xml
- Used tb3_simulation_launch.py and changed the behavior tree xml file to the custom xml.

#### Expected behavior
Configuring and activating the bt_navigator.

#### Actual behavior

> [INFO] [bt_navigator-11]: process started with pid [439124]
> [bt_navigator-11] [INFO] [1603903924.887445223] [bt_navigator]: 
> [bt_navigator-11] 	bt_navigator lifecycle node launched. 
> [bt_navigator-11] 	Waiting on external lifecycle transitions to activate
> [bt_navigator-11] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
> [bt_navigator-11] [INFO] [1603903924.889191370] [bt_navigator]: Creating
> [lifecycle_manager-13] [INFO] [1603903925.253252845] [lifecycle_manager_navigation]: [34m[1mConfiguring bt_navigator[0m[0m
> [bt_navigator-11] [INFO] [1603903925.253488836] [bt_navigator]: Configuring
> [bt_navigator-11] [ERROR] [1603903925.290136583] []: Caught exception in callback for transition 10
> [bt_navigator-11] [ERROR] [1603903925.290155206] []: Original error: Could not load library: /home/nena/foxy_tb3_ws/install/nav2_behavior_tree/lib/libnav2_crate_detection_condition_bt_node.so: undefined symbol: _ZN3tf212getTimestampIN13geometry_msgs3msg12PoseStamped_ISaIvEEEEENSt6chrono10time_pointINS6_3_V212system_clockENS6_8durationIlSt5ratioILl1ELl1000000000EEEEEERKT_
> [bt_navigator-11] [WARN] [1603903925.290188448] []: Error occurred while doing error handling.
> [bt_navigator-11] [FATAL] [1603903925.290196984] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
> [lifecycle_manager-13] [ERROR] [1603903925.290646261] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator

#### Additional information
The thing i want to do here is calculate a pose of some crate and publish it to the tf_tree to make it avalible in the rest of the nodes.

If the code of the custom node is required I can post a link to the fork I made of the mentioned version/commit.

SteveMacenski, 2020-10-28:
We can't support your custom BT nodes here unless they're being contributed back. If you're having issues, see the BT.CPP library documentation and the other BT nodes as examples.

SteveMacenski, 2020-11-04:
Closing - no response from user

closed 2020-11-04 by SteveMacenski