ros-navigation/navigation2 #2064: Consider adding shim rotate-to-path controller plugin

SteveMacenski, 2020-10-28:
Per https://github.com/ros-planning/navigation2/issues/2035 this has come up in now 5 different places I've worked / hobby projects / other users.

It would be useful to have a local trajectory planner plugin that consists of 2 steps:
- Checking the angle between the robot's current orientation and the path, when using a holonomic planner on a non-holonomic robot. If larger than threshold, rotate to path orientation
- Once below threshold, continue with the loaded local trajectory plugin (like DWB or TEB)

SteveMacenski, 2020-11-24:
This has now **again** come up in another context. This clearly deserves action. Maybe another option is to include a rotate to orientation controller plugin itself for use. That way it could be called independently from a BT node (like rotate to path orientation then give to controller; or rotate to goal after XY tol hit) or used in conjunction with other controller plugins. 

Another thought is to have it have both rotate to path orientation and rotate to goal at end as the shim, and the middle of it when not in XY tolerance or at start of path will be given to the "real" plugin shimmed (DWB, etc). Then controller plugins wouldn't need to handle the rotate to goal itself - but if it were to - then the second-level shim would just return true and do nothing.

shrijitsingh99, 2020-12-07:
I too have implemented this functionality in projects in the past, so I support this.
Having it as a BT node seems like a good way to go. Any potential downsides?

> Another thought is to have it have both rotate to path orientation and rotate to goal at end as the shim, and the middle of it when not in XY tolerance or at start of path will be given to the [...]

This is not quite clear to me. Could you expand on this?

SteveMacenski, 2020-12-07:
I don't think a BT node is the objectively right option for everything. I think the best option here is to make a class that does a rotate to relative heading task. Then we  can use that class to implement a `rotate to heading` BT node, `rotate to node` controller plugin, add to pure pursuit planner or optionally to DWB, and the shim node (any or all).

Sure, what I meant is that typically we have 3 stages in this situation:
- Rotate to path heading, if you have a holonomic planner like NavFn, it can create a path that's not in the orientation of the front of the robot. So first step is rotating to path heading (roughly)
- The normal local trajectory planning
- Then rotating to goal, once you've reached your cartesian tolerance for goal. Then rotate to match the right heading. 

So we could have a controller plugin that does the following:
- Rotates to path heading
- Calls an internal pluginlib plugin (like DWB or pure pursuit, anything, through a plugin interface)
- Rotates to goal.

That way the "shim plugin" has a plugin internal to it that it uses for actual trajectory planning and surrounds it with the rotating necessary. That's one of the most common things I've done in the past because alot of the local trajectory planners really suck when they're not starting at the right orientation. TEB will swing around sharply, DWB will wander. So if we can rotate to the rough path heading before using those algorithms, then that behavior goes away and we only use them for the thing they're good at: following a path.

shrijitsingh99, 2020-12-09:
This seems like a good way to go. Will definitely solve the issues that I have encountered in the past.
It gives flexibility to the developer to implement rotation functionality using BT or a controller plugin, which is a great plus.

SteveMacenski, 2020-12-09:
And its the last time I need to implement this ;-) 

... its been many times.

SteveMacenski, 2021-12-01:
https://github.com/ros-planning/navigation2/pull/2718 there's a PR open with this feature! @rayman @nikolasE @simutisernestas

Rayman, 2021-12-01:
Good work!

closed 2021-12-02 by SteveMacenski