ros-navigation/navigation2 #2069: `nav2_util/test/test_actions.cpp` uses obsolete `fibonacci` action

zultron, 2020-11-04:
The test in https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_util/test/test_actions.cpp#L22 uses the `fibonacci` action from the `test_msgs` package, which is obsolete in ROS2.

The `fibonacci` action was removed from `test_msgs` in PR ros2/rcl_interfaces#77, tagged in `0.7.2`; the current `ros-foxy-test-msgs` package version is `1.0.0-1focal.20201007.204652`.

This might not have been caught yet because of a strange packaging issue, where the `fibonacci` action is still included in the `1.0.0` package; see ros2/rcl_interfaces#114.

At any rate, the `fibonacci` action has been removed in dependency packages, and should be removed here, too.

## Bug report

- Operating System:
  - Ubuntu Focal
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - `foxy-devel` branch as of tag `2.1.0` and commit 0caf3bc9
- DDS implementation:
  - N/A

#### Steps to reproduce issue
No steps yet; once ros2/rcl_interfaces#114 is resolved, a `colcon build` of this repo against Foxy `test_msgs` packages will fail.

SteveMacenski, 2020-11-04:
What do you suggest to replace it?

SteveMacenski, 2020-11-04:
Msg still exists, just moved. Will update when it causes a problem.

closed 2020-11-04 by SteveMacenski