ros-navigation/navigation2 #2073: Interface for Complete Coverage Path Planning

5730289021-NN, 2020-11-05:
## Feature request
Interface for Complete Coverage Path Planning

#### Feature description
According to [createPlan() virtual method](https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2planner_plugin.html#creating-a-new-planner-plugin), the input parameter are start pose and goal pose.

I've read from [concept](https://navigation.ros.org/concepts/index.html#planners), and from my understanding, the generated path should be completed or at least a part of the complete path.

So, the problem is the createPlan() does not properly accept the complete coverage parameter. Personally, I need percent_overlapping parameter which will configure how robot move to the same lane.

#### Implementation considerations
It could be more generic like "map<string>* objective" instead of start pose and goal pose.

SteveMacenski, 2020-11-05:
> Personally, I need percent_overlapping parameter which will configure how the robot move to the same lane.

Why can't you accomplish that using a parameter loaded in your plugin? There's no strict need for that to be a part of the interface definition.

> The input [I assume you mean in API argument] could be more generic like "map< string >* objective" instead of start pose and goal pose.

How would that work, exactly? If the Planner Server loads a set of plugins to use, that argument would still never be populated because its a specific value for your algorithm that the server wouldn't be able to be modified to add. You should create a ROS parameter in your plugin that loads those specific param values required, with no change to the API.

5730289021-NN, 2020-11-06:
I understand that we can load the parameter into the plugin. 

However, in complete coverage(CC) application , what do you suggest to do with “goal” argument? Should it be the endpoint after performing the CC task or just leave it be?

SteveMacenski, 2020-11-06:
Depends on your algorithm and application. I won’t comment on that

5730289021-NN, 2020-11-07:
Alright, I was just too hesitant due to this API. Thank you for your time.

closed 2020-11-07 by 5730289021-NN