ros-navigation/navigation2 #2077: Map server tests are failing with 'double free or corrupted memory'

SteveMacenski, 2020-11-05:


AlexeyMerzlyakov, 2020-11-06:
There are 2 tests failed in map server: `test_map_saver_node.cpp` and `test_map_server_node.cpp` with the same problem. 
Problem appears only on debug builds, reproducible locally and second test (test_map_server_node) is flaky. It is suggestive of multi-thread nature of problem.

Double free appears in `node_` static object when it is being deallocated after `main()` function will exit. The problem is being triggered by `node_->create_client()` call for this node before deallocation. Another thread try to deallocate `node_` before it will be destructed itself, maybe because of `create_client()` makes some instances in ROS2 which points to the same `node_` (unfortunately I do not know well ROS2 internals, so it may be wrong).

However, if destroy manually the `node_` (and related `lifecycle_client_` which also has reference to `node_` inside) before it will be deallocated automatically in `__GI_exit()`, both tests became to pass. Proposed snap for that:
```
diff --git a/nav2_map_server/test/component/test_map_saver_node.cpp b/nav2_map_server/test/component/test_map_saver_node.cpp
index 864ca4c3..1f57fa20 100644
--- a/nav2_map_server/test/component/test_map_saver_node.cpp
+++ b/nav2_map_server/test/component/test_map_saver_node.cpp
@@ -59,6 +59,8 @@ public:
   {
     lifecycle_client_->change_state(Transition::TRANSITION_DEACTIVATE);
     lifecycle_client_->change_state(Transition::TRANSITION_CLEANUP);
+    lifecycle_client_.reset();
+    node_.reset();
   }
 
   template<class T>
diff --git a/nav2_map_server/test/component/test_map_server_node.cpp b/nav2_map_server/test/component/test_map_server_node.cpp
index 6fa14d9b..e381afec 100644
--- a/nav2_map_server/test/component/test_map_server_node.cpp
+++ b/nav2_map_server/test/component/test_map_server_node.cpp
@@ -59,6 +59,8 @@ public:
     lifecycle_client_->change_state(Transition::TRANSITION_DEACTIVATE);
     lifecycle_client_->change_state(Transition::TRANSITION_CLEANUP);
     lifecycle_client_->change_state(Transition::TRANSITION_UNCONFIGURED_SHUTDOWN);
+    lifecycle_client_.reset();
+    node_.reset();
   }
 
   template<class T>
```

SteveMacenski, 2020-11-09:
@AlexeyMerzlyakov the map tests are still failing

AlexeyMerzlyakov, 2020-11-10:
The problem confirmed. This does not related to PR #2078. 
This another another one issue which is related to incorrect usage of `transient_local` messages re-publishing in the testcase. Testcase fix required and already prepared.
Possibly, also working with shared resources should be added to Map Saver. The investigation on this is in progress.

AlexeyMerzlyakov, 2020-11-10:
Update: when using Map Saver saving messages published in `volatile` mode (this is non-standard situation, but discovered during test failure analysis), we should protect its resources between concurrently running threads: main thread and lambda-callback `mapCallback()`.

So, fix consists from 3 parts:
1. Protect Map Saver's resource by mutex
2. Remove unnecessary map republisher for `transient local` messages in `test_map_saver_node`
3. Increase map saving timeout from 1sec -> to 5sec. Sometimes it also causes rare test fails
**plus**
4. This is not related to current issue, but added forgotten reset of save_map_service_ in `MapSaver::on_cleanup()` routine.

Made PR #2084 on this.

closed 2020-11-10 by SteveMacenski