ros-navigation/navigation2 #2079: rclcpp::Duration::from_nanoseconds() not available in foxy

soldierofhell, 2020-11-09:
Hi,
In this recent commit (https://github.com/ros-planning/navigation2/pull/2076) in 'main' branch @SteveMacenski introduced function rclcpp::Duration::from_nanoseconds() which is not available on foxy. I know that the old approach is probably depreciated but this prevent from building main on ros2 foxy and probably I have to build ros2 master from source. Can we somehow hold back from "depreciating" foxy on main. I want to have the latest main of navigation2 because it has the newest features/fixes, on the other hand I don't want to build everything from source and struggling with potential unstability of master branches where I don't have to :) Regards,

soldierofhell, 2020-11-09:
Maybe we could use some macro definitions and build parameter to be able to build on foxy or on ros2 master?

soldierofhell, 2020-11-09:
I've tried to build navigation2 main on ros2 master and it failed because of bondcpp, which doesn't use new rclcpp:Duration interface yet
```
/home/username/ws/install/bondcpp/include/bondcpp/bond.hpp:135:76: error: ‘rclcpp::Duration::Duration(rcl_duration_value_t)’ is deprecated: Use Duration::from_nanoseconds instead or std::chrono_literals. For example:rclcpp::Duration::from_nanoseconds(int64_variable);rclcpp::Duration(0ns); [-Werror=deprecated-declarations]
```

SteveMacenski, 2020-11-09:
Main branch of nav2 covers ros2 master, not foxy. There is no expectation that non-foxy branch will work on foxy as our documentation and process lay out

closed 2020-11-09 by SteveMacenski