ros-navigation/navigation2 #2082: [Input requested] Topological operation costmap layers

SteveMacenski, 2020-11-10:
Hi,

I figured I'd ask and see if there was interest in something like this. In order to help remove salt and pepper noise from images, typically you might use dilation and erosion. The same concept could apply to navigation2 costmap layers, where an erroneous voxel layer or obstacle layer measurement (or static layer with SLAM session and things moving around) would make in the middle of free space incorrectly. 

The idea would be to then offer some convolution or topological costmap layer to remove this noise. Is this something folks think is useful and would use?

ardabbour, 2020-11-25:
This is an actual use case in a real life application I developed, especially for adjusting traversability calculations for different robots with different needs. Is this in the same light as the grip_map package's filters? Because eventually that is what we moved to.

SteveMacenski, 2020-11-25:
I don't see this particular filter in the filter's package, but its in the same theme. Topological operations are typically used to remove salt and pepper noise from images so it could be good here to remove random single extraneous measurements.  

It's also our intent to move to grid_maps, we'd certainly be happy to have you help with that :-)

ardabbour, 2020-11-26:
We _should_ be able to use the [MathExpressionFilter](https://github.com/ANYbotics/grid_map/blob/a104f41e4ada75a6322e633693987540f7394902/grid_map_demos/config/filters_demo_filter_chain.yaml#L67) to that extent, I think. Maybe a similar concept for costmap2D would be useful (or not, if nav2 is porting to grid_map 🤷). I remember there was such a porting task (I'll try to help if I can😬), but I couldn't find the corresponding issue to link here.

SteveMacenski, 2020-11-30:
Its been ported to ROS2 but we haven't started the design effort in replacing costmap with it yet. That's under the environmental modelling ticket.

fmrico, 2020-12-02:
Hi @SteveMacenski , I would like to focus my efforts these weeks on helping to port nav2 to grid_map. I am starting now to do some tests with the grid_map library.

SteveMacenski, 2020-12-02:
It might be good to start gathering thoughts into a design doc

jediofgever, 2020-12-15:
where to follow up the ticket/discussion related to replacing `costmap_2d` with `grid_map` ?, I am interested in this with a potential to take a part in the effort of port.

SteveMacenski, 2020-12-15:
#1278 and mostly on slack in various groups.

AlexeyMerzlyakov, 2022-12-22:
Just for the case: attaching DenoiseLayer HLD and brief scheme of main algorithms, that I've prepared during the code analysis of the #2567. That was useful for me, hope this might help someone in the future.
[DenoiseLayer_CodeStructure.pdf](https://github.com/ros-planning/navigation2/files/10287971/DenoiseLayer_CodeStructure.pdf)

ryzhikovas, 2022-12-25:
@AlexeyMerzlyakov, you did a great job creating these schemes. Thank you so much for your help. I'm sorry I didn't think of doing it myself, it would have made the review easier.

closed 2023-06-16 by SteveMacenski