ros-navigation/navigation2 #2085: Hybrid-A* / Smac planner constantly switching between two alternatives routes

dkuenster, 2020-11-11:
## Bug report

**Required Info:**

- Operating System:
  - Debian 10
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - 13e01d2d6bcd230bd5eb069d109eee2bf2c39fe5
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Start a simulation on an open map with the following smac parameters:
```
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    GridBased:
      plugin: "smac_planner/SmacPlanner"
      tolerance: 5.0                    # tolerance for planning if unable to reach exact pose, in meters
      downsample_costmap: True          # whether or not to downsample the map
      downsampling_factor: 4            # multiplier for the resolution of the costmap layer (e.
#### Additional information

<!-- If you are reporting a bug delete everything below
     If you are requesting a feature deleted everything above this line -->
----
## Feature request

#### Feature description
<!-- Description in a few sentences what the feature consists of and what problem it will solve -->

#### Implementation considerations
<!-- Relevant information on how the feature could be implemented and pros and cons of the different solutions -->g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: false              # allow traveling in unknown space
      max_iterations: 1000000           # maximum total iterations to search for before failing
      max_on_approach_iterations: 1000  # maximum number of iterations to attempt to reach goal once in tolerance
      max_planning_time_ms: 2000.0      # max time in ms for planner to plan and smooth. Will scale maximum smoothing times based on remaining time after planning.
      smooth_path: false                # Whether to smooth searched path
      motion_model_for_search: "REEDS_SHEPP"  # 2D Moore, Von Neumann; SE2 Dubin, Redds-Shepp
      angle_quantization_bins: 72       # For SE2 node: Number of angle bins for search, must be 1 for 2D node (no angle search)
      minimum_turning_radius: 10.0      # For SE2 node & smoother: minimum turning radius in m of path / vehicle
      reverse_penalty: 0.0              # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
      change_penalty: 1000.0            # For SE2 node: penalty to apply if motion is changing directions, must be >= 0
      non_straight_penalty: 1.10        # For SE2 node: penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 5.3                 # For SE2 node: penalty to apply to higher cost zones, must be >= 0

      smoother:
        smoother:
          w_curve: 30.0                 # weight to minimize curvature of path
          w_dist: 0.0                   # weight to bind path to original as optional replacement for cost weight
          w_smooth: 30000.0             # weight to maximize smoothness of path
          w_cost: 0.025                 # weight to steer robot away from collision and cost
          cost_scaling_factor: 0.05     # this should match the inflation layer's parameter

        optimizer:
          max_time: 0.10                # maximum compute time for smoother
          max_iterations: 500           # max iterations of smoother
          debug_optimizer: false        # print debug info
          gradient_tol: 1.0e-10         # gradient change termination tolerance
          fn_tol: 1.0e-20               # function change termination tolerance
          param_tol: 1.0e-15            # parameter block change termination tolerance
          advanced:                     # I do not recommend users mess with this unless they're doing production tuning
            min_line_search_step_size: 1.0e-20
            max_num_line_search_step_size_iterations: 50
            line_search_sufficient_function_decrease: 1.0e-20
            max_num_line_search_direction_restarts: 10
            max_line_search_step_expansion: 50
```

Set a goal pose close to the vehicle with a 180° difference in orientation or anything that requires a more complex manouver involving backwards driving.

#### Expected behavior
Robot reaches goal with no problems.

#### Actual behavior
The global trajectory often switches extremely fast between two alternatives. So fast that the controller has trouble making any progress. See:
[image]
[image]

Do you have any idea how to prevent this kind of behavior?

SteveMacenski, 2020-11-11:
This is a problem of any and all path planners when you have a situation with evenly distributed, or no costs, in the potential fields. You should not be planning at such a high rate, it serves no purpose. 1 hz is plenty enough since a local trajectory planner will handle newly introduced obstacles for minor path deviation.

SteveMacenski, 2020-11-11:
https://github.com/ros-planning/navigation2/issues/1570

dkuenster, 2020-11-13:
> Edit: oh, I see that IS your global costmap, read the smac readme and potential fields. This is a very poor configuration of costmap_2d. To be clear, you’d see this exact same issue with any search or sampling based planner with your configuration, this isn’t unique to this particular planner or hardware.

Could you elaborate? I read the stuff you mentioned before configuring the costmap and thought I had followed the advice given there by creating a smooth potential field that covers most of the map.

SteveMacenski, 2020-11-13:
I don't see a potential field. Costmap in your rviz config uses the 'costmap' color scheme which is blue -> red, the blue stops after inflating. I'm not 100% sure what the grey gradient is, but I don't think that's from the costmap.

dkuenster, 2020-11-16:
No it's the "map" color scheme for the global costmap. So it's the different shades of grey.
Here is a screenshot with the "costmap" colorscheme.
[image]

SteveMacenski, 2020-11-16:
Ah, its odd that your walls were blue - my mistake. This looks pretty good. My guess is that it has so many potential options of paths that as the robot moves even a _little bit_ that the lowest cost is now different. But once you move something more substantial, then you've locked yourself into one branch of solutions so then the replans should be more helpful.

SteveMacenski, 2020-11-20:
Updates / did we fix your issue?

dkuenster, 2020-11-30:
Yes, your suggestions helped to reduce it to a tolerable level, thank you.

closed 2020-11-30 by SteveMacenski