ros-navigation/navigation2 #2086: Switching dynamically between planners and controllers

pabloinigoblasco, 2020-11-11:
I need to switch in runtime between different planners and controllers. I used to do that in ROS1 navigation stack using the dynamic_reconfigure server.
Is this functionality available in navigation2. How could I do that?

I see that default behaviors trees use fixed values for the planner_id (GridBased) and controller_id (FollowPath) inputs, so It looks to me that a dynamic solution for this is not provided yet.

I have a few possible solutions in mind but I am not sure if they are possible and which one would be the most convenient.
 - Creating a custom behavior tree for each planner (and select it via nav2_msgs/action/NavigateToPose::Goal::behavior_tree)
 - Overwriting the the FollowPath and GridBased "parameter trees" of the planner_server and controller_server nodes.
 - Creating a custom behavior that receives (via topic/service) the desired plan and writes that into the BT blackboard

Any other proposal?

SteveMacenski, 2020-11-11:
Yes, create a custom behavior tree to accommodate changes to different algorithms depending on the context you want to use each within. For instance, you could create a condition Bt node to check `if in dynamic scene` and if true, uses the `idk-really-awesome-controller`, else uses the `line follower` controller.

pabloinigoblasco, 2020-11-11:
Thanks for the fast answer. 
It makes sense.

SteveMacenski, 2020-11-11:
Can this be closed or is there further feature request / bug report?

pabloinigoblasco, 2020-11-12:
I agree, I guess this is a feature request.
Switching between planners could be done just updating a dynamic parameter in ROS1 (even from the command line), it was clear and simple.
In navigation2 this can be potentially done but you need to code own behavior to achieve it. 

This weekend I will think a bit more to that idea of creating a custom behavior and I may implement it.

Nonetheless, thanks for adding it to the feature list.

SteveMacenski, 2020-11-12:
> In navigation2 this can be potentially done but you need to code own behavior to achieve it. It would be great if that behavior were already provided by default by the navigation2 stack.

Can you lay out a vision for how that could be addressed scalably? I'm not opposed to creating behaviors for changing algorithms outside of the BT, but there has to be some reasonable design behind it. Changing a BT to specify a `controller_id` input port defined by the `FollowPath` BT node is extremely reasonable requiring no code updates beyond the XML (which you could think of as a parameter file; all users should have their own tuned for their own needs) and scalable to `N` algorithms in any definable context or event. 

I'd say that what you're asking can be completed in a really reasonably and easy way, so if you don't think that's the best solution for your needs, I'm all ears to see how we can be more flexible.

pabloinigoblasco, 2020-11-12:
> Cross linking docs to this to remind me this is something non-obvious to people 

I am afraid we are not aligned yet.  For me it is obvious that it can be done, but It is not obvious if it is already implemented or If I still need to implement some new C++ BTNodes.

> Changing a BT to specify a controller_id input port defined by the FollowPath BT node is extremely reasonable requiring no code updates beyond the XML (which you could think of as a parameter file; [...]

I agree, I see the XML as a parameter file. And I understand that it must be configured in the way it is more convenient for each specific application. I also agree that a single XML could be created to adapt to N situations or algorithms. 

My mindset may still be "attached" to the ROS1 navigation stack and there might be something I do not see yet. 
However, If I see the navigation2 stack as a black-box ( with a properly configured BT) there should be some ROS network mechanism (topic, service or parameter) to switch the current planner and controller based on just the name of them (smac_planner, dwb - or the name you specified in the controller_plugins parameter or the planner_plugins parameter).

But still the question is simple (almost a yes/no question): 

is it possible to change the planner and controller from an external input (topic, service or parameter)? Is there a short code snippet (bash, c++, BT-XML) I could use to achieve it?

For example, in ROS1 the answer is yes, and below it is shown an example in C++ (but it also is possible in bash or python):

    dynamic_reconfigure::StrParameter local_planner, global_planner;
    local_planner.name = "base_local_planner";
    local_planner.value = desired_local_planner_;
    conf.strs.push_back(local_planner);

    global_planner.name = "base_global_planner";
    global_planner.value = desired_global_planner_;
    conf.strs.push_back(global_planner);
    ros::service::call("/move_base/set_parameters", srv_req, srv_resp);

Does make this question sense?

> For instance, you could create a condition Bt node to check if in dynamic scene and if true, uses the idk-really-awesome-controller, else uses the line follower controller.

- Is that BTNode that allows me switch between planner and controllers already implemented in the nav stack2? 
- Should I implement a custom BTNode condition C++ class (I show later how I imagine the BT XML)?
- Is there something I am not understanding yet?

For example, I see an implemented BTNode called IsBatteryLowCondition. I understand that I could use that condition to select a planner or other depending on what is the battery level. For me it is important the fact that  the IsBatteryLowCondition BTNode uses a topic to receive information from the external ROS network (/battery_state) and change the result.

> I'm not opposed to creating behaviors for changing algorithms outside of the BT, but there has to be some reasonable design behind it.

I do not claim that the planner and controller must be changed from outside of the BT. In fact, the solution I have in mind uses the BTXML to switch the planner and controller but It would required a new not implemented C++ BTNode class.

What I claim is that there should be a way to switch between planners and controllers from the ROS network (topic, service or parameter) based on the names of the desired planner and controller. 

> Can you lay out a vision for how that could be addressed scalably? 

Assuming that BTNode lets say"PlannerSelector" is not yet implemented it should be implemented:

- That "PlannerSelector" could be a condition node or just a simple BTNode that lasts one single tick and it is called before ComputePathToPose. 
- That BTNode reads from the ROS network (parameter, topic or service) the desired planner and controller and then they are exposed to the BTTree XML so that they can bee passed as inputs in ComputePathToPose/planner_id and FollowPath/controller_id

The resoulting XML I have in mind would be similar to the following:

```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <PipelineSequence name="NavigateWithReplanning">
      <RateController hz="1.0">
        <Sequence>
          <GoalUpdater input_goal="{goal}" output_goal="{updated_goal}">
               <PlannerSelector output_planner="{selected_planner_id}" output_controller="{selected_controller_id}" />
               <ComputePathToPose goal="{updated_goal}" path="{path}" planner_id="{selected_planner_id}"/>
          </GoalUpdater>
         <TruncatePath distance="1.0" input_path="{path}" output_path="{truncated_path}"/>
        </Sequence>
      </RateController>
      <KeepRunningUntilFailure>
         <FollowPath path="{truncated_path}" controller_id="{selected_controller_id}"/>
      </KeepRunningUntilFailure>
    </PipelineSequence>
  </BehaviorTree>
</root>
```

> I'd say that what you're asking can be completed in a really reasonably and easy way and so close this ticket,

If you feel that the question is already answered or you think what I say makes no sense, please feel free to close the issue.

For me it is not obvious yet If there is a snippet/short answer to solve the selection of the planner based on an external input from the ROS network (topic, service or parameter) with the name of the desired planner and controller (as it was shown in the c++ snippet before).

SteveMacenski, 2020-11-13:
>  Is it possible to change the planner and controller from an external input (topic, service or parameter)? Is there a short code snippet (bash, c++, BT-XML) I could use to achieve it?

Yes, you can create a trivial BT node that sets an input port of the {compute path, follow path} nodes to select the mapped algorithm ID in the server (e.g. GridBased, whatever name you chose). You mention that you think there should be some "included" code do to this, but by your description, you have your own non-included code to change with dynamic reconfigure. So this really isn't any different except the BT node solution is scalable and not abusing dynamic reconfigure for what is it not meant for.

> Is that BTNode that allows me switch between planner and controllers already implemented in the nav stack2?

Yes, see nav2 docs. The follow path / compute path to pose BT nodes have input ports allowing you to set the controller / planner ID it requests from the planner/controller servers. By default they're set to GridBased and FollowPath as are default in the servers themselves. You can just change to another value hardcoded in the BT or using the blackboard parameters if you'd like it to change over time in the same run from a service / topic call. 

> Should I implement a custom BTNode condition C++ class (I show later how I imagine the BT XML)?

Yes. There are many solutions here. You could have the condition node which goes down one branch if true and another if false, each with a controller/planner ID that makes the condition. Another would be an action node that just looks for a topic / service / parameter and sets the blackboard parameter for controller/planner ID to the new value to use and have the BT XML input port for the tasks use the blackboard value. Probably another 2-3 ways to do it. 

> PlannerSelector

A planner selector / controller selector BT node that changes the blackboard values and a BT XML that uses those blackboard values in the follow path / compute path would be a valuable addition.  I don't think its the right solution for every user that wants to change algorithms, but it could be useful for enough that I'd support it. The way I would rather have people use multiple planners / controllers would be to have conditions for when each are useful so they change change dynamically themselves through the BT rather than external input. Whatever external node is requesting the change shouldn't exist and should be integrated into the condition node so that change is fully autonomous within the tree.

pabloinigoblasco, 2020-11-13:
>>Is it possible to change the planner and controller from an external input (topic, service or parameter)? Is there a short code snippet (bash, c++, BT-XML) I could use to achieve it?

> Yes, you can create a trivial BT node that sets an input port of the {compute path, follow path} nodes to select the mapped algorithm ID in the server (e.g. GridBased, whatever name you chose). You [...]

I think that is not correct. From my point of view, you cannot compare the code of a "remote procedure call" invokation (that I can even invoke from a bash terminal ) to create a new BTNode C++ class for the navigation architecture (no matter if it is scalable) .

In fact, I think that goes against the ROS philosophy where clients consume remote functionalities.

Do not get me wrong, It is great if you can do complex and scalable applications inside the navigation BT but I shouldnt be forced to do that in that way.

The navigation stack should (from my point of view) really solve well the issues related with navigation. And it should provide some basic functionalities to the ROS Network (at least the same it was exposed in ROS1) like selecting the planner, navigating to a goal.

>>Is that BTNode that allows me switch between planner and controllers already implemented in the nav stack2?

>Yes, see nav2 docs. The follow path / compute path to pose BT nodes have input ports allowing you to set the controller / planner ID it requests from the planner/controller servers.

Again we are not aligned. those specific BT Nodes only offers the possibility of being changed from BT but not from the ROS network (that is what I was asking about)

>By default they're set to GridBased and FollowPath as are default in the servers themselves. You can just change to another value hardcoded in the BT or using the blackboard parameters if you'd like it to change over time in the same run from a service / topic call.

Can I directly write into the BT blackboard directly from the ROS network? That could also be a valid solution for me, but I think that it is not supported and it is not planned to be supported in the future. Am I wrong?

>PlannerSelector

>A planner selector / controller selector BT node that changes the blackboard values and a BT XML that uses those blackboard values in the follow path / compute path would be a valuable addition.

Okay.

>I don't think its the right solution for every user that wants to change algorithms, but it could be useful for enough that I'd support it.

Not all the functionalities are useful for everybody. I also may not need the LowBatteryBT. :-)

> The way I would rather have people use multiple planners / controllers would be to have conditions for when each are useful so they change change dynamically themselves through the BT rather than external input.

Again, I think that goes against the ROS distributed philosophy (did this change?). It is great if you can do it internally from the BT, and it is great if you propose some "best practices" for navigation. But you should not be forced to write c++ code to configure the navigation if (If the navigation stack 2 is going to be the canonical solution for navigation in ROS2) what if I am a python programmer that want to control robots from a remote node? Cant I change the planner and controller dynamically?

It looks to me like the navigation stack is requesting to be the center of all the robot application, even if in some cases the navigation may be a secondary functionality.

>Whatever external node is requesting the change shouldn't exist and should be integrated into the condition node so that change is fully autonomous within the tree.

>But I recognize that it could be a good node to add to our BT node library. I'd review a PR on it :-)

We think differently, but it is great if we can find some agreement at least with this planner selector :-)

SteveMacenski, 2020-11-14:
> What if I am a python developer that do not know anything about c++ or behavior trees?

My (unfiltered) opinion on python: if you're unwilling to learn anything but python, you shouldn't be building robots. Python has no place in a navigation system, its not fast enough and doesn't give you the tools required to create a stable product or result. You cannot control memory allocations and the python implementations in ROS are very heavy.  That's not to say everything needs to be C++, rust, C, java, and other are equilally valid, but it certainly can't be python. On behavior trees: learn. This implementation uses behavior trees, there is a required learning curve to learn about them to use Navigation2 effectively. Every new technology is going to require you to learn something new to use it, this is no different. You had to learn about costmaps to use Navigation in ROS1, and then had C++ plugins to implement your algorithms. This is exactly analog. 

> Again we are not aligned. those specific BT Nodes only offers the possibility of being changed from BT but not from the ROS network (that is what I was asking about)

Please read carefully my previous comments. I clearly laid out a strategy for accessing or changing the blackboard as a network interface. 

Regardless, we have alignment in a planner / controller selector BT action node that we can agree would have sufficient re-use to place into the main nav2 stack.

pabloinigoblasco, 2021-03-16:
This conversation is old.
It derived in this other issue talking about how to implement the planner_selector, the controller_selector and the goal_checker_selector concepts: https://github.com/ros-planning/navigation.ros.org/issues/147

SteveMacenski, 2021-03-16:
That's for design, if anything that ticket should be closed. Lets leave this open until something is merged. That other ticket is on our documentation website, not a Nav2 feature

SteveMacenski, 2021-03-23:
Done!

closed 2021-03-23 by SteveMacenski