ros-navigation/navigation2 #2094: [foxy][tutorial] tb3_simulation_launch.py not launching properly

mag-seasony, 2020-11-16:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version: Foxy binaries
- Version or commit hash: ros-foxy-navigation2	0.4.3-1focal.20201103.171126
- DDS implementation: not sure

#### Steps to reproduce issue
As specified in [this](https://navigation.ros.org/getting_started/index.html) tutorial:

```
source /opt/ros/<ros2-distro>/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models
```
then
```
ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior
turtlebot 3 in a simulated environment, with the navigation stack up and running

#### Actual behavior
Gazebo fail to launch properly (get stuck then do not load the sdf), and the navigation stack does not run with the following error:
```
Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
```

#### Additional information

all the output from the launch:
```
os2 launch nav2_bringup tb3_simulation_launch.py
[INFO] [launch]: All log files can be found below /home/mg/.ros/log/2020-11-16-14-53-06-981842-varus-60948
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [60950]
[INFO] [gzclient-2]: process started with pid [60952]
[INFO] [robot_state_publisher-3]: process started with pid [60954]
[INFO] [rviz2-4]: process started with pid [60956]
[INFO] [map_server-5]: process started with pid [60958]
[INFO] [amcl-6]: process started with pid [60960]
[INFO] [lifecycle_manager-7]: process started with pid [60962]
[INFO] [controller_server-8]: process started with pid [60965]
[INFO] [planner_server-9]: process started with pid [60992]
[INFO] [recoveries_server-10]: process started with pid [61006]
[INFO] [bt_navigator-11]: process started with pid [61016]
[INFO] [waypoint_follower-12]: process started with pid [61022]
[INFO] [lifecycle_manager-13]: process started with pid [61027]
[map_server-5] [INFO] [map_io]: Loading yaml file: /opt/ros/foxy/share/nav2_bringup/maps/turtlebot3_world.yaml
[map_server-5] [DEBUG] [map_io]: resolution: 0.05
[map_server-5] [DEBUG] [map_io]: origin[0]: -10
[map_server-5] [DEBUG] [map_io]: origin[1]: -10
[map_server-5] [DEBUG] [map_io]: origin[2]: 0
[map_server-5] [DEBUG] [map_io]: free_thresh: 0.196
[map_server-5] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-5] [DEBUG] [map_io]: mode: trinary
[map_server-5] [DEBUG] [map_io]: negate: 0
[map_server-5] [INFO] [map_io]: Loading image_file: /opt/ros/foxy/share/nav2_bringup/maps/turtlebot3_world.pgm
[map_server-5] [DEBUG] [map_io]: Read map /opt/ros/foxy/share/nav2_bringup/maps/turtlebot3_world.pgm: 384 X 384 map @ 0.05 m/cell
[robot_state_publisher-3] [WARN] [1605534787.283666952] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 7 children
[robot_state_publisher-3] Link camera_link had 2 children
[robot_state_publisher-3] Link camera_depth_frame had 1 children
[robot_state_publisher-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[robot_state_publisher-3] Link camera_rgb_optical_frame had 0 children
[... 155 lines cut ...]
[controller_server-8] [INFO] [1605534787.814579662] [controller_server]: Activating
[controller_server-8] [INFO] [1605534787.814603825] [local_costmap.local_costmap]: Activating
[controller_server-8] [INFO] [1605534787.814613553] [local_costmap.local_costmap]: Checking transform
[controller_server-8] [INFO] [1605534787.814640479] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-8] [INFO] [1605534788.314705874] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1605534788.337316691] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1605534788.337509985] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1605534788.357361925] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [WARN] [1605534788.498757339] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [WARN] [1605534788.510852313] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [WARN] [1605534788.529816708] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [INFO] [1605534788.570599841] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1605534788.570797473] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-4] [INFO] [1605534788.571369953] [rviz2]: Trying to create a map of size 384 x 384 using 1 swatches
[rviz2-4] [ERROR] [1605534788.587226059] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-4] active samplers with a different type refer to the same texture image unit
[controller_server-8] [INFO] [1605534788.814694183] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-8] [INFO] [1605534789.314725818] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1605534789.571181821] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1605534789.571462711] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[controller_server-8] [INFO] [1605534789.814769573] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-8] [INFO] [1605534790.314779043] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1605534790.571815358] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1605534790.572147448] [rviz2]: Sending lifecycle_manager_navigation/is_active request
```

SteveMacenski, 2020-11-16:
Have you modified anything off default? Are you sure gazebo is running (e.g. do you see the gazebo window is fully loaded and updating laser scans)? That's an error that it can't get odometry, so gazebo's plugins for odometry might not be running / up or crashed. Did you also set the initial pose as the instructions list? The other big reason that TF complains on startup is if you didn't set an initial pose, which is required.

mag-seasony, 2020-11-17:
I did not modify anything from the default, gazebo manage to run after few mins, but the sdf is not loaded. I did not set the initial pose (my fault as I did not see it in the tutorial). 
Thanks for the help, It is working right now.
The 2 things I had to do for making it work is to indeed set the initial pos (note: isn't it possible to set it by default?), and to download the models from the github as the foxy version does not seem to be available through apt

Again, thanks for the help,
cheers!

closed 2020-11-17 by mag-seasony