ros-navigation/navigation2 #2100: Update nav2_behavior_tree/nav2_tree_nodes.xml

moritzadlatus, 2020-11-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy Fitzroy
- Version or commit hash:
  - 3d166b748bd704eca05ef671b0b2ca99968a8c73

#### Steps to reproduce issue
import "nav2_tree_nodes.xml" to Groot, Load Tree "navigate_w_replanning_and_recovery.xml"

#### Expected behavior
opening Behavior Tree

#### Actual behavior
"Exception! It was not possible to parse the file. Error: This model has not been registered: GoalUpdated"
#### Additional information
I fixed it by renaming GoalReached to GoalUpdated. Is it the same though?

SteveMacenski, 2020-11-24:
https://github.com/ros-planning/navigation2/blob/0cdf4f44c7a4171bfa4c6eb885d46c3c19f26a87/nav2_behavior_tree/plugins/condition/goal_updated_condition.cpp

goal updated is also a valid node name

gramss, 2020-11-24:
Currently (as in this moment) trying to figure out the zmq performance stuff.

But this issue should be straight forward.

Groot needs to have a "factory" / model base to create your BT from (visually). In this case only the visual part with how many in/outputs, node_type, and the node ID is needed.
This information lives in: https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/nav2_tree_nodes.xml
@moritzadlatus did the right steps. But the new (at least I guess so) plugin called `goal_updated` is still missing from this.

So one would need to add (not tested):

```
    <Control ID="GoalUpdated">
      <input_port name="goal">Current Goal</input_port>
    </Control>
``` 
exactly here: https://github.com/ros-planning/navigation2/blob/f6bbc5c81e21324f66c663194e93f3f0d2109dee/nav2_behavior_tree/nav2_tree_nodes.xml#L47

Interesting to note here is that the blackboard is not transferred to groot and also not checked or displayed.
Therefor, the `Goal_Reached` plugin has 2 additional fields https://github.com/ros-planning/navigation2/blob/dcc1d4dafba6f6c7588cb5f7f6ca918b2a6c48ef/nav2_behavior_tree/plugins/condition/goal_reached_condition.cpp#L35

Those 2 fields are missing as well in the nav2_tree_nodes.xml ..

There exists some efforts to directly read the plugins and skip the nodes.xml part completely. But I think those do also not interact with the blackboard inputs/outputs .. I need to dig into this deeper.

But for now adding the above code should show the right node.
Swapping out the ID of the node in the actual BT and running an alternate version of it, swaps out the complete node and alters the function of the BT.

@moritzadlatus could you test this please? :slightly_smiling_face: 

@SteveMacenski probably a thing to add to the tutorial for creating a new node? (adding it to the nodes.xml for Groot to display it nicely?)

moritzadlatus, 2020-11-26:
Adding the node worked, thanks!

SteveMacenski, 2020-11-30:
Grrr....

closed 2020-11-30 by SteveMacenski