ros-navigation/navigation2 #2102: bt_navigator reporting nullptr decorator/control node when setting waypoint

diogopbrandao, 2020-11-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  -Foxy

#### Steps to reproduce issue

```

```

#### Expected behavior
Start to navigate to the set goal.
#### Actual behavior

Does not move.

Prints the following:

[bt_navigator-19] [INFO] [1606231704.751179399] [bt_navigator]: Begin navigating from current location to (0.19, -0.21)
[bt_navigator-19] [ERROR] [1606231704.752934680] [bt_navigator]: Action server failed while executing action callback: "One of the children of a DecoratorNode or ControlNode is nullptr"
[bt_navigator-19] [WARN] [1606231704.756022148] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.

SteveMacenski, 2020-11-24:
Did you try with our default provided settings? Try to back out from out defaults until you can figure out where your custom modifications have created this error potentially.  I can't say that I've run into this error before to give you a clear sense of what's wrong.  

This is clearly coming from BT.CPP https://docs.ros.org/en/melodic/api/behaviortree_cpp_v3/html/behavior__tree_8cpp_source.html which should be of no surprise. Maybe you created a custom node in your BT that doesn't exist so it's just a nullptr as a placeholder?

SteveMacenski, 2020-12-02:
@diogopbrandao can you respond to my questions?

diogopbrandao, 2020-12-02:
@SteveMacenski I apologize for being silent for so long. I've been following your suggestions and finally found the problem that was giving this error. As I said the problem started when nav2 was upgraded from the eloquent-devel to foxy-devel. The problem was in the behavior tree set up.

I found out that on eloquent-devel, on package nav2_bt_navigator, the file bt_navigator.cpp takes as an argument "**bt_xml_filename**". On the foxy-devel this argument has changed to "**default_bt_xml_filename**", This correction solved the error mentioned above after correcting it in all all launch files and nav2_params.yaml . Also, the bt called navigate_w_replanning.xml  is not present anymore, which led me to change this tree that was being used in this project to the bt navigate_w_replanning_and_recovery.xml, and then added the missing nodes to make this bt work.

At this point, after setting a way point in rviz, the error does not appear and a path is generated. However, the robot is still not moving and a warning message keeps appearing:

**[controller_server-17] [WARN] [1606899310.530681506] [controller_server]: Control loop missed its desired rate of 20.0000Hz**

I'm gonna keep looking to see why the robot is not moving and this warning keeps showing up.

Bottom line, as I took this project the launch files and config file nav2_params.yaml were basically set to work on eloquent-devel, but the nav2 was already in the foxy-devel version.

closed 2020-12-02 by SteveMacenski