ros-navigation/navigation2 #2109: Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"

Mohit-Ak, 2020-11-28:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Foxy binaries
- Version or commit hash:
```
ros-foxy-nav2-map-server	0.4.3-1focal.20201103.155816
ros-foxy-nav2-map-server-dbgsym	0.4.3-1focal.20201103.155816
```
#### Steps to reproduce issue

```
 ros2 run nav2_map_server map_saver_cli -f knoxville_maps/sample2
```

#### Expected behavior
- To save the map.

#### Actual behavior
```
 ros2 run nav2_map_server map_saver_cli -f knoxville_maps/sample2
[INFO] [1606526736.194038079] [map_saver]: 
	map_saver lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1606526736.194279197] [map_saver]: Creating
[INFO] [1606526736.194598763] [map_saver]: Saving map from 'map' topic to 'knoxville_maps/sample2' file
[WARN] [1606526736.194665291] [map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN] [1606526736.194696357] [map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[WARN] [map_io]: Image format unspecified. Setting it to: pgm
[INFO] [map_io]: Received a 1530 X 3424 map @ 0.05 m/pix
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
```
#### Additional information
Related tickets in graphicsmagick.
https://github.com/nagadomi/waifu2x/issues/131
Related unanswered ROS Question.
https://answers.ros.org/question/343711/when-running-nav2_map_server-magick-abort-due-to-signal-11/

Things tried:
- Tried saving ```/map``` topic from different bag files.
- Tried ```sudo apt-get upgrade```
- Tried uninstalling and re-installing ```sudo apt-get remove libgraphicsma*```. This also uninstalled the dependant ```ros-foxy-nav2``` packages which I then installed back.

PC and CPU Specs
- Lambda Labs DeepLearning Workstation with 64Gb RAM and two Nvidia 2080ti
```aa@bb:~/ros2_ws$lscpu
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   46 bits physical, 48 bits virtual
CPU(s):                          24
On-line CPU(s) list:             0-23
Thread(s) per core:              2
Core(s) per socket:              12
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           85
Model name:                      Intel(R) Core(TM) i9-7920X CPU @ 2.90GHz
Stepping:                        4
CPU MHz:                         3799.999
CPU max MHz:                     4400.0000
CPU min MHz:                     1200.0000
BogoMIPS:                        5799.77
Virtualization:                  VT-x
L1d cache:                       384 KiB
L1i cache:                       384 KiB
L2 cache:                        12 MiB
L3 cache:                        16.5 MiB
```

SteveMacenski, 2020-11-30:
If this is a `magick` library issue, as this ticket implies with bringing up the same error in other non-ROS libraries, then this ticket should be closed and filed against magick with any minimal reproduction instructions.  

Do you see this building from source?

Mohit-Ak, 2020-12-03:
Thankyou @SteveMacenski . Just built it from the source and the issue got resolved and hence closing the ticket.

closed 2020-12-03 by Mohit-Ak

SteveMacenski, 2020-12-03:
Glad you found a way around it!

S

BastianMuratory, 2022-06-29:
Hello, I am having the same issue, I'm still in the process of learning ros2, and I don't quite understand what you mean build building it from source ? How can I do that ?

brean, 2022-12-04:
I like to reopen this issue. I am getting the same error in Ubuntu 22.04 with ROS 2 humble and the latest debian packages for navigation2 for running with Gazebo Garden in a docker container:

```
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
```

after setting a pose estimate in RVIZ2. (before that I am getting the messages that planner_server, lifecycle_manager_navigation, behavior_server, lifecycle_manager_navigation, bt_navigator, waypoint_follower and velocity_smoother were activating and I don't see any other errors in the log).

I am installing it from the deb-packages here: https://github.com/brean/ros2-turtlebot3-gazebo-docker/tree/humble

I tried to use GraphicsMagick 1.3.37 inside my docker-container on the humble-naav2-source branch but after when it tries to compile the nav2_map_server I am getting this error:
```
--- stderr: nav2_map_server
/usr/bin/ld: /usr/local/lib/libGraphicsMagick++.a(libGraphicsMagick___la-Drawable.o): warning: relocation against `_ZTVN6Magick29PathSmoothQuadraticCurvetoAbsE' in read-only section `.text'
/usr/bin/ld: /usr/local/lib/libGraphicsMagick++.a(libGraphicsMagick___la-Color.o): relocation R_X86_64_PC32 against symbol `_ZTVN6Magick5ColorE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/map_io.dir/build.make:327: libmap_io.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:347: CMakeFiles/map_io.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

(It says I should recompile with -fPIC but I am already compiling GraphicsMagick with the fPIC-flag)
```

SteveMacenski, 2022-12-05:
I'd ask for the logs or for you to look at them more carefully. Usually when you see that error from Magick its because servers are crashing and that's just the way the map server crashes in response, but almost always its not related to this log - its something prior.

Namo98, 2023-09-12:
Hey Steve,

i have the same Error message:
```
ros2 run nav2_map_server map_saver_cli -f maps/airport_terminal 
1694509645.805340 [0] map_saver_: using network interface enp34s0 (udp/192.168.178.46) selected arbitrarily from: enp34s0, docker0
[INFO] [1694509645.815983913] [map_saver]: 
	map_saver lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1694509645.816086415] [map_saver]: Creating
[INFO] [1694509645.816510484] [map_saver]: Saving map from 'map' topic to 'maps/airport_terminal' file
[WARN] [1694509645.816568575] [map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN] [1694509645.816599396] [map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[WARN] [map_io]: Image format unspecified. Setting it to: pgm
[INFO] [map_io]: Received a 2731 X 1929 map @ 0.04 m/pix
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
```
You said as an option to  build from source. Which package do you mean?