ros-navigation/navigation2 #2110: nav2_recoveries actions aren't getting cancelled when alternating quickly between actions

tal-grossman, 2020-11-29:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04
- ROS2 Version:
  Foxy binaries
- Version or commit hash:
 0.4.3-1focal.20201103.171126
- DDS implementation:
Fast-RTPS

#### Steps to reproduce issue
1. Have a behavior tree to call and alternate recovery actions (spin, wait for example) _fast enough_. I assume **fast is higher than the cycle_frequency** 
Example of behavior tree that alternates every 'RateController hz' between spin and wait:
```
<!--
Every RateController hz "{the_answer}" value is switched from -1 to 1
if the "{the_answer}" value is 1 - Spin is called, if -1 Wait is called.
when "{the_answer}" value alternates is cancels the previous action and calls for the second one.    
-->
<root main_tree_to_execute="MainTree">
    <BehaviorTree ID="MainTree">
        <Repeat num_cycles="-1">
            <PipelineSequence name="test procedure">
                <RateController hz="50.0">
                    <IfThenElse>
                        <BlackboardCheckString value_A="{the_answer}" value_B="1" return_on_mismatch="FAILURE">
                            <AlwaysSuccess/>
                        </BlackboardCheckString>
                        <SetBlackboard value="-1" output_key="the_answer"/>
                        <SetBlackboard value="1" output_key="the_answer"/>
                    </IfThenElse>
                </RateController>
                <WhileDoElse>
                    <BlackboardCheckString value_A="{the_answer}" value_B="1" return_on_mismatch="FAILURE">
                        <AlwaysSuccess/>
                    </BlackboardCheckString>
                    <Spin spin_dist="3.14"/>
                    <Wait wait_duration="10"/>
                </WhileDoElse>
            </PipelineSequence>
        </Repeat>
    </BehaviorTree>
</root>
```
2. Using CLI call /navigate_to_pose action with the predefined behavior tree for example: _"test_cancelling.xml"_. goal pose does not matter.
```
ros2 action send_goal /navigate_to_pose nav2_msgs/action/NavigateToPose '{behavior_tree: "FULL_PATH_TO/test_cancelling.xml"}'
```
#### Expected behavior
1. the previous action is cancelled cleanly when the next one starts.
2. if client aborts ('ctrl + c' from the same 'send_goal' terminal), all actions are cancelled. (see logs).
#### Actual behavior
1. The previous action sometimes keeps running with the new action. In this case we will get 'wait' + 'spin' running in the same time 
2. Sometimes after abort the actions continue running until completion.
```
[recoveries_server-3] [1606649811.700406859][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.720016529][recoveries_server][INFO]: Attempting wait
[recoveries_server-3] [1606649811.789984629][recoveries_server][INFO]: Canceling spin
[recoveries_server-3] [1606649811.790088626][recoveries_server][WARN]: [spin] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.790781526][recoveries_server][WARN]: [spin] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.800247976][recoveries_server][INFO]: Attempting spin
[recoveries_server-3] [1606649811.800316978][recoveries_server][INFO]: Turning 3.14 for spin recovery.
[recoveries_server-3] [1606649811.820186779][recoveries_server][INFO]: Canceling wait
[recoveries_server-3] [1606649811.820263308][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.820408260][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.830151258][recoveries_server][INFO]: Attempting wait
[bt_navigator-4] [1606649811.849598774][bt_navigator][INFO]: Navigation canceled
[bt_navigator-4] [1606649811.849615685][bt_navigator][WARN]: [navigate_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
[bt_navigator-4] [1606649811.950996904][bt_navigator][ERROR]: Action server failed while executing action callback: "client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.2/src/rmw_response.cpp:123, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/service.c:356"
[recoveries_server-3] [1606649812.800588485][recoveries_server][INFO]: spin running...
[recoveries_server-3] [1606649812.830405557][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649813.500577831][recoveries_server][WARN]: Collision Ahead - Exiting Spin
[recoveries_server-3] [1606649813.500618105][recoveries_server][INFO]: spin completed successfully
[recoveries_server-3] [1606649813.830388440][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649814.830337959][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649815.830372117][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649816.830401794][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649817.830389271][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649818.830508042][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649819.830452554][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649820.830494531][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649821.830279852][recoveries_server][INFO]: wait completed successfully

```
#### Additional information
1. I first encountered this issue when running costumed behavior tree with costumed recoveries actions. The same bug reproduced with more than 2 actions. I got several actions running in the same time when they clearly got cancel command (so I don't believe it a bt issue).  

After adding some log prints in _**nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp**_:
[image]
- It gets to the "should_cancel_goal()" ok. (no "Failed to cancel action server" log print)
 

- But somehow it is not cancelled here **_nav2_recoveries/include/nav2_recoveries/recovery.hpp_**:
[image]

2. It might also be rclcpp action bug, but i'd like to hear what you think
3. Same issue can be also reproduced in controller_server?

SteveMacenski, 2020-11-30:
So what you're saying is that you're expecting that a recovery action will stop running when you request another recovery action to run? That's not supported and I wouldn't think sensible, given they are all widely independent action servers with their own tasks.

If something like that were to be implemented, there would need to be a background thread in the recovery server https://github.com/ros-planning/navigation2/blob/main/nav2_recoveries/src/recovery_server.cpp checking on the states of the recovery's action servers and killing any of them running if a new one is requested. However, I don't think this is something worth implementing.  

It seems to me that the BT nodes you're using to interact with the recovery servers could also implement that logic, but also that your BT may be misconfigured to allow for multiple BT recovery nodes to be allowed to be in the `RUNNING` state at once, that should never be possible. Only 1 leaf node can be `RUNNING` in a properly created BT and BT node plugins.  It seems like you have other fishy problems if you're able to be calling multiple recoveries at once and they're both running but at least one isn't returning a `RUNNING` state.

tal-grossman, 2020-12-01:
> To make sure we're on the same page: what you're saying is that you're expecting that a recovery action will stop running when you request another recovery action to run? That's not supported and I wouldn't think sensible, given they are all widely independent action servers with their own tasks.

Generally I agree. I don't expect or think it should be design and implemented that way, they are independent actions and should stay as such. If a client calls for 2 or more independent actions, they should run simultaneously. 

Having said that, In the case I'm trying to describe here where the BT is stopping one action and calls for the second one repeatedly, I do expect that result.
> 

> It seems to me that the BT nodes you're using to interact with the recovery servers could also implement that logic, but also that your BT may be misconfigured to allow for multiple BT recovery [...]

I don't think the BT is the problem becuase as mentioned I do see cancel requests get here [bt_action_node.hpp - halt](https://github.com/ros-planning/navigation2/blob/11b25a51981eb53db6a6a3601729ae85238d9241/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L183) (I added some prints to verify)

My assumption it is related to the implementation of the recovery_server itself because:
1. When lowering the _'stop one action and start new action'_ requests frequencies - (in the example above it is the RateController hz), to be lower that the cycle_frequency the result will be as expected.

2. The cancelling requests checking [in recovery interface](https://github.com/ros-planning/navigation2/blob/11b25a51981eb53db6a6a3601729ae85238d9241/nav2_recoveries/include/nav2_recoveries/recovery.hpp#L203) is depended on the cycle_frequency. meaning that the client's cancel request will only get preformed after 1 cycle iteration. for example, set the cycle frequency to 0.5 second, will get you 2 second of running while it should be cancelled. lower than this sometimes the recovery will action missed the cancelling and just keeps running

3. Using BT to have repeatedly request and cancel an action gets messier results as the frequency is getting higher. On my costumed application, I'm using the BT's frequnecy (defualt is 100 hz) to alternate between action by a perception condition.

SteveMacenski, 2020-12-02:
You didn't answer my question in the quote.... please verify that we're on the same page before its meaningful to continue discussion on topics that build on the concept. 

>  I don't expect or think **it** should be design and implemented that way, **they** are independent actions and should stay as such

You're also talking about recoveries, BT nodes, and other stuff. Please be painfully specific or we're not going to be able to get anywhere fast due to miscommunication. The recoveries are independent already so I'm not sure what you mean. It sounds like you have an issue with how something is designed or implemented, but then describe the current state. 

> I don't think the BT is the problem becuase ...

Just from a first principles look, if we all agree that the recoveries right now are not related to each other, then if things aren't stopping correctly, its most likely from improperly calling those action servers (which are done in BT nodes). There's more to building a BT node than just simply calling cancel, there's a whole state machine design you need to be aware of. 

It should be trivial for you to make a quick python script on cancellation with recovery server to completely remove the BT from the equation to test.

daisukes, 2020-12-03:
Hi @tal-grossman and @SteveMacenski 
I came here because my PR was tagged ;-)

I found the issue which is talking about the same error message "client will not receive response."
https://github.com/ros2/ros2/issues/1024

It looks like the behavior tree is terminated with an exception.
> `[bt_navigator-4] [1606649811.950996904][bt_navigator][ERROR]: Action server failed while executing action callback: "client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.2/src/rmw_response.cpp:123, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/service.c:356"`

https://github.com/ros-planning/navigation2/blob/e77560d87ac816ff72663c36055ce35c1a45c942/nav2_util/include/nav2_util/simple_action_server.hpp#L136-L144

`WhileDoElse` halts the current action before ticking the next one, but cancel is called asynchronously, so I think this can happen.

```
[recoveries_server-3] [1606649811.700406859][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.720016529][recoveries_server][INFO]: Attempting wait
 # spin1 cancel is called async, wait1 started to run
[recoveries_server-3] [1606649811.789984629][recoveries_server][INFO]: Canceling spin
[recoveries_server-3] [1606649811.790088626][recoveries_server][WARN]: [spin] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.790781526][recoveries_server][WARN]: [spin] [ActionServer] Client requested to cancel the goal. Cancelling.
 # spin1 is cancelled 
[recoveries_server-3] [1606649811.800247976][recoveries_server][INFO]: Attempting spin
 # wait1 cancel is called async, spin2 started to run
[recoveries_server-3] [1606649811.800316978][recoveries_server][INFO]: Turning 3.14 for spin recovery.
[recoveries_server-3] [1606649811.820186779][recoveries_server][INFO]: Canceling wait
[recoveries_server-3] [1606649811.820263308][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
[recoveries_server-3] [1606649811.820408260][recoveries_server][WARN]: [wait] [ActionServer] Client requested to cancel the goal. Cancelling.
 # wait1 is cancelled 
[recoveries_server-3] [1606649811.830151258][recoveries_server][INFO]: Attempting wait
 # spin2 cancel is called async, wait2 started to run
[bt_navigator-4] [1606649811.849598774][bt_navigator][INFO]: Navigation canceled
[bt_navigator-4] [1606649811.849615685][bt_navigator][WARN]: [navigate_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
[bt_navigator-4] [1606649811.950996904][bt_navigator][ERROR]: Action server failed while executing action callback: "client will not receive response, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-shared-cpp-1.2.2/src/rmw_response.cpp:123, at /tmp/binarydeb/ros-foxy-rcl-1.1.9/src/rcl/service.c:356"
 # somehow spin2 cancel call failed and the bt_navigator is terminated, meaning spin2 keeps running
[recoveries_server-3] [1606649812.800588485][recoveries_server][INFO]: spin running...
[recoveries_server-3] [1606649812.830405557][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649813.500577831][recoveries_server][WARN]: Collision Ahead - Exiting Spin
[recoveries_server-3] [1606649813.500618105][recoveries_server][INFO]: spin completed successfully
 # spin2 completed
[recoveries_server-3] [1606649813.830388440][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649814.830337959][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649815.830372117][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649816.830401794][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649817.830389271][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649818.830508042][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649819.830452554][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649820.830494531][recoveries_server][INFO]: wait running...
[recoveries_server-3] [1606649821.830279852][recoveries_server][INFO]: wait completed successfully
 # wait2 completed
```

SteveMacenski, 2020-12-04:
@daisukes sorry, what's the statement you're trying to make? The error is a service call failing, but I'm not sure what you mean by "... but cancel is called asynchronously, so I think this can happen."

Are you suggesting there's something wrong with Nav2 we can patch to fix this?

daisukes, 2020-12-04:
> @daisukes sorry, what's the statement you're trying to make? The error is a service call failing, but I'm not sure what you mean by "... but cancel is called asynchronously, so I think this can happen."

I meant that if cancellation is a synchronous call, two actions should not happen simultaneously. However here cancellation is asynchronous and the BT tree does not wait for the cancellation, so two actions can happen at a time in this case.

https://github.com/ros-planning/navigation2/blob/e77560d87ac816ff72663c36055ce35c1a45c942/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L186-L187

`spin_until_future_complete` just waits until `cancel` is acknowledged by the server.

> Are you suggesting there's something wrong with Nav2 we can patch to fix this?

No, I think it is a DDS layer issue.

SteveMacenski, 2020-12-04:
Got it. I think tl;dr for @tal-grossman, your issue is with Fast-DDS service issues, cyclone should fix this. It technically also has service issues, but I've never heard reports of them in nav2 (but consistently for Fast-DDS), so whatever Cyclone does differently seems to make this issue **much** less frequent. 

CC @EduPonz @JaimeMartin 

I'm going to close this issue since I think that's the solution from the error. I'm more than happy to reopen it if you find that your issue is not in fact DDS only and there's something we need to address in Nav2 or rclcpp (or something above DDS).

closed 2020-12-04 by SteveMacenski

tal-grossman, 2020-12-06:
Thank you @daisukes for the clarification and the link to the Fast-DDS issue. As I pointed out in the issue I wasn't sure if is pure nav2 issue or something else. 
@SteveMacenski I'll be sure to test it out on Cyclone and if it is reproduced I'll try to dig more into it.

EduPonz, 2020-12-09:
Hi everyone,

First of all, thanks for digging into this. The fix backport (ros2/rmw_fastrtps#490) was just merged into rmw_fasrtps' foxy branch. This should solve most of the issues that arise in Nav2 regarding actions and services. It'd be really great if you guys could give it a try and tell us how it goes. @jacobperron , is this going to be included in this Friday's (11th December 2020) Foxy sync?

jacobperron, 2020-12-09:
Yes, ros2/rmw_fastrtps#490 should be included in the sync this Friday.

SteveMacenski, 2020-12-09:
@tal-grossman so you may be able to try again next week and see if your issue is resolved to continue using Fast-DDS.