ros-navigation/navigation2 #2112: Navigating with keepout zones tutorial not successful

Antoinette-abi, 2020-12-01:
- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  -  Rolling built from source
- DDS implementation:
  -  Fast-RTPS

#### Steps to reproduce issue

I was working on the tutorial for keepout filter as given in "https://navigation.ros.org/tutorials/docs/navigation2_with_keepout_filter.html#"

1. The Python launch script to enable the costmap filter info publisher server and the map server as lifecycle nodes, the yaml file with ros parameters and an example for the filter mask is launched using the given navigation2_tutorials directory. 

```
ros2 launch nav2_keepout_filter_demo costmap_filter_info.launch.py params_file:=src/navigation2_tutorials/nav2_keepout_filter_demo/params/keepout_params.yaml mask:=src/navigation2_tutorials/nav2_keepout_filter_demo/maps/keepout_mask.yaml
```
2. Once the costmap_filter_info.launch.py is run, the keepout filter is enabled in costmap 2D by adding keepout_filter to the plugins parameter in nav2_params.yaml file. The keepout_filter plugin is enabled for both the local and global costmaps.
3. Nav2 stack is then started using 
```
ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior
Keepout filter should be enabled in the global costmap and would be visible as pink zones on rviz. 

#### Actual behavior

On running the demo costmap_filter_info.launch.py file with the given params and mask, the following output is obtained which is as shown in the tutorial video.
```
~/tutorials_ws$ ros2 launch nav2_keepout_filter_demo costmap_filter_info.launch.py params_file:=src/navigation2_tutorials/nav2_keepout_filter_demo/params/keepout_params.yaml mask:=src/navigation2_tutorials/nav2_keepout_filter_demo/maps/keepout_mask.yaml
[INFO] [launch]: All log files can be found below /home/nena/.ros/log/2020-12-01-14-02-52-499641-nena-NUC7i7DNHE-18976
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [18978]
[INFO] [map_server-2]: process started with pid [18980]
[INFO] [costmap_filter_info_server-3]: process started with pid [18982]
[costmap_filter_info_server-3] [INFO] [1606827772.592896211] [costmap_filter_info_server]: This is costmap filter info publisher
[lifecycle_manager-1] [INFO] [1606827772.596917235] [lifecycle_manager_costmap_filters]: Creating
[lifecycle_manager-1] [INFO] [1606827772.604382355] [lifecycle_manager_costmap_filters]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [1606827772.606637380] [lifecycle_manager_costmap_filters]: Starting managed nodes bringup...
[lifecycle_manager-1] [INFO] [1606827772.606687038] [lifecycle_manager_costmap_filters]: Configuring filter_mask_server
[costmap_filter_info_server-3] [INFO] [1606827772.622847477] [costmap_filter_info_server]: 
[costmap_filter_info_server-3] 	costmap_filter_info_server lifecycle node launched. 
[costmap_filter_info_server-3] 	Waiting on external lifecycle transitions to activate
[costmap_filter_info_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1606827772.623645577] [filter_mask_server]: 
[map_server-2] 	filter_mask_server lifecycle node launched. 
[map_server-2] 	Waiting on external lifecycle transitions to activate
[map_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1606827772.623800881] [filter_mask_server]: Creating
[map_server-2] [INFO] [1606827772.762830603] [filter_mask_server]: Configuring
[map_server-2] [INFO] [map_io]: Loading yaml file: src/navigation2_tutorials/nav2_keepout_filter_demo/maps/keepout_mask.yaml
[map_server-2] [DEBUG] [map_io]: resolution: 0.1
[map_server-2] [DEBUG] [map_io]: origin[0]: -0.45
[map_server-2] [DEBUG] [map_io]: origin[1]: -0.45
[map_server-2] [DEBUG] [map_io]: origin[2]: 0
[map_server-2] [DEBUG] [map_io]: free_thresh: 0.25
[map_server-2] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-2] [DEBUG] [map_io]: mode: scale
[map_server-2] [DEBUG] [map_io]: negate: 0
[map_server-2] [INFO] [map_io]: Loading image_file: src/navigation2_tutorials/nav2_keepout_filter_demo/maps/keepout_mask.pgm
[map_server-2] [DEBUG] [map_io]: Read map src/navigation2_tutorials/nav2_keepout_filter_demo/maps/keepout_mask.pgm: 38 X 42 map @ 0.1 m/cell
[lifecycle_manager-1] [INFO] [1606827772.778885892] [lifecycle_manager_costmap_filters]: Configuring costmap_filter_info_server
[costmap_filter_info_server-3] [INFO] [1606827772.780778763] [costmap_filter_info_server]: Configuring
[lifecycle_manager-1] [INFO] [1606827772.789850273] [lifecycle_manager_costmap_filters]: Activating filter_mask_server
[map_server-2] [INFO] [1606827772.790743442] [filter_mask_server]: Activating
[map_server-2] [INFO] [1606827772.791211227] [filter_mask_server]: Creating bond (filter_mask_server) to lifecycle manager.
[lifecycle_manager-1] [INFO] [1606827772.902918237] [lifecycle_manager_costmap_filters]: Server filter_mask_server connected with bond.
[lifecycle_manager-1] [INFO] [1606827772.903127169] [lifecycle_manager_costmap_filters]: Activating costmap_filter_info_server
[costmap_filter_info_server-3] [INFO] [1606827772.904024136] [costmap_filter_info_server]: Activating
[costmap_filter_info_server-3] [INFO] [1606827772.904299193] [costmap_filter_info_server]: Creating bond (costmap_filter_info_server) to lifecycle manager.
[lifecycle_manager-1] [INFO] [1606827773.013789926] [lifecycle_manager_costmap_filters]: Server costmap_filter_info_server connected with bond.
[lifecycle_manager-1] [INFO] [1606827773.013988478] [lifecycle_manager_costmap_filters]: Managed nodes are active
[lifecycle_manager-1] [INFO] [1606827773.014095065] [lifecycle_manager_costmap_filters]: Creating bond timer...

```
On running the navigation 2 stack, I can move around the robot but the keepout zones are not enabled in the costmap as shown in the image. 
[image]

I've followed the exact steps as given in the tutorial but i keep running into this issue. Any inputs on why this is not working for me ? @AlexeyMerzlyakov

SteveMacenski, 2020-12-01:
Can you provide the part of your nav2 params file that you changed to add keepout zones? Maybe a topic isn't correctly aligned.

Antoinette-abi, 2020-12-01:
Yes. So i've added the keepout_filter in the plugins parameter in the local and global costmap.

`local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22
      **plugins: ["voxel_layer", "inflation_layer", "keepout_filter"]**
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.55
      **keepout_filter:
        plugin: "nav2_costmap_2d::KeepoutFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"**
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
      static_layer:
        map_subscribe_transient_local: True
      always_send_full_costmap: True
  local_costmap_client:
    ros__parameters:
      use_sim_time: True
  local_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True

global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.22
      resolution: 0.05
      track_unknown_space: true
      **plugins: ["static_layer", "obstacle_layer", "inflation_layer", "keepout_filter"]**
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.55
      **keepout_filter:
        plugin: "nav2_costmap_2d::KeepoutFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"**
      always_send_full_costmap: True
  global_costmap_client:
    ros__parameters:
      use_sim_time: True
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True`

SteveMacenski, 2020-12-01:
Is `/costmap_filter_info` being published to? Did the keepout layer filter give you any logging out that it received the info and map messages?

AlexeyMerzlyakov, 2020-12-02:
I've proceeded all steps again, from the scratch for latest `main` branch in nav2. It works fine, so this may be some mis-configuration of environment.
Are you using the latest `main` branch in navigation2 repo?
Could you please also save navigation logs and attach them here, for example by running navigation as follows:
```
$ ros2 launch nav2_bringup tb3_simulation_launch.py 2>&1 | tee run.log
<CTRL+C after robot will be localized and some move>
$ zip run.zip run.log
```

Antoinette-abi, 2020-12-02:
I checked the data that's being published on the topic `/costmap_filter_info` and i get the following response:
`The message type 'nav2_msgs/msg/CostmapFilterInfo' is invalid`. @SteveMacenski

Antoinette-abi, 2020-12-02:
@AlexeyMerzlyakov , I did build the main branch of navigation2 as given in the tutorials. I've got the following log file. 

[run.zip](https://github.com/ros-planning/navigation2/files/5630301/run.zip)
[run.log](https://github.com/ros-planning/navigation2/files/5630299/run.log)

Should I try building everything again and start over ?

AlexeyMerzlyakov, 2020-12-02:
@Antoinette-abi, from the log attached I do not see the "keepout_filter" is really enabled in the system. Did you modify `nav2_bringup/bringup/params/nav2_params.yaml` params file when added there a keepout filter?

Antoinette-abi, 2020-12-02:
Hi @AlexeyMerzlyakov , Yes i did modify the nav2_params file in `nav2_bringup/bringup/params/nav2_params.yaml` and I also double checked in the `~/navigation2_ws/install/nav2_bringup/share/nav2_bringup/params/nav2_params.yaml` file. [image] 
The keepout_filter is also added in the local costmap.
I did use `colcon build --symlink-install` to build it so I'm not sure what I've missed.

SteveMacenski, 2020-12-02:
Are you sure you're using that parameter file?

AlexeyMerzlyakov, 2020-12-03:
@Antoinette-abi, from log file you've attached before it is also seem that map file is loaded from system place:
```
[map_server-5] [INFO] [map_io]: Loading yaml file: /opt/ros/foxy/share/nav2_bringup/maps/turtlebot3_world.yaml
```
So, this might be happen when you have navigaion2 installed twice: one time as ROS2 packages and another - built manually in `~/navigation2_ws`. You need to clean-out nav2 packages from the system and use manually built ones.

Also, please check the environment when you starting ROS2+Nav2 usage. As written in the ticket header, there is ROS2 rolling build from source. For that you just need to include the following items in your environment preparation (and nothing more, I belive), as shown in the example below:
```
source $HOME/ros2_rolling/install/setup.bash
source $HOME/nav2_depend_ws/install/setup.bash
source $HOME/navigation2_ws/install/setup.bash
source $HOME/tutorials_ws/install/setup.bash
```

Antoinette-abi, 2020-12-03:
Now I understand what the problem is. I'll remove the duplicating nav2 packages and try it again. Will update you once it works. Thank you!

SteveMacenski, 2020-12-10:
Did this work?

Antoinette-abi, 2020-12-10:
@SteveMacenski Yes! I removed all the duplicate packages and re did everything.
So the problem was with the existing navigation packages.
Thank you both for your help.

closed 2020-12-10 by SteveMacenski