ros-navigation/navigation2 #2117: RewrittenYaml - living in our launchfiles, waiting to show their full potential?

gramss, 2020-12-09:
Continuing the discussion out of https://github.com/ros-planning/navigation2/pull/1958#discussion_r538937338

What does [Rewritten_Yaml.py](https://github.com/ros-planning/navigation2/blob/8318930682f9280b0d1a740537d805228ed688b2/nav2_common/nav2_common/launch/rewritten_yaml.py) do? 

Let's say you have a few dozen parameters in your new plugin and want to test sub categories of it in bigger `system_tests`.
So how would you do that?
Either, you try to create a dedicated .yaml just for that corner case (that you need to keep up-to-date) or you try to add the functionality through LaunchConfiguration + DeclareLaunchArgument + node_impl.

Current situation: https://github.com/ros-planning/navigation2/search?q=configured_params

It lives in quite a few launch files (most of them testing related). But the configured_params (instance of RewrittenYaml) is mostly not used correctly.

```
import os
import sys

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch import LaunchService
from launch.actions import ExecuteProcess, IncludeLaunchDescription, SetEnvironmentVariable
from launch.launch_context import LaunchContext
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node
from launch_testing.legacy import LaunchTestService

from nav2_common.launch import RewrittenYaml

import yaml

bringup_dir = get_package_share_directory('nav2_bringup')
params_file = os.path.join(bringup_dir, 'params', 'nav2_params.yaml')

# Replace the default parameter values for testing special features
# without having multiple params_files inside the nav2 stack
context = LaunchContext()
param_substitutions = {}

param_substitutions.update({'use_sim_time': 'False'})

if (os.getenv('GROOT_MONITORING') == 'True'):
    param_substitutions.update({'enable_groot_monitoring': 'True'})

configured_params = RewrittenYaml(
    source_file=params_file,
    root_key='',
    param_rewrites=param_substitutions,
    convert_types=True)

new_yaml = configured_params.perform(context)

print ("This is definitely not a dict: " + str(configured_params))

with open(params_file) as file:
    list = yaml.load(file, Loader=yaml.FullLoader)
    print ("Orig: " + str(list['bt_navigator']['ros__parameters']['use_sim_time']))

with open(new_yaml) as file:
    list = yaml.load(file, Loader=yaml.FullLoader)
    print ("New: " + str(list['bt_navigator']['ros__parameters']['use_sim_time']))
```

Output:

> This is definitely not a dict: <nav2_common.launch.rewritten_yaml.RewrittenYaml object at 0x7f1a5e858430>
> Orig: True
> New: False

Also:
> print ("This is definitly not a dict: " + str([configured_params]))
>> This is definitly not a dict: [<nav2_common.launch.rewritten_yaml.RewrittenYaml object at 0x7f1a5e858430>]

But heavily used here: https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/bringup/launch/navigation_launch.py My only guess is that each node gracefully has good defaults set.. (that it takes from nav2_params.yaml)

Or I'm doing something substantially wrong. I'm up to change my mind!

No the question stays, as pointed out in my comment and also @SteveMacenski suggestion.. Is this valuable enough or should we throw it out?

I vote for keep and fix with the following upcoming agenda:

- [ ] go through all instances of RewrittenYaml and look if it is used correctly by not only looking but also testing it
- [ ] remove / fix the broken instances of it
- [ ] add the feature in documentation
- [ ] Use it in actual test scenarios
- [ ] Improve it and/or try to merge it into the Launch system (there exists Substitutions, but this is also hard to grasp 
without any proper Documentation -> RewrittenYaml had some unique advantages over Substitutions, at least from the undocumented user perspective)

SteveMacenski, 2020-12-09:
@orduno @cdelsey @crdelsey @mkhansenbot can you shed some light on this? I remember when these scripts were added, but I wasn't paying enough attention to the launch customization.

mkhansenbot, 2020-12-15:
@crdelsey was the author of the Rewritten Yaml, I'll ping him via email too and ask him to chime in here. It's been awhile but this was definitely useful to be able to override key parameters in the parameter file, back on Crystal. I don't know if ROS launch now supports doing this in a better way now on Foxy?

SteveMacenski, 2020-12-15:
Well I think the thing we identified is that it's not actually doing anything, there's a`perform` function which does the rewriting which none of the launch files actually call. That's what sparked this discussion

crdelsey, 2020-12-16:
The perform function gets called by ros launch when it actually performs the launch.

The purpose of this module is to allow you to modify a yaml file based on ros launch parameters. 

This here is a good example of how it is intended to be used:
https://github.com/ros-planning/navigation2/blob/0cdf4f44c7a4171bfa4c6eb885d46c3c19f26a87/nav2_bringup/bringup/launch/navigation_launch.py#L53-L63

Here we have a default params file that is the template `params_files`. This template has 4 fields in it that we want to modify based on ros launch parameters:

 - use_sim_time
 - default_bt_xml_filename
 - autostart
 - map_subscribe_transient_local

Ideally we could do something like:
```python
f = yaml.load(params_file)
params_file['use_sim_time'] = use_sim_time # set field in params file to ros launch parameter
``` 

The trouble is, `use_sim_time` (the launch parameter) is not a string yet. It is a Substition that doesn't get resolved till the launch is performed sometime after `generate_launch_description` is called.

What RewrittenYaml does is turns the whole yaml file into a Substition. 

When ros launch performs the launch, it tries to launch a node, sees that the `parameters` field is a Substition (our RewrittenYaml) and calls our perform function.

The peform function in RewrittenYaml then:
 - resolves all the ros launch parameter substitions in `params_rewrites` and converts them to strings.
 - goes through the yaml template and replaces each key listed with the resolved string.
 - Saves the modified yaml as a temporary file
 - returns the path to that temporay file

We then go through the `params_rewrites` dictionary and resolve all those substitions (the ros launch parameters). Only at that point can they be turned into strings. We then modify the yaml with those strings, create a tempory file and pass that back to be used as the parameters for the node.

gramss, 2020-12-16:
Thank you for this great write up!!
Now I know how inheritance works in Python.....

I tried to work up the Substitution part from the Launch system but failed due to the complex structure lacking good documentation.

But am I right about the nested key part? 
https://github.com/ros-planning/navigation2/blob/6a0c92cad7fb8bb1e3e8500742bc293cddbedc36/nav2_system_tests/src/costmap_filters/test_speed_launch.py#L46-L54

With invoking `perform` manually, I was not able to address a key in a multi-key dict with specifying the key-chain.

So this:
```
param_substitutions = { 
     'planner_server.ros__parameters.GridBased.use_astar': 'True'}
```
Does not work with the manually invoked `perform`. 
But this does:

```
param_substitutions = { 
     'use_astar': 'True'}
````

Meaning that I would not be able to rewrite `use_sime_time` in the same yaml file differently. This does not make sense in the `use_sim_time` context, but might introduce strange errors if parameter-names might not be unique.

crdelsey, 2020-12-16:
Yes. That is a limitation of what I wrote. It was good enough at the time, but doesn't work for your use case. Let me look to see what it would take to add what you describe.

SteveMacenski, 2020-12-16:
That's a helpful description, thanks Carl! Question: then why include `use_sim_time` in the param files at all? Why not omit from param files (so no modifications required between real-life and sim use) and then use this to "add" the commandline defined sim time param value (that might default to `True` as a launch config)?

How does `use_sim_time` work (when that expands out to `<node name>.ros__params.use_sim_time` for all nodes) but other nested parameters don't?

crdelsey, 2020-12-17:
Well, I think we wanted a config file that was a good default without having to tweak anything, so keeping use_sim_time in the file with the standard value made some sense.

But also, RewrittenYaml is pretty dumb. It doesn't understand the Yaml structure at all. It just finds all leave keys in the yaml and, if a key by that name appears in `params_rewrites`, it changes it. If we left use_sim_time or other keys out of the original yaml, we'd have to somehow convey to RewrittenYaml where exactly it was supposed to add the key. Effectively, the original yaml defines the correct structure as well as the default values.

crdelsey, 2020-12-17:
@gramss Here's what I'd recommend to allow you to specify a particular key to change instead of changing all keys of a given name:
1. Instead of separating names with a `.` , I'd recommend making a tuple of keys since `.` is a valid character in a yaml identifier. So `'planner_server.ros__parameters.GridBased.use_astar'` becomes `('planner_server', 'ros__parameters', 'GridBased', 'use_astar')`
2.  At the end of this function:
https://github.com/ros-planning/navigation2/blob/8318930682f9280b0d1a740537d805228ed688b2/nav2_common/nav2_common/launch/rewritten_yaml.py#L108-L112
I'd add a section to deal with the absolute keys
```python
def substitute_params(self, yaml, param_rewrites): 
     for key in self.getYamlLeafKeys(yaml): 
         if key.key() in param_rewrites: 
             raw_value = param_rewrites[key.key()] 
             key.setValue(self.convert(raw_value)) 
    for param in param_rewrites.keys():
        if type(param) == tuple:
            raw_value = param_rewrites[param] 
            current_key = yaml
            for key in param:
               current_key = current_key[key]  # get the next key down
           current_key.set_value(self.convert(raw_value))
```

SteveMacenski, 2021-01-05:
@gramss does that answer everything? A Pr would be great :smile:

SteveMacenski, 2021-01-22:
I did some playing with this on the pure pursuit PR to add full remapping of parameter paths. I found that you probably _do_ need to call the perform function in the launch file, as is with the system tests was doing that before I modified it.

Rayman, 2021-02-12:
Is it even necessary to rewrite the yaml? [This PR](https://github.com/ros2/launch_ros/pull/158) introduces a new concept to ros2 launch to set a parameter for every node. If you use `SetParameter` for `use_sim_time`, you don't have to pass it trough every launch file and it can simplify the launch files.

crdelsey, 2021-02-12:
That PR looks promising. I haven't played with it to know exactly what it can and can't do, but on first look, it seems like it could be used in most cases where we currently use RewrittenYaml

SteveMacenski, 2021-02-12:
This issue was fixed - many launch files were missing the `perform` call (some had, some didn't). This change was merged in and tested to work as part of the new regulated pure pursuit controller I developed. 

It is tested working because I use it to set the new controller / planner algorithms to be able to test more than just the default navfn/DWB in CI. 

Closing, since the issue brought up has been resolved just due to inconsistent use in different files.

SteveMacenski, 2021-02-12:
The use sim time thing is interesting though! I'd love to remove that...

closed 2021-02-12 by SteveMacenski