ros-navigation/navigation2 #2120: publish_voxel_map: false in voxel_layer crashes navigation stack

xivien, 2020-12-11:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 20.04
- ROS2 Version:
  - Ros2 Foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.3-1focal.20201103.171126
- DDS implementation:
  - default

#### Steps to reproduce issue
When using voxel layer in a costmap, setting publish_voxel_map to false causes the navigation stack to crash. Rest of parameters are default from wiki
```
voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: False
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /intel_realsense_r200_depth/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          obstacle_range: 2.5
          raytrace_range: 3.0
          clearing: True
          marking: True
          data_type: "PointCloud2"
```
```
[ERROR] [controller_server-1]: process has died [pid 25843, exit code -11, cmd '/opt/ros/foxy/lib/nav2_controller/controller_server --ros-args -r /tf:=tf -r /tf_static:=tf_static --params-file /tmp/tmp0343mgs1'].

```

#### Expected behavior
Costmap launches without publishing /local_costmap/voxel_grid topic

#### Actual behavior
Navigation stack crashes
#### Additional information
It may be related to this line https://github.com/ros-planning/navigation2/blob/8bd158af79b942f4053206674a33b403e6c144e4/nav2_costmap_2d/plugins/voxel_layer.cpp#L98

SteveMacenski, 2020-12-11:
Please get a trace back so we can know why. We included tutorial on our documentation website navigation.ros.org for doing so

xivien, 2020-12-12:
Thanks for the response, I downloaded the package with apt, and cannot currently download from source and recompile it with -g flag, to get full traceback, will try to update in the future.
```
Thread 1 "controller_serv" received signal SIGSEGV, Segmentation fault.
0x00007ffff0cceaa0 in nav2_costmap_2d::VoxelLayer::onInitialize() () from /opt/ros/foxy/lib/liblayers.so
(gdb) backtrace
#0  0x00007ffff0cceaa0 in nav2_costmap_2d::VoxelLayer::onInitialize() () from /opt/ros/foxy/lib/liblayers.so
#1  0x00007ffff7c76556 in nav2_costmap_2d::Costmap2DROS::on_configure(rclcpp_lifecycle::State const&) () from /opt/ros/foxy/lib/libnav2_costmap_2d_core.so
#2  0x0000555555595604 in ?? ()
#3  0x00007ffff7dedc88 in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::execute_callback(unsigned int, rclcpp_lifecycle::State const&) () from /opt/ros/foxy/lib/librclcpp_lifecycle.so
#4  0x00007ffff7dee1cd in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::change_state(unsigned char, rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn&) () from /opt/ros/foxy/lib/librclcpp_lifecycle.so
#5  0x00007ffff7dee6d8 in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::on_change_state(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >) () from /opt/ros/foxy/lib/librclcpp_lifecycle.so
#6  0x00007ffff7de5562 in std::_Function_handler<void (std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >), std::_Bind<void (rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::*(rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>))(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<rmw_request_id_t>&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >&&) () from /opt/ros/foxy/lib/librclcpp_lifecycle.so
#7  0x00007ffff7de7eab in rclcpp::Service<lifecycle_msgs::srv::ChangeState>::handle_request(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<void>) () from /opt/ros/foxy/lib/librclcpp_lifecycle.so
#8  0x00007ffff7ed290d in ?? () from /opt/ros/foxy/lib/librclcpp.so
#9  0x00007ffff7ed335c in ?? () from /opt/ros/foxy/lib/librclcpp.so
#10 0x00007ffff7ed35cc in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>) () from /opt/ros/foxy/lib/librclcpp.so
#11 0x00007ffff7ed4472 in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () from /opt/ros/foxy/lib/librclcpp.so
#12 0x00007ffff7ed8bdc in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/foxy/lib/librclcpp.so
#13 0x00007ffff7ed6678 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /opt/ros/foxy/lib/librclcpp.so
#14 0x0000555555592469 in ?? ()
#15 0x00007ffff75500b3 in __libc_start_main (main=0x5555555923a0, argc=8, argv=0x7fffffffd8e8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd8d8) at ../csu/libc-start.c:308
#16 0x000055555559260e in ?? ()
```

SteveMacenski, 2020-12-12:
I see the issue

https://github.com/ros-planning/navigation2/blob/f237751f557b2e921e08f96d4749bfe3f2b2b091/nav2_costmap_2d/plugins/voxel_layer.cpp#L93-L98

The `on_activate` is not in the if statement, so if you set to false, it tries calling on `nullptr`. A quick patch is just to move that function under the statement

SteveMacenski, 2020-12-12:
https://github.com/ros-planning/navigation2/pull/2121 please test that this fully resolves your issue, but I suspect it does.

SteveMacenski, 2020-12-14:
Closing, I validated locally this fixed the issue

closed 2020-12-14 by SteveMacenski