ros-navigation/navigation2 #2129: KinematicsHandler::on_parameter_event_callback gets called many times for no reason

tal-grossman, 2020-12-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - 0.4.5-1focal.20201210.084248
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
print something in KinematicsHandler::on_parameter_event_callback: [here](https://github.com/ros-planning/navigation2/blob/0ed83f084d105f3eae667e0b7a28431939a07c96/nav2_dwb_controller/dwb_plugins/src/kinematic_parameters.cpp#L136).

#### Expected behavior
callback is called only when sending a ParameterEvent with kinematics params changed.
#### Actual behavior
callback is called on many unrelated events in the system. 
#### Additional information
- I think these [PR](https://github.com/ros2/rclcpp/pull/1257/files) and [commit](https://github.com/ros2/rclcpp/commit/f2cd2fbf0e2737a94585666dd6c4eeffaa3b50e9) has something do to with it since they change the topic "parameter_events" to global "/parameter_events"
- Maybe it is worth considering using service instead of topic event callback

SteveMacenski, 2020-12-17:
What "unrelated" events in the system are you referring to? This is not enough information to be actionable. Please give us full context and a way to reproduce.

tal-grossman, 2020-12-17:
> What "unrelated" events in the system are you referring to? This is not enough information to be actionable. Please give us full context and a way to reproduce

You are right, I wasn't specific because if you have the print you see it a lot. **Even when just run or launch another node or launch file**.  

I filed a ticket here since I guess that is not the behavior that you are looking for. If I'm not mistaking, on every callback the parameters are loaded and updated even if there was no actual change.
So by getting in the callback function on a random system event (such as running a new node), is just not needed work.

SteveMacenski, 2020-12-17:
> I wasn't specific because if you have the print you see it a lot

However, I need you to be specific so we can do anything about it. I will say that its expected that if you change `10` parameters, that it'll trigger `10` times, the callbacks are triggered for every individual parameter change. 

>  So by getting in the callback function on a random system event (such as running a new node), is just not needed work.

tal-grossman, 2020-12-17:
> What I suspect is that your issue is with expected behavior of the parameter events, which isn't really the role of Nav2 to dictate, we just use it
 Yes sure, but is this the right tool for the job?
Currently the result is the callback being triggered due to any parameter event and mostly not a desired kinematic parameter change.
Furthermore, it iterates over the names of the parameters and changes them by name only. So if I have another parameter with the same name but from different application or node or name space, it will change the controller parameter as well.

Anyway I hear you, and I will close this issue here as long we are on the same page. Should I close it?

SteveMacenski, 2020-12-17:
>  Currently the result is the callback being triggered due to any parameter event and mostly not a desired kinematic parameter change.

That's expected behavior. There's no way for the node to discriminate by which plugin / user within the node will be interested in that new parameter value. 

I agree that parameter events should be isolated to a given node (e.g.  `max_vel` param changes should be made to `controller_server/max_vel` such that it doesn't impact other nodes in other processes like `planner_server/max_vel`). It would be useful to have a general "change all params named XYZ" but I don't think that should be the default behavior.

That's an issue for ROS2 design or rclcpp though, we're just consumers of it

tal-grossman, 2020-12-17:
Ok, thank you. Closing it.

closed 2020-12-17 by tal-grossman