ros-navigation/navigation2 #2134: allow_unknown can't configure on dashing-devel branch

drtritmwin, 2020-12-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- ROS2 Version:
  - Dashing binary

#### Expected behavior
navfn plan a path without passing an unknown space
#### Actual behavior
navfn plan a path passing an unknown space
#### Additional information
**the parameter "allow_unknown" has been set as const in navfn_planner packages in ROS dashing branch**

----

As I mentioned, I can see that the variable  ""allown_unknown_" in only dashing branch has been set as
 
"const bool allow_unknown_{true};"

That variable is in /navigation2/nav2_navfn_planner/include/nav2_navfn_planner/navfn_planner.hpp, line number 147
So I can't configue that parameter

SteveMacenski, 2020-12-22:
Please submit a PR to add that parameter, thanks!

SteveMacenski, 2021-01-05:
Merged!

closed 2021-01-05 by SteveMacenski