ros-navigation/navigation2 #2137: Configuring dwb_controller to output higher velocities and support holonomic motion

Yadunund, 2020-12-23:
## Bug report
Hi @SteveMacenski ,

I have a custom holonomic base which I have configured to work with Navigation 2. I'm able to navigate around but there are a couple of issues I want to highlight/ seek your advise on.
1. Increasing the `max_vel_x`, `min_vel_x` , `acc_lim_x`, `decel_lim_x`and other such kinematic parameters of the `dwb_controller` does not seem to have any effect. The controller still outputs the same (slow) velocities for the robot to follow. How do I configure the controller to make the robot move faster while navigating?
2. Despite populating the `max_vel_y` and `min_vel_y` parameters, the controller only outputs X and theta velocities and hence the motion of the robot is similar to a diff drive setup. I have the complete list of parameters below for your reference. I also tried the values specified [here](https://github.com/ros-planning/navigation2/issues/954#issuecomment-542308256) but the robot still never receives any Y velocities from the controller.

Thanks for your time with this ticket.

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy Binaries
- Version or commit hash:
  - ros-foxy-navigation2	0.4.5-1focal.20201210.084248
- DDS implementation:
  - cyclone 

#### Steps to reproduce issue

```
dwb_controller:
  ros__parameters:
    use_sim_time: False
    debug_trajectory_details: True
    min_vel_x: -2.0
    min_vel_y: -2.0
    max_vel_x: 2.0
    max_vel_y: 2.0
    max_vel_theta: 2.0
    min_speed_xy: -2.0
    max_speed_xy: 2.0
    min_speed_theta: 0.7
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.001
    min_theta_velocity_threshold: 0.001
    acc_lim_x: 5.0
    acc_lim_y: 5.0
    acc_lim_theta: 6.4
    decel_lim_x: -5.0
    decel_lim_y: -5.0
    decel_lim_theta: -6.4
    vx_samples: 20
    vy_samples: 20
    vtheta_samples: 20
    sim_time: 1.7
    linear_granularity: 0.05
    xy_goal_tolerance: 0.30
    yaw_goal_tolerance: 0.17
    transform_tolerance: 0.1
    critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
    BaseObstacle.scale: 0.02
    PathAlign.scale: 0.0
    GoalAlign.scale: 0.0
    PathDist.scale: 32.0
    GoalDist.scale: 24.0
    RotateToGoal.scale: 32.0
```

SteveMacenski, 2020-12-23:
PRs are always appreciated 🙂

Yadunund, 2020-12-23:
If the problem is with the controller and not the configuration file then I can look into it 👍🏼

Yadunund, 2020-12-23:
Looks like the launch argument name for the parameter file was changed from `params` in [Dashing](https://github.com/ros-planning/navigation2/blob/942ea483863e3268023cb0cd504c93632fcbf155/nav2_bringup/launch/nav2_navigation_launch.py#L59) to `params_file` in [Foxy](https://github.com/ros-planning/navigation2/blob/3d166b748bd704eca05ef671b0b2ca99968a8c73/nav2_bringup/bringup/launch/navigation_launch.py#L82)

As my launch file was using the older `params` argument, the default parameter file was being loaded each time. Hence the changes I was making to my parameter file were useless. 

After correcting the argument name, I am happy to share that the `dwb_controller` works as expected for holonomic bases. I am also able to adjust the velocities with the relevant parameters!

closed 2020-12-23 by Yadunund