ros-navigation/navigation2 #2138: Latest release docker image nav2_util errors

AlexKaravaev, 2020-12-25:
## Bug report

**Required Info:**

- Docker image:
rosplanning/navigation2:main.release
  
#### Steps to reproduce issue
1. Run ```docker run -it --rm rosplanning/navigation2:main.release /bin/bash```

#### Expected behavior
Launch docker image without any errors

#### Actual behavior
```
not found: "/opt/overlay_ws/install/nav2_util/share/nav2_util/local_setup.bash"
root@888914ad3acc:/opt/overlay_ws# 
```

#### Additional information

Furthermore, if I go into /opt/overlay_ws and run
```
colcon build
```

The ws will not compile, giving an error
```
Finished <<< dwb_msgs [4.41s]                                                           
--- stderr: nav2_util                                
In file included from /opt/ros/foxy/include/rclcpp/client.hpp:40,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:35,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /opt/overlay_ws/src/navigation2/nav2_util/include/nav2_util/costmap.hpp:21,
                 from /opt/overlay_ws/src/navigation2/nav2_util/src/costmap.cpp:17:
/opt/overlay_ws/src/navigation2/nav2_util/src/costmap.cpp: In constructor ‘nav2_util::Costmap::Costmap(rclcpp::Node*, bool, bool, int, int)’:
/opt/overlay_ws/src/navigation2/nav2_util/src/costmap.cpp:42:28: error: format ‘%f’ expects argument of type ‘double’, but argument 5 has type ‘int’ [-Werror=format=]
   42 |       node_->get_logger(), "Costmap: Lethal threshold set to %.2f, it should be within"
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   43 |       " bounds 0-100. This could result in potential collisions!", lethal_threshold_);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~
      |                                                                    |
      |                                                                    int
```

AlexKaravaev, 2020-12-25:
Building Dockerfile from source, did not help too.
I've experienced erros like
```
rclcpp::Duration dur1(rclcpp::Duration::from_nanoseconds(connect_timeout_));          
```

Which lead me to think, that it may be eloquent and foxy issue.

I've stumbled on this commit. On november 25, osrf changed image osrf/ros2 to be not eloquent but foxy. 

https://github.com/osrf/docker_images/commit/f90b381b9b30b6656f461ddb424a73b6bbaa27c7#diff-a18be51943c1a1bb8fded1e027f2ee31b1af3d8f58571ecd6517a4aceff3c01b

I guess it could be the source of an issue here

SteveMacenski, 2020-12-25:
Please fill in the PR template, and it sounds like the issue is a compiler warning that a PR could fix to switch the print type.

wilcobonestroo, 2020-12-28:
> ```
> colcon build
> ```
>
> The ws will not compile, giving an error
>
> ```
> Finished <<< dwb_msgs [4.41s]
> --- stderr: [...]

I had the same issue. This one looks like it can easily be solved by changing the %.2f to %i, so it prints an integer. I will try this one and make a PR of it.

AlexKaravaev, 2020-12-29:
@wilcobonestroo I am just not sure, that this is simply changing the format. The main buld of navigation2 is should be running on eloquent, isn't it? But the docker image is foxy, not eloquent

SteveMacenski, 2021-01-01:
Eloquent is EOL, no one will support eloquent at this point

wilcobonestroo, 2021-01-02:
@AlexKaravaev Main is Foxy at the moment. Eloquent has already reached end of life, so it is no longer supported or maintained. Unless you have very specific dependencies on Eloquent, I would switch to Foxy if that is possible.

wilcobonestroo, 2021-01-02:
@AlexKaravaev Can this issue be closed?

SteveMacenski, 2021-01-05:
This has been fixed with https://github.com/ros-planning/navigation2/pull/2139

closed 2021-01-05 by SteveMacenski