ros-navigation/navigation2 #2143: Original error: Could not load library: libnav2_change_goal_node_bt_node.so

allstarry, 2021-01-04:
I'm trying to use nav2 under foxy and launch it with default navigation_launch.py, then this happened. Here's the error log:

[bt_navigator-4] [INFO] [1609754847.723031462] [bt_navigator]: Configuring
[bt_navigator-4] [ERROR] [1609754847.755155329] []: Caught exception in callback for transition 10
[bt_navigator-4] [ERROR] [1609754847.755174084] []: Original error: Could not load library: libnav2_change_goal_node_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-4] [WARN] [1609754847.755196507] []: Error occurred while doing error handling.
[bt_navigator-4] [FATAL] [1609754847.755206467] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-6] [ERROR] [1609754847.755652729] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-6] [ERROR] [1609754847.755674959] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

SteveMacenski, 2021-01-04:
Please properly fill out the PR template.

SteveMacenski, 2021-01-13:
@allstarry can you fill in the template with your issue and a complete description? If not, this will be closed on Friday

closed 2021-01-22 by SteveMacenski