ros-navigation/navigation2 #2145: Navigation2 nodes missing from `ros2 node list` upon relaunch

nielsvd, 2021-01-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - ros-foxy-navigation2	0.4.5-1focal.20201210.084248
- DDS implementation:
  - Fast-RTPS (default)

#### Steps to reproduce issue

###### 1
From the workspace root, launch (e.g.) a TurtleBot3 simulation:
```
export TURTLEBOT3_MODEL=burger
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:$(pwd)/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
Then, in a second terminal, launch the navigation:
```
export TURTLEBOT3_MODEL=burger
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=true
``` 
Print the node list:
```
ros2 node list
```
Close (ctrl-c) the navigation and the simulation.

###### 2
Relaunch from the same respective terminals, the simulation:
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
and the navigation:
```
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=true
```
Print the node list again (2nd time):
```
ros2 node list
```
Close (ctrl-c) the navigation and the simulation. Stop the `ros2` daemon:
```
ros2 daemon stop
```

###### 3
Relaunch from the same respective terminals, the simulation:
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
and the navigation:
```
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=true
```
Print the node list again (3rd time):
```
ros2 node list
```

#### Expected behavior
The node list should be the same all three times (up to some hash in the `/transform_listener_impl_...` nodes).

#### Actual behavior
The second time, the following nodes are missing (the remainder is practically the same):
```
/controller_server
/controller_server_rclcpp_node
/global_costmap/global_costmap
/global_costmap/global_costmap_rclcpp_node
/global_costmap_client
/local_costmap/local_costmap
/local_costmap/local_costmap_rclcpp_node
/local_costmap_client
/planner_server
/planner_server_rclcpp_node
```
The third time, after stopping the daemon, it works as expected again.

SteveMacenski, 2021-01-07:
Ok, but is anything actually wrong? Are any of the servers not actually up and doing their job?

If not, this is the wrong place for this ticket. Try filing something with ros2 cli since this is just what `ros2 node list` is outputting. There's no action here we can take or anything specific to this project.

nielsvd, 2021-01-08:
Apart from listing the nodes and getting parameters from the CLI, everything works fine.

The reason I first opened the issue here, and not with `ros2cli`, is because only nodes from the `navigation2` stack have so far shown this behavior here. If you say this is a coincidence, I can move the issue to `ros2cli`.

SteveMacenski, 2021-01-08:
It might not be a coincidence (eg launching a dozen lifecycle nodes simultaneously) but almost certainly not specific to nav2. We’re one of the major ROS2 projects that makes extensive use of alot of the tools so its not infrequent for us to identify these kinds of issues

SteveMacenski, 2021-01-13:
Closing here - no action we can take

closed 2021-01-13 by SteveMacenski