ros-navigation/navigation2 #2147: Create nav2 composition package

SteveMacenski, 2021-01-08:
Something to contain manually composed nav2 system nodes so that even though the launch system doesn't deal with component-lifecycle nodes well, we can still get the benefits of both (even if not as flexible). This will be good enough for most users as they're not going to be swapping out the task servers.

SteveMacenski, 2021-01-08:
e.x. https://github.com/ros2/demos/blob/master/composition/src/manual_composition.cpp

NithishkumarS, 2021-01-22:
> Something to contain manually composed nav2 system nodes so that even though the launch system doesn't deal with component-lifecycle nodes well, we can still get the benefits of both (even if not as flexible). This will be good enough for most users as they're not going to be swapping out the task servers.
> 
> Per idea / request from Emerson in the WG meeting
So ideally where should this composition file be located?

NithishkumarS, 2021-01-22:
So ideally where should this composition file be located?

SteveMacenski, 2021-01-22:
In a new package, `nav2_composition_bringup`

gezp, 2021-08-19:
i will do this work in the next few weeks. i make a plan list at first , and any suggestions are welcome.

- [ ] fix export dependency and library
- [ ] add argument (`node_name`,  `node_option`, etc) for Nav2 Nodes.
- [ ] create `nav2_composition` package,  and compose all nav2 nodes in  a single `main()`.
- [ ] add test, and benchmark improvements in performance.
- [ ] improve `nav2_composition` by using `zero copy` between publisher and subscription (if time allots)

gezp, 2021-09-08:
It provides a composition-based bringup method, which could replace launching Nav2 nodes separately in package `nav2_bringup`, so it's not necessary to create a new package `nav2_composition`, just put it in package `nav2_bringup`.

gezp, 2021-09-09:
i try to implement `dynamically composed bringup` for Nav2, but i failed because of some problems, i will share my experience:

* first, i try to register Nav2 nodes as the component(for exmaple, register `AmclNode` : https://github.com/gezp/navigation2/blob/dynamic_composed_bringup/nav2_amcl/src/amcl_node.cpp#L1373-L1378)
* then, launch these component in `ComposableNodeContainer` (https://github.com/gezp/navigation2/blob/dynamic_composed_bringup/nav2_bringup/launch/composed_bringup_launch.py#L103-L167)

there are some problems about loading ROS parameters :

* the first problem: `ComposableNode` couldn't load ros parameter yaml file (https://github.com/ros2/launch_ros/issues/156).
  * this issue is fixing in progress ( https://github.com/ros2/launch_ros/pull/259), this problem will be solved.
  * in order to avoid this problem, i load a default yaml file to parameter list (https://github.com/gezp/navigation2/blob/dynamic_composed_bringup/nav2_bringup/launch/composed_bringup_launch.py#L29-L37)
* the second problem: the internal Node of `ComposableNode ` couldn't load any parameter
  * `ComposableNode` is designed for a single Node, loading `ComposableNode` is based on ROS service (`composition_interfaces/srv/load_node`),  and parameters will be loaded by setting `rclcpp::NodeOptions` in service callback (https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_manager.cpp#L190), so we can't load parameter for the internal Node of `ComposableNode `. （maybe wrong）
  * In Nav2, some Nodes need use internal Node, for example, `ControllerServer` use internal Node `Costmap2DROS` for `local_costmap`. 

i have no idea about how to solve the second problem, any discussion is welcome.

gezp, 2021-09-14:
in order to solve the second problem in https://github.com/ros-planning/navigation2/issues/2147#issuecomment-915890304, i try to use argument ` --params-file` for containter Node so that the internal Node could use this argument to load parameter from file. 

it still has no effect, because  `NodeOption` of component Node uses `use_global_arguments(false)` to disable global arguments (https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_manager.cpp#L151), in addition, some other
 arguments also has no effect for component Node, you can find some details here : https://github.com/ros2/rclcpp/issues/978

gezp, 2021-09-14:
According to https://discourse.ros.org/t/nav2-composition/22175, `a large multi-threaded executor` consumes higher cpu(increase 30%-50%) than `a bunch of single-threaded executors`  in manually composed bringup of nav2. 
but `rclcpp_components` doesn't support `a bunch of single-threaded executors`  for run-time composition, which is important for  `dynamically composed bringup` of nav2 . a related issue here: https://github.com/ros2/rclcpp/issues/1772 .

gezp, 2021-09-18:
> in order to solve the second problem in [#2147 (comment)](https://github.com/ros-planning/navigation2/issues/2147#issuecomment-915890304), i try to use argument ` --params-file` for container Node so that the internal Node could use this argument to load parameter from file.
> 
sorry, it work well , i forgot to load parameters for each component Node when i use  params-file for container.

in this way, we need load component Node's parameter from `ComposableNode` (code block in launch file) and load  internal Node's parameter from Container Node (which seems like a bit strange).  

the internal Node (e.g. `local_costmap`) create their own `rclcpp::NodeOptions`, which couldn't be affected by component Node, so default `use_global_arguments(true)` is used by them, which could load parameters from container's  argument ` --params-file` .

SteveMacenski, 2021-09-27:
> in this way, we need load component Node's parameter from ComposableNode (code block in launch file) and load internal Node's parameter from Container Node (which seems like a bit strange).

Indeed, sounds like we still want to enable that option, but if there's a work around we can use for now, that works for me. Lets just continue to document the issue that does exist for solutions.

SteveMacenski, 2021-09-30:
Manual composition merged!

SteveMacenski, 2022-01-20:
Essentially done, just waiting on docs updates

closed 2022-01-20 by SteveMacenski

gezp, 2022-06-06:
> while I noticed that the ComposableNode correctly gets the parameters from the .yaml file, the costmap node within doesn't.

i guess your  .yaml file for costmap is incorrect, could you show your  .yaml file?

imstevenpmwyca, 2022-06-06:
> > while I noticed that the ComposableNode correctly gets the parameters from the .yaml file, the costmap node within doesn't.
> 
> i guess your .yaml file for costmap is incorrect, could you show your .yaml file?
```
ns_local:
    local_costmap:
        ros__parameters:
            update_frequency: 10.0
            publish_frequency: 10.0
            global_frame: odom
            robot_base_frame: base_footprint
            rolling_window: True
            (...)
```
I'm using a .yaml file that was working before converting the node to a ComposableNode.

gezp, 2022-06-06:
i didn't use `ComposableNodeContainer` to lanuch ComposableNode, so i'm not sure why it didn't work. In Nav2, i launch container as a normal node, and `--params-file` isn't used, maybe you could try this method:
```python
# use this method
params = os.path.join(get_package_share_directory('xxx'),  'config', 'parameters.yaml' )
_container = ComposableNodeContainer(
            xxx
            parameters = [params]
    )
# instead of this method
#_container = ComposableNodeContainer(
#           xxx
#           arguments = ['--params-file', path/to/parameters.yaml],
#    )

```