ros-navigation/navigation2 #2148: Bug in bt ports of type double

KvE-sax, 2021-01-11:
## Bug report
Bug found when using doubles in the behavior tree ports.

**Required Info:**

- Operating System:
  - Ubuntu 20.04 LTS
- ROS2 Version:
  - ROS2 Foxy binaries
- Version or commit hash:
- ros-foxy-navigation2	0.4.3-1focal.20200826.191445
  
- DDS implementation:
  - eProsima Fast RTPS 

#### Steps to reproduce issue

- Create behavior condition node,
- Use port to provide double (defined as double on xml and cpp side)
- Print the variable using rclcpp its logger.

#### Expected behavior
When using a port to provide the behavior tree condition node with a double, the output of the logged variable filled by the port should be the same.

#### Actual behavior
When using a port to provide the behavior tree condition node with a double, the number logged by the rclcpp logger is not the same as the input.

SteveMacenski, 2021-01-12:
Please provide a code snippet / code links of what you're reporting, I'm not sure I understand the issue

SteveMacenski, 2021-01-20:
@KvE-sax this seems like a BT.CPP issue not a nav2 issue, have you identified this as something that's a Nav2 issue or an issue with use of BT.CPP without any of our wrappers? It seems like what you're describing in that example is a BT.CPP issue since I don't see any Nav2 code running - just calling its API.

KvE-sax, 2021-01-28:
I have not tried anything without the wrappers of Nav2. So you could be right. 
Is it possible to move this issue to the BT.CPP repo by use of git or by tagging the developer of de BT.CPP lib?

SteveMacenski, 2021-01-28:
Not that I'm aware of, I'm not an admin on BT.CPP. I think you should just close it here and open one there with just the BT.CPP lines (maybe link to this though)

closed 2021-02-01 by SteveMacenski