ros-navigation/navigation2 #2150: Normalize FollowWaypoints action names to be snake cased

ymd-stella, 2021-01-14:
https://github.com/ros-planning/navigation2/blob/61710875ed1d73e807b7cd1a9e577d995f2e4ead/nav2_waypoint_follower/src/waypoint_follower.cpp#L75

Action name "FollowWaypoints" should be changed to "follow_waypoints".

This is the problem that was pointed out in https://github.com/ros-planning/navigation2/pull/1484#issue-369676718 .

SteveMacenski, 2021-01-14:
100% correct, I swear I keep fixing these and they come back somehow. I'm going crazy. Thanks for pointing that out. 

Here are all the places https://github.com/ros-planning/navigation2/search?q=FollowWaypoints

laiaaa0, 2021-01-16:
Hi, I'd like to have a go at this as my first issue, I will follow the download and build guidelines and attempt to have something by the end of the weekend :)

laiaaa0, 2021-01-18:
Oh, I see @NithishkumarS got ahead of me :) 

Just a clarification on this issue, did  the action files also need to be renamed?  I was checking the changes in   #1484 , and the `.action` file wasn't renamed there, so I was wondering what are the guidelines here.

NithishkumarS, 2021-01-19:
> 100% correct, I swear I keep fixing these and they come back somehow. I'm going crazy. Thanks for pointing that out.
> 
> Here are all the places https://github.com/ros-planning/navigation2/search?q=FollowWaypoints

When I change the name of the message in cmakelist from FollowWaypoints to follow_waypoints, looks like I need to change the name of the message in the nav_msg package.
If not all the places in the search cant be changed. Please let me know how to proceed

SteveMacenski, 2021-01-19:
The action .msg doesn't need to change, just the string action topic name used for it in the code!

SteveMacenski, 2021-01-22:
Thanks! Closing since merged

closed 2021-01-22 by SteveMacenski