ros-navigation/navigation2 #2155: Add doxygen for all files

SteveMacenski, 2021-01-22:
We have pretty good coverage on this already, but there are a couple of packages missing doxygen in the headers

SteveMacenski, 2021-01-22:
A good first time issue user might want to just add these for a couple of header files missing them! You don't have to do everything at once, even a little bit at a time is super helpful

SteveMacenski, 2021-02-04:
From a survey, these are the packages needed doxygen
- [ ] AMCL
- [ ] behavior tree
- [ ] costmap 2d
- [ ] navfn
- [ ] recoveries
- [ ] utils

mlsdpk, 2021-02-11:
@SteveMacenski this is my first time contributing to the project, I would like to take navfn first if possible.

SteveMacenski, 2021-02-11:
That would be greatly helpful, thanks @mlsdpk!

SteveMacenski, 2021-03-12:
I did utils and recoveries

SteveMacenski, 2021-04-06:
Phew.... 2,000 lines later, finally complete!

closed 2021-04-06 by SteveMacenski