ros-navigation/navigation2 #2157: Nav2 Getting Started with TurtleBot3 and tb3_simulation_launch.py on Ubuntu 20.04 and ROS Foxy

Darkproduct, 2021-01-25:
I'm writing this, because this [Getting Started Page](https://navigation.ros.org/getting_started/index.html) has several issues and this is my record of solving them, hopefully. Maybe this will help other in the future.

### First Problem:
I followed all steps and ran `ros2 launch nav2_bringup tb3_simulation_launch.py` for the first time and after a long wait I see an empty gazebo like https://github.com/ros-planning/navigation2/issues/1557 and a looping terminal message:
```
[rviz2-4] [INFO] [1611529397.058893052] [rviz2]: Message Filter dropping message: frame 'odom' at time 18,725 for reason 'Unknown'
[rviz2-4] [INFO] [1611529397.090720528] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 18,800 for reason 'Unknown'
[rviz2-4] [INFO] [1611529397.156629924] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1611529397.157158073] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-4] [INFO] [1611529397.250773404] [rviz2]: Message Filter dropping message: frame 'odom' at time 18,925 for reason 'Unknown'
[rviz2-4] [INFO] [1611529397.283108215] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 19,000 for reason 'Unknown'
[planner_server-9] [INFO] [1611529397.451650997] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```

After a bit of investigation I found out that the `turtlebot3_gazebo` folder in `/opt/ros/foxy/share` doesn't exist and this line `export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models` from the [Getting Started Page](https://navigation.ros.org/getting_started/index.html) does nothing.

Unfortunately, I couldn't find a way to get these models without cloning the [turtlebot3_simulations](https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git) repo. 
What I tried:
 - Searching through all turtlebot3 folders in `/opt/ros/foxy/share` but there is no model folder that I could use for Gazebo. 
 - It looks like no one is using the binary packages of turtlebot3 [[1]](https://github.com/ros-planning/navigation2/issues/1557#issuecomment-589860530) [[2]](https://ubuntu.com/blog/simulate-the-turtlebot3) [[3]](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#simulation)

I don't really know if it was ever possible to install the models through `apt install` so they would show up in `/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models`, but on Ubuntu 20.04 with ROS Foxy this doesn't seem possible.

I solved this by cloning [turtlebot3_simulations](https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git) by hand and using the new model path in `export GAZEBO_MODEL_PATH`.

### Second Problem:
ros2 crashes after setting the `2D Pose Estimate` in rviz. Here is the output log after setting the pose.
```
[amcl-6] [INFO] [1611536662.876677162] [amcl]: Setting pose (153.234000): -1.699 -0.459 0.166
[rviz2-4] [INFO] [1611536662.968005764] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 151,400 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.968173176] [rviz2]: Message Filter dropping message: frame 'odom' at time 151,336 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.969955064] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.970229335] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.970421472] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.970602433] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.970990687] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.971207687] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.971407436] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.971632189] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.971848142] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[rviz2-4] [INFO] [1611536662.972041395] [rviz2]: Message Filter dropping message: frame 'odom' at time 0,000 for reason 'Unknown'
[planner_server-9] [INFO] [1611536663.039952271] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation at time 153.476000, but only time 154.401000 is in the buffer, when looking up transform from frame [base_link] to frame [map]
[rviz2-4] [INFO] [1611536663.154084615] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 151,600 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.155362939] [rviz2]: Message Filter dropping message: frame 'odom' at time 151,536 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.346821818] [rviz2]: Trying to create a map of size 60 x 60 using 1 swatches
[rviz2-4] [INFO] [1611536663.348729771] [rviz2]: Message Filter dropping message: frame 'odom' at time 151,736 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.378993167] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 151,801 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.538643038] [rviz2]: Message Filter dropping message: frame 'odom' at time 151,936 for reason 'Unknown'
[planner_server-9] [INFO] [1611536663.539850752] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 153.952000 but the earliest data is at time 154.401000, when looking up transform from frame [base_link] to frame [map]
[rviz2-4] [INFO] [1611536663.570598993] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 152,001 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.762722003] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 152,201 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.762876983] [rviz2]: Message Filter dropping message: frame 'odom' at time 152,135 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.827972931] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1611536663.828189373] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-4] [INFO] [1611536663.955715667] [rviz2]: Message Filter dropping message: frame 'odom' at time 152,336 for reason 'Unknown'
[rviz2-4] [INFO] [1611536663.987277292] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 152,401 for reason 'Unknown'
[planner_server-9] [INFO] [1611536664.039900651] [global_costmap.global_costmap]: start
[planner_server-9] [INFO] [1611536664.091625419] [planner_server]: Activating plugin GridBased of type NavfnPlanner
[lifecycle_manager-13] [INFO] [1611536664.092353665] [lifecycle_manager_navigation]: Activating recoveries_server
[recoveries_server-10] [INFO] [1611536664.093283826] [recoveries_server]: Activating
[recoveries_server-10] [INFO] [1611536664.093335858] [recoveries_server]: Activating spin
[recoveries_server-10] [INFO] [1611536664.093354786] [recoveries_server]: Activating backup
[recoveries_server-10] [INFO] [1611536664.093371129] [recoveries_server]: Activating wait
[lifecycle_manager-13] [INFO] [1611536664.094415202] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-11] [INFO] [1611536664.095110832] [bt_navigator]: Activating
[lifecycle_manager-13] [INFO] [1611536664.095731312] [lifecycle_manager_navigation]: Activating waypoint_follower
[waypoint_follower-12] [INFO] [1611536664.096035615] [waypoint_follower]: Activating
[lifecycle_manager-13] [INFO] [1611536664.097055383] [lifecycle_manager_navigation]: Managed nodes are active
[ERROR] [rviz2-4]: process has died [pid 26484, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /opt/ros/foxy/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].
```

Solved with ^C and try again. On the second run everything worked.

## TLDR
### First Fix: Missing Models
Clone [turtlebot3_simulations](https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git) and use
```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/<PATH_TO>/turtlebot3_simulations/turtlebot3_gazebo/models
```
### ROS2 Crash
I don't really know what happened there.

SteveMacenski, 2021-01-25:
Please ask on ROS Answers please - the issue tracker are for bugs / feature requests not setup advise :smile:

closed 2021-01-25 by SteveMacenski

Darkproduct, 2021-01-25:
@SteveMacenski I'm not really sure what to say. Did you read the issue?

My problem is with this [Getting Started Page](https://navigation.ros.org/getting_started/index.html) and I have to assume that this is something from this project here?
At no point did I ask for advice. I don't use TurteBot and I don't want to use it. I'm in a Team of Students building Robots for competition and fun and I use ROS since Indigo and I think it is time to look into ROS 2. My setup is an almost new installation of Ubuntu 20.04 LTS with ROS foxy. I read all the tutorials and moved on to the navigation stack because this is in my eyes one of the best projects created in ROS.
To get the base setup from here [Getting Started Page](https://navigation.ros.org/getting_started/index.html) to work, was everything I wanted to archive and it took several hours. 

All I want to archive is a fest bug-free setup and help others to archive the same, so I wrote an Issue to let you know that there is something missing in your instructions on this [page](https://navigation.ros.org/getting_started/index.html).

Never the less, the second problem, that I couldn't solve is almost certainly a bug.

I hope this message isn't root or anything, this is not my intention. The navigation stack is a great project and I think it is awesome that everyone can work with it.

doisyg, 2021-01-28:
+1 for fixing the Getting Stared Page (same issues/frustration here)

SteveMacenski, 2021-01-28:
I'm not in control of Tb3's releases, you'd have to poke Robotis / go on their repos to have them fix it. These instructions assume a release exists and cloning / building I don't feel is a big enough issue to warrant an update during periods when this isn't available - it should be immediately clear that it didn't install on the apt command failure

doisyg, 2021-01-28:
Fair enough.
Though not necessarily the priority for the nav2 project, I believe being able to launch in a couple of minute a simulation of a basic mobile base along with nav2 on the last ROS2 LTS would really help new users and serve as a good working example.
And if depending on Tb3 causes issues (we are at Foxy release + 8 months), maybe maintaining a basic example directly in nav2 ?
We may be able to contribute in this direction if there is an interest and when my team will have a bit more experience with ROS2
(and using Ignition instead of gazebo classic for the simulator).
Maybe this basic simulation could also serve for integration tests too ?

SteveMacenski, 2021-01-29:
> I believe being able to launch in a couple of minute a simulation of a basic mobile base along with nav2 on the last ROS2 LTS would really help new users and serve as a good working example.

Agreed! That's the point of the page. Long term, I don't see us continuing to support Tb3 for this and other reasons so that might be a nonissue in a year. 

>  We may be able to contribute in this direction if there is an interest and when my team will have a bit more experience with ROS2

We could certainly talk about what that would look like! I'm always open to let people play in the sandbox if they bring their own tools. This is mostly just a statement that "I'm" not going to do anything about it. I'm more than happy to direct / enable others to though.

aboarya, 2021-07-25:
@SteveMacenski I think you should consider removing that from your Getting Started page, since it is currently broken.

I ran into the same issue as @Darkproduct.  I appreciate that this is not the turtlebot3 project, however your introduction of this nav2 project on the YouTube videos , have a common message:

> "Nav2 is well documented and easy for someone to pick it up and join the community"

Meanwhile the Getting Started example is broken!  And when someone reaches out for help, a student I might add, your initial reaction is to tell them look somewhere else for help.

altineller, 2023-03-20:
I have came across this problem while trying to 'run nav2_simple_commander nav_to_pose_example_launch.py' 

the nav2_simple_commander/warehouse.world contains a turtlebot3_waffle model inside. I do have turtlebot3_gazebo/models in my GAZEBO_MODEL_PATH, as well as the aws_robomaker_small_warehouse_world/models and I can perfectly run the simulations as well as the nav2_simple_commander programs, when I manually launch world and spawn robot

This is more like a distribution problem, I am using ubuntu22 with ros humble, so the /opt/ros/humble/share/nav2_simple_commander/warehouse.world file is the cause of the problem, and since all the demo launch files use this warehouse.world this is a slight bug, imho.

maybe modifying the launch files to spawn the robot using turtlebot_gazebo, and adding these to the requirements in package.xml could fix it.

best regards,
Can

rahulswa08, 2023-06-05:
I faced same issue. The problem was initial pose was not set and ROS is showing up the error. 
I followed this [link](https://youtu.be/QP-cxh8qUJQ?t=653) to resolve.