ros-navigation/navigation2 #2160: [ros2] Nav2 parameters tuning GUI

doisyg, 2021-01-28:
Hi,
New to ros2 an navigation2. I am trying to tune the multiple navigation parameters. In ROS1 I was using the `rqt_reconfigure` tool, what are you guys using for nav2 ? There is a port of `rqt_reconfigure` but is seems completely unusable with foxy (at least on my system). https://github.com/ros-visualization/rqt_reconfigure/tree/dashing

(Can move to answers.ros.org if needed)

SteveMacenski, 2021-01-28:
There is no GUI, but a ticket open to re-enable runtime reconfigurable parameterizations https://github.com/ros-planning/navigation2/issues/956. I think DWB is the only one with largely complete reconfigurability we were using as a POC before moving onto other topics. Dynamic reconfigure is dead in ROS 2 so I'm not sure what `rqt_reconfigure` would be / do in ROS 2.

SteveMacenski, 2021-01-28:
But step 1 would be just "get runtime reconfigurable parameters even available", before that point, a GUI is pretty useless :laughing:

doisyg, 2021-01-28:
Thanks for the clarification!
I would look into the reconfigurability of the parts I am intending to spend the most time tuning (TEB controller and Navfn planner). I see Costmap2D https://github.com/ros-planning/navigation2/pull/1370 seems already reconfigurable (at least each layer can be enable/disable), so that's a relief ^^

SteveMacenski, 2021-01-28:
I'm pretty sure we removed costmap's at some point, I'd check the actual code.

... NavFn has like 2 parameters :-) TEB isn't in this project directly so not an opinion there. 

Can this be closed? Doesn't seem actionable

closed 2021-01-28 by doisyg