ros-navigation/navigation2 #2163: Is Costmap Filters enable in ros foxy? As I can see branch ors-foxy devel has no Costmap Filters at all, but naviagtion2 tutorial said that filters are enable in ros-foxy.

Fricodelco, 2021-02-02:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2021-02-02:
Please fill in the template  (or I suppose for this type of ticket, putting info in the ticket itself and not all in the title :wink: ) 

The costmap filters change ABI of costmap 2d and won't be released into foxy, Galactic in a couple of months will be the first distribution having it batteries included, but you can use it today in the main branch. 

Please link to the tutorial you say mentions it being available in foxy, I searched through the keepout/speed tutorials and configuration guide and couldn't find the word "foxy" on the pages.

Fricodelco, 2021-02-04:
Here is the link: https://navigation.ros.org/tutorials/docs/navigation2_with_keepout_filter.html#requirements there is link to build and install part of tutorial and this part said that cos foxy is enough. Anyway now Im trying to compile the main branch with ros rolling and get this error:
--- stderr: nav2_util                                                         
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.

Any ideas how could I fix it?

Fricodelco, 2021-02-04:
I fix that one, but now get this:
In file included from /home/server/ros2_rolling/src/navigation2/nav2_util/src/lifecycle_node.cpp:15:
/home/server/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_node.hpp:25:10: fatal error: bondcpp/bond.hpp: No such file or directory
   25 | #include "bondcpp/bond.hpp"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/nav2_util_core.dir/build.make:128: src/CMakeFiles/nav2_util_core.dir/lifecycle_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/server/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/odometry_utils.hpp:29,
                 from /home/server/ros2_rolling/src/navigation2/nav2_util/src/odometry_utils.cpp:18:
/home/server/ros2_rolling/src/navigation2/nav2_util/include/nav2_util/lifecycle_node.hpp:25:10: fatal error: bondcpp/bond.hpp: No such file or directory
   25 | #include "bondcpp/bond.hpp"
      |          ^~~~~~~~~~~~~~~~~~
I have already install rot-rolling-bond, what's the problem?

SteveMacenski, 2021-02-04:
Please follow our build and install instructions verbatim. You're probably missing the entire dependencies workspace that includes bond. 

Closing - original ticket answered.

closed 2021-02-04 by SteveMacenski

charlielito, 2021-05-20:
> Please fill in the template (or I suppose for this type of ticket, putting info in the ticket itself and not all in the title  )
> 
> The costmap filters change ABI of costmap 2d and won't be released into foxy, Galactic in a couple of months will be the first distribution having it batteries included, but you can use it today in the main branch.
> 
> Please link to the tutorial you say mentions it being available in foxy, I searched through the keepout/speed tutorials and configuration guide and couldn't find the word "foxy" on the pages. I'll update them if there's something incorrect.

@SteveMacenski  do you have a ETA for the galactic release?