ros-navigation/navigation2 #2171: Can't build regulated_pure_pursuit

diogopbrandao, 2021-02-05:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- ROS2 Version:
  -  Foxy 
- Version or commit hash:
  - 98d1c8c537f55bbc5175babc0a7f6f312521e73a
  - ros-foxy-navigation2   0.4.5-1focal.20210106.033048

A few weeks ago I posted an issue on navigation2_tutorials that follows the issue that I'm writting right now.
I'm creating a trajectory controller wrapper for navigation2, following the plugin tutorial : [https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2controller_plugin.html](url)

As a first step I'm trying to make nav2_regulated_pure_pusuit package work, but I can't get pass trying to build it. No changes were made to this package.
I've tried two ways:

### First way:
 I tried to simply clone the package nav2_regulated_pure pursuit into my project, and try to build it there, but got the following errors:

```
/ros/marta_ws  colcon build --packages-select nav2_regulated_pure_pursuit_controller
Starting >>> nav2_regulated_pure_pursuit_controller
[Processing: nav2_regulated_pure_pursuit_controller]                             
--- stderr: nav2_regulated_pure_pursuit_controller                               
In file included from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp:20:
/home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/include/nav2_regulated_pure_pursuit_controller/regulated_pure_pursuit_controller.hpp:58:8: error: ' void nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController::configure(const WeakPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&) marked ‘override’, but does not override
   58 |   void configure(
      |        ^~~~~~~~~
/home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/include/nav2_regulated_pure_pursuit_controller/regulated_pure_pursuit_controller.hpp:104:8: error: ‘void nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController::setSpeedLimit(const double&)’ marked ‘override’, but does not override
  104 |   void setSpeedLimit(const double & speed_limit) override;
      |        ^~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/foxy/include/class_loader/class_loader.hpp:55,
                 from /opt/ros/foxy/include/class_loader/multi_library_class_loader.hpp:52,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:58,
                 from /opt/ros/foxy/include/nav2_costmap_2d/costmap_2d_ros.hpp:53,
                 from /opt/ros/foxy/include/nav2_core/controller.hpp:42,
                 from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/include/nav2_regulated_pure_pursuit_controller/regulated_pure_pursuit_controller.hpp:24,
                 from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp:20:
/opt/ros/foxy/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController; B = nav2_core::Controller]’:
/opt/ros/foxy/include/class_loader/meta_object.hpp:190:7:   required from here
/opt/ros/foxy/include/class_loader/meta_object.hpp:192:12: error: invalid new-expression of abstract class type ‘nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController’
  192 |     return new C;
      |            ^~~~~
In file included from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp:20:
/home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/include/nav2_regulated_pure_pursuit_controller/regulated_pure_pursuit_controller.hpp:38:7: note:   because the following virtual functions are pure within ‘nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController’:
   38 | class RegulatedPurePursuitController : public nav2_core::Controller
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/include/nav2_regulated_pure_pursuit_controller/regulated_pure_pursuit_controller.hpp:24,
                 from /home/user/ros/marta_ws/src/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp:20:
/opt/ros/foxy/include/nav2_core/controller.hpp:74:16: note:     ‘virtual void nav2_core::Controller::configure(const SharedPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’
   74 |   virtual void configure(
      |                ^~~~~~~~~
make[2]: *** [CMakeFiles/nav2_regulated_pure_pursuit_controller.dir/build.make:63: CMakeFiles/nav2_regulated_pure_pursuit_controller.dir/src/regulated_pure_pursuit_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:119: CMakeFiles/nav2_regulated_pure_pursuit_controller.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Failed   <<< nav2_regulated_pure_pursuit_controller [30.2s, exited with code 2]

Summary: 0 packages finished [30.9s]
  1 package failed: nav2_regulated_pure_pursuit_controller
  1 package had stderr output: nav2_regulated_pure_pursuit_controller
```

### Second way:
The other thing that I have tried is to just clone all the navigation2 repo, change the branch to the regulated_pure_pursuit branch, and build it, which also fails. It outputs the following:

```
 ~/ros/nav2_ws/src/navigation2   regulated_pure_pursuit  cd ..
 ~/ros/nav2_ws/src  cd ..                         
 ~/ros/nav2_ws  colcon build --symlink-install
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Finished <<< nav2_common [2.97s]                                                          
Starting >>> nav2_msgs
[Processing: nav2_msgs, nav_2d_msgs]                                                            
[Processing: nav2_msgs, nav_2d_msgs]                                                                       
Finished <<< nav_2d_msgs [1min 9s]                                                                         
Starting >>> nav2_voxel_grid
[Processing: nav2_msgs, nav2_voxel_grid]                                                                       
Finished <<< nav2_voxel_grid [43.3s]                                                                            
Starting >>> dwb_msgs
[Processing: dwb_msgs, nav2_msgs]                                                                       
[Processing: dwb_msgs, nav2_msgs]                                                                           
[Processing: dwb_msgs, nav2_msgs]                                                                            
Finished <<< dwb_msgs [1min 40s]                                                                              
Starting >>> nav2_gazebo_spawner                               
Finished <<< nav2_gazebo_spawner [2.71s]                                                                
[Processing: nav2_msgs]                                        
[Processing: nav2_msgs]                                         
Finished <<< nav2_msgs [4min 40s]                            
Starting >>> nav2_util
--- stderr: nav2_util                                
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< nav2_util [6.76s, exited with code 1]
                                       
Summary: 6 packages finished [4min 50s]
  1 package failed: nav2_util
  1 package had stderr output: nav2_util
  24 packages not processed
```

This one I find really strange since when I build on the foxy-devel or master branch I have no issues and these branches also use bondcpp. 

I apologize if this is not the right place for it, but this seems like a bug since I cannot even get it to build.
I would appreciate any suggestions to solve the building errors in both cases presented above.

#### Steps to reproduce issue

**First way** : ``` colcon build --packages-select nav2_regulated_pure_pursuit_controller``` on project workspace 
**Second way** :  `colcon build --symlink-install` on nav2_ws

#### Expected behavior
Build successfully
#### Actual behavior
Fails building

SteveMacenski, 2021-02-05:
Reg PP isn't released yet in Nav2 and won't be made available for Foxy for some time. There's a PR pending working on getting it into main now that will likely be in by the end of this week or next, but all of those issues are around changing the code for Foxy which isn't supported yet.

SteveMacenski, 2021-02-05:
Please see the getting started guide for build / install of Nav2 for the bond issues. They are clearly covered as dependencies.

closed 2021-02-05 by SteveMacenski