ros-navigation/navigation2 #2175: [DWB] poor obstacle avoidance when using small inflation radius (0.01)

sachinkum0009, 2021-02-09:
DWB Planner is not avoiding obstacles when using small inflation radius

## Param Used
```bash
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: true
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
    - nav2_goal_reached_condition_bt_node
    - nav2_goal_updated_condition_bt_node
[... 288 lines cut ...]
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  - Foxy Source

#### Steps to reproduce issue

- Used the above params or decrease the inflation radius to the 0.01

#### Expected behavior
- Should plan a path to avoid the obstacles

#### Actual behavior
- Colliding with the Obstacles

#### Additional information

[image]
[image]
[image]
[image]

sachinkum0009, 2021-02-09:
also, the planner is not able to avoid obstacles even for a larger radius as it does not consider the footprint of the robot when calculating the path.
That the path planner seems to consider the robot as a point object and plan the path just outside the inflation radius even if we increase it. The center of the robot follows the path and end up crashing the sides.
I tried to view the local_costmap/footprint but it is not showing even though I have added that in the param file
but I am able to view the global_costmap/footprint
is

SteveMacenski, 2021-02-09:
Hi,

While I feel for your concerns, this question has a number of assumptions / statements that show a somewhat lack of knowledge about how these algorithms work and what configurations mean. While usually I'm really open to helping folks get things going, the breadth of topics I'd have to explain is too much for me to be able to explain in this venue.

It looks like you're using the TB3 in the sandbox world. I'd recommend reverting to our default configs that are known to work in this robot / environment. Configuration help can be asked on ROS Answers, but again, I'm not sure anyone can give you what you need here due to body of knowledge required to fully explain. 

There is no statement that you can't misconfigure a system so badly that it will collide.

sachinkum0009, 2021-02-10:
Thank you for your response
will work on this