ros-navigation/navigation2 #2176: Nav2 rolling release ?

doisyg, 2021-02-09:
## Feature request
A nav2 rolling debian release `ros-rolling-nav2*`

#### Implementation considerations
Just opening the debate here without being sure if it is a nice idea.
Pros:
- No compilation
- Ease of installation
- Ease of compilation / testing of packages that depends on nav2, e.g STVL, TEB, etc...

Cons:
- Release overhead
- Sync issues
- People complaining about unstable release

SteveMacenski, 2021-02-09:
To start, I agree that Nav2 in the most ideal world should be on `rolling`. However, my opinion flips given the real-world time constraints of a limited maintenance team. 

I've been involved with a couple of other projects that have rolling releases and the barrage of emails / issues of the flaky build farm that are produced from the edge builds would literally drive me insane with anxiety. Let alone that Nav2 has 22 (soon to be 23) packages, each would provide an email to me on failure every couple of hours for days. 

Most of these issues are small, but the flood of notification makes it impractically cumbersome for me to handle myself. I'd be open to talking about a rolling release if there were other active maintainers and someone else stepped up to handle this type of thing, but until then, its just not something I can do and stay on top of other Nav2 issues + development + PR reviewing. 

Keep in mind our CI runs on ROS2 master builds which are slightly more bleeding edge than rolling. So by all accounts, we do "support" ROS 2 master the same way rolling does, but we do not allow for that 'apt install' of our bleeding edge code in the same way. But in place, we do have Docker containers that have everything installed and ready for use that are built nightly! So we have our own flavor of Rolling in Nav2, but with another toolset that doesn't produce a ton of noise and is pretty reliable.

doisyg, 2021-02-09:
> Most of these issues are small, but the flood of notification makes it impractically cumbersome for me to handle myself.

That was my main concern, I feel too that with any release (even the most bleeding one), will come people looking for support of those.

> But in place, we do have Docker containers that have everything installed and ready for use that are built nightly!

Good to know

Thanks for your answer, closing.

closed 2021-02-09 by SteveMacenski

gc-robotics, 2021-04-30:
Hi @SteveMacenski , 

I am trying to fix the errors that come up while building for rolling. 

There is one on which I got stuck:

`Starting >>> nav2_common
Starting >>> nav_2d_msgs
Starting >>> nav2_gazebo_spawner
Finished <<< nav2_common [1.60s]                                                                                          
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav2_gazebo_spawner [1.91s]                                                                                 
Finished <<< nav_2d_msgs [2.22s]                                                                                               
Starting >>> dwb_msgs
Finished <<< nav2_voxel_grid [0.79s]                                                                                    
Finished <<< dwb_msgs [1.65s]                                                                
Finished <<< nav2_msgs [2.67s]                   
Starting >>> nav2_util
Finished <<< nav2_util [0.63s]                     
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server
Finished <<< nav_2d_utils [1.24s]                                                                            
Starting >>> nav2_amcl
Finished <<< nav2_lifecycle_manager [1.28s]
Starting >>> nav2_waypoint_follower                                                                                            
Finished <<< nav2_map_server [2.09s]                                                                            
Starting >>> nav2_costmap_2d
Finished <<< nav2_waypoint_follower [0.88s]
Starting >>> nav2_rviz_plugins
Finished <<< nav2_amcl [1.27s]                                                                          
Starting >>> slam_toolbox
Finished <<< nav2_rviz_plugins [1.84s]                                                                           
Finished <<< nav2_behavior_tree [6.75s]                                                                           
Starting >>> nav2_bt_navigator
Finished <<< slam_toolbox [12.4s]                                                                           
--- stderr: nav2_bt_navigator                                                                                 
In file included from /test/navigation2/nav2_bt_navigator/src/bt_navigator.cpp:15:
test/navigation2/nav2_bt_navigator/include/nav2_bt_navigator/bt_navigator.hpp:140:28: error: ‘Buffer’ is not a member of ‘tf2_ros’
  140 |   std::shared_ptr<tf2_ros::Buffer> tf_;
      |                            ^~~~~~
/test/navigation2/nav2_bt_navigator/include/nav2_bt_navigator/bt_navigator.hpp:140:34: error: template argument 1 is invalid
  140 |   std::shared_ptr<tf2_ros::Buffer> tf_;
      |                                  `
Any suggestion on how to fix it? Thanks.

SteveMacenski, 2021-04-30:
> error: ‘Buffer’ is not a member of ‘tf2_ros’ 140 | std::shared_ptr<tf2_ros::Buffer> tf_;

That's... weird. You sure that's a nav2 issue? Our nightly CI runs on the ROS2 master branches and are turning over OK, are you sure you set up your dependencies correctly?